letscontrolit/ESPEasy #3347: P074_TSL2591: Incorrect sensor values after a couple of hours

heinemannj, 2020-10-28:
Plugin: P074_TSL2591

Incorrect sensor values after a couple of hours

Green LED on

[image]

All LEDs off

[image]

After opening of device settings and just close the setting page => the same issue

[image]

But after submitting but without any changes of device settings => correct sensor values after next Data Acquisition interval

[image]

Plugin Count:⋄ | 81 [Normal] [Testing] [Development] [TEST ESP32]
Binary Filename:⋄ | ESP_Easy_mega_20201025_dev_ESP32_4M316k

TD-er, 2020-10-28:
This apparently is something that may be fixed in software and perhaps also detected in software.
But since you're the first to report it with this sensor (as far as I know), I would also like to know about your setup.

Can you tell me more about:
- Cable length to sensor
- I2C clock frequency
- Presence of I2C pull-up resistors (and if any, what value)
- Are there more sensors connected to the same I2C bus?

heinemannj, 2020-10-28:
Cable length to sensor: 20 cm
I2C clock frequency: 400000 Hz
Presence of I2C pull-up resistors (and if any, what value): None
Are there more sensors connected to the same I2C bus? NO

The only devices (connected/configured) you can see in the sreenshot:

- Light sensor,
- LEDs and 
- virtual switches

The sensor is reachable over the I2C bus at every time.
I can request sensor value updates via rules at every time.
But all retuned values are 100% less than expected.

Is there any command I can execute via rules to reset the sensor before starting my measurement?
For the TSL2561 sensor plugin the "Send sensor to sleep" function is implemented. This can be configured for the TSL2591 too.

TD-er, 2020-10-28:
You can disable and enable the task from the rules.
This does respectively call PLUGIN_EXIT and PLUGIN_INIT.

See [documentation on "taskenable" and "taskdisable"](https://espeasy.readthedocs.io/en/latest/Reference/Command.html#commands-listed)
To trigger an immediate measurement, call "taskrun".

heinemannj, 2020-10-28:
taskdisable,1
Status after execution: OK

**taskenable,1
Status after execution: failed**
But under "Devices" sensor is enabled again and up and running

The taskrun command I'm already using in rules and are working fine:

on EmptySample#State=1 do
   TaskValueSet,2,1,0
   TaskValueSet,2,2,0
   TaskValueSet,2,3,0
   TaskValueSet,2,4,0
   TaskValueSet,3,1,0
   TaskValueSet,3,2,0
   TaskValueSet,3,3,0
   TaskValueSet,3,4,0
   TaskValueSet,4,1,0
   TaskValueSet,4,2,0
   TaskValueSet,4,3,0
   TaskValueSet,4,4,0
   GPIO,12,1
   timerSet,1,1
endon

On Rules#Timer=1 do
   TaskRun,1
   TaskValueSet,2,1,[TSL2591#Lux]
   GPIO,12,0
   GPIO,13,1
   timerSet,2,1
endon

On Rules#Timer=2 do
   TaskRun,1
   TaskValueSet,2,2,[TSL2591#Lux]
   GPIO,13,0
   GPIO,14,1
   timerSet,3,1
endon

On Rules#Timer=3 do
   TaskRun,1
   TaskValueSet,2,3,[TSL2591#Lux]
   GPIO,14,0
   GPIO,23,1
   timerSet,4,1
endon

On Rules#Timer=4 do
   TaskRun,1
   TaskValueSet,2,4,[TSL2591#Lux]
   GPIO,23,0
   GPIO,25,0
endon

TD-er, 2020-10-28:
I added block markings to your code, making it slightly better readable.

Tested here on the taskenable/taskdisable command and indeed the taskenable gives an error.
I will have a look at it.

TD-er, 2020-10-28:
OK, found the issue and it looks like it is not calling the PLUGIN_INIT as it should have.
I will make a fix for it.

TD-er, 2020-10-28:
It is fixed and has been merged.

So now back to this issue :)
I have looked into the code of the library and this is roughly the flow to get data:

Plugin init:
- Plugin_INIT  -> not `_initialized`
- call `begin()` -> set timing & set gain
- perform read

Normal call after init:
- `_initialized` is set, so no call to `begin()
- perform read

So I think your sensor somehow looses its set timings and gain and thus reports wrong values.

A very simple fix for this can be to check the return value of `getFullLuminosity` and if it is reporting a value of 0 (or any other known value) then set `_initialized` to false again.

Like I said, this is one of the first reports on this specific plugin, so it might be your sensor has a flaky connection which may cause the sensor to loose its set parameters.
But that is something that we should be able to handle to make the code more robust.

heinemannj, 2020-10-28:
Thanks for the task command fix.
I need to recompile for ESP32 dev again.

It would be helpful to add a dev env flag for the ESP32

in platformio.ini
`default_envs = dev_ESP32_4M316k`

and in platformio_esp32_envs.ini
```
[env:dev_ESP32_4M316k]
extends                   = esp32_common
platform                  = ${esp32_common.platform}
build_flags               = ${esp32_common.build_flags}  
                            -DFEATURE_ARDUINO_OTA
                            -DPLUGIN_SET_TEST_ESP32
                            -DPLUGIN_BUILD_DEV
board                     = esp32dev
extra_scripts             = ${esp32_common.extra_scripts}
```

It takes me a couple of painful hours to upload the ESP32 firmware in general (Step 1) and to understand and to implement the compilation procedure for ESP32 (step 2).

TD-er, 2020-10-28:
Why not use the "custom" build env then?
If you create a Custom.h file (template is available) you can define just how you need it in there and that file is excluded from the Git repo so you won't erase it on updates.

In all editors with PlatformIO support you can just select what env to build. So no need to set the default (which also affects the intellisense behavior)
For example in VS code:
[image]

The reason why I don't want to change the default env is that it does affect intellisense behavior.
At this moment intellisense already needs quite some memory (RAM isn't an issue on my system), which causes a crash in VScode of the extention host service. Intellisense does exceed the 4GB max RAM usage for the sandbox environment it is running in.
By setting the default env you essentially tell Intellisense to activate a lot more defines (USES_Pxxx) which makes more source code visible to intellisense -> more crashes.

TD-er, 2020-10-28:
I added a fix for your problem. (and also a speed-up, which should be noticable in the timing stats)

Can you test it or do you need me to make a test build?

heinemannj, 2020-10-28:
The test build I will manage by myself ;-)
Maybe I need support for understanding the proper usage of PlatformIO and GitHub ... but out of scope on sensor issue ;-)

How can I monitor timing stats?
I'm an old tired German guy ...

TD-er, 2020-10-28:
Timing stats are via the web interface -> Tools -> Timing stats.

All lines that have a value > 100 msec are highlighted as that's considered bad behavior (as it will block one or more calls to PLUGIN_TEN_PER_SECOND)
N.B. Each time you reload the page, the stats will be reset.

The current changes I made are in the P074 ino file:
```c++
    case PLUGIN_READ: {
      P074_data_struct *P074_data =
        static_cast<P074_data_struct *>(getPluginTaskData(event->TaskIndex));

      if (nullptr != P074_data) {

        // Fix to re-set the gain/timing before every read.
        // See https://github.com/letscontrolit/ESPEasy/issues/3347
        if (!P074_data->tsl.begin()) {
          break;
        }

        // Simple data read example. Just read the infrared, fullspecrtrum diode
        // or 'visible' (difference between the two) channels.
        float lux, full, visible, ir;
        {
          uint32_t fullLuminosity = P074_data->tsl.getFullLuminosity();

          // TSL2591_FULLSPECTRUM: Reads two byte value from channel 0 (visible + infrared)
          full = (fullLuminosity & 0xFFFF);

          // TSL2591_INFRARED: Reads two byte value from channel 1 (infrared)
          ir =  (fullLuminosity >> 16);

          // TSL2591_VISIBLE: Reads all and subtracts out just the visible!
          visible =  ( (fullLuminosity & 0xFFFF) - (fullLuminosity >> 16));

          lux     = P074_data->tsl.calculateLuxf(full, ir); // get LUX
        }

        UserVar[event->BaseVarIndex + 0] = lux;
        UserVar[event->BaseVarIndex + 1] = full;
        UserVar[event->BaseVarIndex + 2] = visible;
        UserVar[event->BaseVarIndex + 3] = ir;
```

So that will make testing easier if you have problems using Github to checkout a pull request.

By the way, I'm looking through the code of the used library for this plugin and it can be made so much simpler.
So expect one more commit later (not sure if finished today)
At least let me know if the sensor keeps working with current changes after running for some hours.

heinemannj, 2020-10-28:
I've copied and paste your changes into my file system.
Compilation was OK.
But uploading fails ...

tonhuisman, 2020-10-28:
> But uploading fails ...

On ESP32 you should be able to do an OTA update from the Tools page, unless you disabled OTA in your build, of course 🤔

heinemannj, 2020-10-28:
Compilation is failed - Sorry for German notes:

Funktion "operator new" kann nicht mit der angegebenen Argumentliste aufgerufen werden. -- Argumenttypen sind: (unsigned long, const std::nothrow_t)

TD-er, 2020-10-28:
Added a new commit for async reading.
I will make a test build.

heinemannj, 2020-10-28:
"Timing stats are via the web interface -> Tools -> Timing stats."

I can't see "Timing stats" on actual ESP8266 and ESP32 firmware

TD-er, 2020-10-28:
That's not included in "test" or "dev" builds, due to build size issues.

I have a test build with the latest commit (let's hope it will work, not tested)
- [ESPEasy_ESP32_mega-20201022-59-PR_3349.zip](https://www.dropbox.com/s/mzw8tynhxcwxhm8/ESPEasy_ESP32_mega-20201022-59-PR_3349.zip?dl=0)
- [ESPEasy_ESP82xx_mega-20201022-59-PR_3349.zip](https://www.dropbox.com/s/ivz0e1cnycnk22r/ESPEasy_ESP82xx_mega-20201022-59-PR_3349.zip?dl=0)

heinemannj, 2020-10-29:
**Your test build is working!**
In addition I have adapted a Custom.h file.

Can I also include the settings for my devices into the Custom.h?

Is there any possibility to include custom files into the build?
- like the rule1.txt, rule2.txt, rule3.txt, rule4.txt or
-  a custom dashboard like photometer.html

With my custom build "Timing stats" is available:

[image]

heinemannj, 2020-10-29:
The output I2C-Addresses scan is strange and not pointing to the sensor TSL2591 which I have connected:

[image]

TD-er, 2020-10-29:
This max timing of 1.4 seconds (!!) is with the latest code, or is it with the old code?

And no, you cannot include rules in the code, but for mass deployment it is possible to either make an image of the entire flash (thus flashing all including the file system)
Or you can use the "settings archive" option to clone them from another ESP unit, or build a script using `curl` or something similar for creating POST messages to upload the files.

N.B. if you clone or upload security.dat or config.dat files, make sure to reboot before changing anything in the settings or you may corrupt data.

TD-er, 2020-10-29:
> The output I2C-Addresses scan is strange and not pointing to the sensor TSL2591 which I have connected:
> 
> [image]

That's because the I2C scan uses a list of known I2C devices which was probably not updated when P074 was added.
My idea is to add a new function to scan I2C plugins present in the build to list their name if the found address is a valid one for that plugin. That way the list is always uptodate.

TD-er, 2020-10-29:
About your question for a dashboard.
That's possible.
See this (possibly a bit dated) description on the [wiki](https://www.letscontrolit.com/wiki/index.php?title=Mini_Dashboard)

heinemannj, 2020-10-29:
Much more better performance but from functional perspective unstable - no valid reproduceable light values

https://github.com/TD-er/ESPEasy/tree/bugfix/task_enable_disable

[image]

Build based on your zip file from yesterday evening:

[image]

I perform 2 measurements/samples:

a) Empty Sample
red on - wait 1 sec - measure and write into column I0 - red off
green on - wait 1 sec - measure and write into column I0 - green off
the same for blue
the same for infrared

b) Sample
the same procedure like above but store the values in column IT

c) Transmission = T = IT / I0
Under ideal conditions it should be exactly 1

The light value for the red LED should be ~1298 lux

[image]

TD-er, 2020-10-29:
I'm having a hard time interpreting your post.
The readings are too low?
Or are they only off for a single color?

How far are they off?
What is the used integration time?

What I did was I query the sensor every 100 msec to see if its status register shows it is ready.
But that interpretation may be wrong, so then the sensor may return a value at the first time it calls the PLUGIN_TEN_PER_SECOND call after a PLUGIN_READ. (which may be 0 msec, or 100 msec, or inbetween)

So what I can do, is set to wait for the set integration time (+ some margin) and then read the sensor.

heinemannj, 2020-10-29:
In quit state (all leds are off) the light value must be 0 lux
=> sensor and leds are completely isolated/boxed/dark room
=> time between the different measurement can be up to 15 mins => meaning 15 min inactivity

The reading for red should be ~1298 lux and NOT 0 or 1.0
=> meaning the sensor was not in the proper state - may be initialising was not completely finished

For green I expect ~2010 lux - reading seems to be also to low
=> meaning the sensor was not in the proper state or wrong integration time

For blue I expect ~ 997 lux
=> No gap between both samples - good

For blue I expect ~ 5 lux
=> No gap between both samples - good

I have played with the integration time - 400 msec has reduced the difference between both samples a little bit

[image]

heinemannj, 2020-10-29:
Only one led is ON at the same time.

Can I turn on a hidden debug button to monitor sensor reinitialisations?

TD-er, 2020-10-29:
The reading for red, was that the 1st reading and all others a later reading?
If the integration time hardly changes the outcome, only noise filtering, then the sensor reading is OK.
What I can do, is add a log entry showing the actual time between starting the measurement and reading the value.
This will be higher then the set value, but not lower.

I looked at the code to compute the lux value.
Right now I do use float values, which could prevent detection of an overflow situation, so that's a bug for sure.
Is it possible one of the channels is fully saturated when using the red LED?

heinemannj, 2020-10-29:
_The reading for red, was that the 1st reading and all others a later reading?_
- Yes

_Is it possible one of the channels is fully saturated when using the red LED?_
- No - As I said before normally sensor is in dark room
- I've played with integration time and gain values - fully saturated situation is impossible

_So what I can do, is set to wait for the set integration time (+ some margin) and then read the sensor._
- This really make sense

TD-er, 2020-10-29:
I have updated the code.
Can you test with the current code of this PR?
Or do you need me to make a new build.

It should log when there is a reading, what the duration is.
If it is at least the integration time, then we know it doesn't need to be changed to simply wait for the sensor, but looking at the status bit is a valid approach.

About not being saturated...
If the LED is pointing direct in the sensor and you set the integration time + gain at max, then I can imagine it could be saturated.

heinemannj, 2020-10-29:
You are meaning to update from:
https://github.com/TD-er/ESPEasy/tree/bugfix/task_enable_disable

Yes with the help of your patiance I have learned to make a proper build ;-)
with env:custom ...

heinemannj, 2020-10-29:
About not being saturated...
If the LED is pointing direct in the sensor and you set the integration time + gain at max, then I can imagine it could be saturated.

Beam angel is 180 degree => pointing directly into the sensor - you are right
But the sensor should be able to mesure 44000 lux => less than 3000 lux is penauts

Build is failed:

```
Building in release mode
Compiling .pio\build\custom_ESP32_4M316k\src\ESPEasy.ino.cpp.o
C:/Users/A410882/Downloads/ESPEasy-bugfix-task_enable_disable (1)/ESPEasy-bugfix-task_enable_disable/src/ESPEasy.ino:17:4: warning: #warning "**** Using Settings from Custom.h File ***" [-Wcpp]
   #warning "**** Using Settings from Custom.h File ***"
    ^
Linking .pio\build\custom_ESP32_4M316k\ESP_Easy_mega_20201029_custom_ESP32_4M316k.elf
Building .pio\build\custom_ESP32_4M316k\ESP_Easy_mega_20201029_custom_ESP32_4M316k.bin
Retrieving maximum program size .pio\build\custom_ESP32_4M316k\ESP_Easy_mega_20201029_custom_ESP32_4M316k.elf
Checking size .pio\build\custom_ESP32_4M316k\ESP_Easy_mega_20201029_custom_ESP32_4M316k.elf
esptool.py v2.6
esp32_create_factory_bin([".pio\build\custom_ESP32_4M316k\ESP_Easy_mega_20201029_custom_ESP32_4M316k.bin"], [".pio\build\custom_ESP32_4M316k\ESP_Easy_mega_20201029_custom_ESP32_4M316k.elf"])
Generating factory bin for genuine esp units
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  15.8% (used 51904 bytes from 327680 bytes)
Flash: [=======   *** [.pio\build\custom_ESP32_4M316k\ESP_Easy_mega_20201029_custom_ESP32_4M316k.bin] ValueError : too many values to unpack (expected 2)
]  71.5% (used 1359406Traceback (most recent call last):
 byte  File "C:\Users\A410882\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Action.py", line 1280, in execute
s f    result = self.execfunction(target=target, source=rsources, env=env)
rom  File "C:\Users\A410882\Downloads\ESPEasy-bugfix-task_enable_disable (1)\ESPEasy-bugfix-task_enable_disable\tools\pio\post_esp32.py", line 11, in esp32_create_factory_bin
 1    sect_adr,sect_file = section.split(" ")
900ValueError: too many values to unpack (expected 2)
544 bytes)
============================================================================================ [FAILED] Took 54.07 seconds ============================================================================================

Environment          Status    Duration
-------------------  --------  ------------
custom_ESP32_4M316k  FAILED    00:00:54.070
======================================================================================= 1 failed, 0 succeeded in 00:00:54.070 =======================================================================================
```

TD-er, 2020-10-29:
You have a space in the path.

I made a test build for you:
- [ESPEasy_ESP32_mega-20201022-59-PR_3349.zip](https://www.dropbox.com/s/mzw8tynhxcwxhm8/ESPEasy_ESP32_mega-20201022-59-PR_3349.zip?dl=0)
- [ESPEasy_ESP82xx_mega-20201022-59-PR_3349.zip](https://www.dropbox.com/s/ivz0e1cnycnk22r/ESPEasy_ESP82xx_mega-20201022-59-PR_3349.zip?dl=0)

TD-er, 2020-10-29:
> About not being saturated...
> If the LED is pointing direct in the sensor and you set the integration time + gain at max, then I can imagine it could be saturated.
>
> Beam angel is 180 degree => [...]

Max gain is 9876x the low gain.
And the lux is depending on both ch0 and ch1.
When considering the older version (TSL2561), you see the IR part is saturated quite early compared to its full range.

But please let me know if the new test build I made is now working.

heinemannj, 2020-10-30:
I have different results depending on which build I use:

1. https://github.com/TD-er/ESPEasy/tree/bugfix/task_enable_disable
Not working but with proper log entry with duration time
2. ESPEasy_ESP32_mega-20201022-59-PR_3349.zip as attached
Working but without duration time in log enty

heinemannj, 2020-10-30:
**1. https://github.com/TD-er/ESPEasy/tree/bugfix/task_enable_disable**

```
509074: TSL2591: Lux: 2 Full: 1 Visible: 1 IR: 0 duration: 297
509172: EVENT: TSL2591#Lux=2.00
509272: EVENT: TSL2591#Full=1.00
509372: EVENT: TSL2591#Visible=1.00
509472: EVENT: TSL2591#IR=0.00
516489: TSL2591: Lux: 0 Full: 0 Visible: 0 IR: 0 duration: 313
516572: EVENT: TSL2591#Lux=0.00
516672: EVENT: TSL2591#Full=0.00
516783: EVENT: TSL2591#Visible=0.00
516872: EVENT: TSL2591#IR=0.00
517311: WD   : Uptime 9 ConnectFailures 0 FreeMem 224056 WiFiStatus WL_CONNECTED ESPeasy internal wifi status: Conn. IP Init
522120: Command: TaskValueSet
522121: ACT  : TaskValueSet,3,3,0
522122: Command: TaskValueSet
522123: ACT  : TaskValueSet,3,4,0
522124: Command: TaskValueSet
522125: ACT  : TaskValueSet,4,1,0
522126: Command: TaskValueSet
522127: ACT  : TaskValueSet,4,2,0
522128: Command: TaskValueSet
522129: ACT  : TaskValueSet,4,3,0
522130: Command: TaskValueSet
522131: ACT  : TaskValueSet,4,4,0
522132: Command: TaskValueSet
522133: ACT  : GPIO,12,1
522134: Command: GPIO
522134: GPIO : port#12: set to 1
522135: ACT  : GPIO,13,0
522136: Command: GPIO
522136: GPIO : port#13: set to 0
522137: ACT  : GPIO,14,0
522137: Command: GPIO
522138: GPIO : port#14: set to 0
522139: ACT  : GPIO,23,0
522139: Command: GPIO
522139: GPIO : port#23: set to 0
522140: ACT  : timerSet,1,1
522141: Command: timerSet
522201: SW  : GPIO=12 State=1 Output value=1
522226: EVENT: Red#State=1.00
523141: EVENT: Rules#Timer=1,1
**523169: ACT  : TaskRun,1
523170: Command: TaskRun**
523196: ACT  : TaskValueSet,2,1,0.0
523196: Command: TaskValueSet
523198: ACT  : event
523198: Command: event
523199: EVENT: 
523284: ACT  : GPIO,12,0
523285: Command: GPIO
523285: GPIO : port#12: set to 0
523286: ACT  : GPIO,13,1
523287: Command: GPIO
523287: GPIO : port#13: set to 1
523288: ACT  : timerSet,2,1
523289: Command: timerSet
523352: SW  : GPIO=12 State=0 Output value=0
523377: SW  : GPIO=13 State=1 Output value=1
523391: EVENT: Red#State=0.00
523477: EVENT: Green#State=1.00
523590: TSL2591: Lux: 2508 Full: 1328 Visible: 1278 IR: 50 duration: 237
523672: EVENT: TSL2591#Lux=2508.00
523772: EVENT: TSL2591#Full=1328.00
523911: EVENT: TSL2591#Visible=1278.00
523998: EVENT: TSL2591#IR=50.00
524289: EVENT: Rules#Timer=2,1
524321: ACT  : TaskRun,1
524321: Command: TaskRun
524324: ACT  : TaskValueSet,2,2,2508.0
524325: Command: TaskValueSet
524327: ACT  : GPIO,13,0
524327: Command: GPIO
524327: GPIO : port#13: set to 0
524328: ACT  : GPIO,14,1
524329: Command: GPIO
524329: GPIO : port#14: set to 1
524330: ACT  : timerSet,3,1
524331: Command: timerSet
524672: SW  : GPIO=13 State=0 Output value=0
524697: SW  : GPIO=14 State=1 Output value=1
524711: EVENT: Green#State=0.00
524802: EVENT: Blue#State=1.00
524890: TSL2591: Lux: 1742 Full: 868 Visible: 861 IR: 7 duration: 218
524968: EVENT: TSL2591#Lux=1742.00
525068: EVENT: TSL2591#Full=868.00
525194: EVENT: TSL2591#Visible=861.00
525284: EVENT: TSL2591#IR=7.00
525369: EVENT: Rules#Timer=3,1
525405: ACT  : TaskRun,1
525405: Command: TaskRun
525408: ACT  : TaskValueSet,2,3,1742.0
525409: Command: TaskValueSet
525411: ACT  : GPIO,14,0
525411: Command: GPIO
525411: GPIO : port#14: set to 0
525412: ACT  : GPIO,23,1
525413: Command: GPIO
525413: GPIO : port#23: set to 1
525414: ACT  : timerSet,4,1
525415: Command: timerSet
525469: SW  : GPIO=14 State=0 Output value=0
525494: SW  : GPIO=23 State=1 Output value=1
525508: EVENT: Blue#State=0.00
525600: EVENT: IR#State=1.00
525692: TSL2591: Lux: 6 Full: 19 Visible: 8 IR: 11 duration: 224
525768: EVENT: TSL2591#Lux=6.00
525868: EVENT: TSL2591#Full=19.00
525968: EVENT: TSL2591#Visible=8.00
526068: EVENT: TSL2591#IR=11.00
526415: EVENT: Rules#Timer=4,1
526455: ACT  : TaskRun,1
526455: Command: TaskRun
526458: ACT  : TaskValueSet,2,4,6.0
526459: Command: TaskValueSet
526461: ACT  : GPIO,23,0
526461: Command: GPIO
526462: GPIO : port#23: set to 0
526463: ACT  : GPIO,25,0
526463: Command: GPIO
526463: GPIO : port#25: set to 0
526515: SW  : GPIO=25 State=0 Output value=0
526540: SW  : GPIO=23 State=0 Output value=0
526555: EVENT: EmptySample#State=0.00
526643: EVENT: IR#State=0.00
526733: TSL2591: Lux: 0 Full: 0 Visible: 0 IR: 0 duration: 217
526771: EVENT: TSL2591#Lux=0.00
526868: EVENT: TSL2591#Full=0.00
526968: EVENT: TSL2591#Visible=0.00
527103: EVENT: TSL2591#IR=0.00
```

heinemannj, 2020-10-30:
**2. ESPEasy_ESP32_mega-20201022-59-PR_3349.zip as attached**

```
46648: EVENT: Clock#Time=Fri,14:29
49931: TSL2591: Lux: 2.04 Full: 1.00 Visible: 1.00 IR: 0.00
50165: EVENT: TSL2591#Lux=2.04
50285: EVENT: TSL2591#Full=1.00
50374: EVENT: TSL2591#Visible=1.00
50464: EVENT: TSL2591#IR=0.00
56956: TSL2591: Lux: 2.04 Full: 1.00 Visible: 1.00 IR: 0.00
57521: EVENT: TSL2591#IR=0.00
59813: ACT  : TaskValueSet,4,3,0
59814: Command: TaskValueSet
59815: ACT  : TaskValueSet,4,4,0
59816: Command: TaskValueSet
59817: ACT  : GPIO,12,1
59818: Command: GPIO
59818: GPIO : port#12: set to 1
59819: ACT  : GPIO,13,0
59820: Command: GPIO
59820: GPIO : port#13: set to 0
59821: ACT  : GPIO,14,0
59822: Command: GPIO
59822: GPIO : port#14: set to 0
59823: ACT  : GPIO,23,0
59824: Command: GPIO
59824: GPIO : port#23: set to 0
59825: ACT  : timerSet,1,1
59826: Command: timerSet
59890: SW  : GPIO=12 State=1 Output value=1
59915: EVENT: Red#State=1.00
60271: Dummy: value 1: 0.00
60271: Dummy: value 2: 0.00
60271: Dummy: value 3: 0.00
60271: Dummy: value 4: 0.00
60300: Dummy: value 1: 0.00
60300: Dummy: value 2: 0.00
60300: Dummy: value 3: 0.00
60300: Dummy: value 4: 0.00
60321: EVENT: I0#Red=0.00
60485: Dummy: value 1: 0.00
60485: Dummy: value 2: 0.00
60485: Dummy: value 3: 0.00
60485: Dummy: value 4: 0.00
60489: EVENT: I0#Green=0.00
60581: EVENT: I0#Blue=0.00
60673: EVENT: I0#IR=0.00
60798: EVENT: IT#Red=0.00
60889: EVENT: IT#Green=0.00
60979: EVENT: Rules#Timer=1,1
**61009: ACT  : TaskRun,1
61009: Command: TaskRun
61758: TSL2591: Lux: 2366.40 Full: 1753.00 Visible: 1425.00 IR: 327.00**
62520: EVENT: T#IR=0.00
62610: EVENT: TSL2591#Lux=2366.40
62702: EVENT: TSL2591#Full=1753.00
62792: EVENT: TSL2591#Visible=1425.00
62886: EVENT: TSL2591#IR=327.00
63822: ACT  : TaskValueSet,2,2,2517.3
63823: Command: TaskValueSet
63825: ACT  : GPIO,13,0
63826: Command: GPIO
63826: GPIO : port#13: set to 0
63827: ACT  : GPIO,14,1
63828: Command: GPIO
63828: GPIO : port#14: set to 1
63832: ACT  : timerSet,3,1
63833: Command: timerSet
63892: SW  : GPIO=13 State=0 Output value=0
63917: SW  : GPIO=14 State=1 Output value=1
63932: EVENT: Red#State=0.00
64810: TSL2591: Lux: 1742.31 Full: 870.00 Visible: 863.00 IR: 8.00
65979: ACT  : TaskValueSet,2,3,1742.3
65980: Command: TaskValueSet
65983: ACT  : GPIO,14,0
65984: Command: GPIO
65985: GPIO : port#14: set to 0
66011: ACT  : GPIO,23,1
66012: Command: GPIO
66012: GPIO : port#23: set to 1
66014: ACT  : timerSet,4,1
66015: Command: timerSet
66072: SW  : GPIO=14 State=0 Output value=0
66097: SW  : GPIO=23 State=1 Output value=1
66112: EVENT: TSL2591#Lux=2517.26
66242: EVENT: TSL2591#Full=1334.00
66334: EVENT: TSL2591#Visible=1284.00
66426: EVENT: TSL2591#IR=51.00
67923: ACT  : TaskValueSet,2,4,5.6
67924: Command: TaskValueSet
67928: ACT  : GPIO,23,0
67929: Command: GPIO
67929: GPIO : port#23: set to 0
67956: ACT  : GPIO,25,0
67956: Command: GPIO
67957: GPIO : port#25: set to 0
68011: SW  : GPIO=25 State=0 Output value=0
68035: SW  : GPIO=23 State=0 Output value=0
68050: EVENT: TSL2591#Lux=1742.31
68174: WD   : Uptime 1 ConnectFailures 0 FreeMem 224176 WiFiStatus WL_CONNECTED ESPeasy internal wifi status: Conn. IP Init
68180: EVENT: TSL2591#Full=870.00
68272: EVENT: TSL2591#Visible=863.00
68364: EVENT: TSL2591#IR=8.00
68454: EVENT: Blue#State=0.00
70954: TSL2591: Lux: 2.04 Full: 1.00 Visible: 2.00 IR: 0.00
77973: EVENT: TSL2591#Lux=2.04
78092: EVENT: TSL2591#Full=1.00
78182: EVENT: TSL2591#Visible=1.00
78272: EVENT: TSL2591#IR=0.00
```

heinemannj, 2020-10-30:
My Custom.h:

[Custom.txt](https://github.com/letscontrolit/ESPEasy/files/5465406/Custom.txt)

TD-er, 2020-10-30:
OK, found what's causing the issue with the build I made...
Last time I changed the build script for the "nightly build", but had not reverted the changes.
So the build was not based on the PR, but on the latest code in the mega branch.... again (I really hate it when I make the same mistake several times...)

So consider your own build test as the true test of the source code, which apparently doesn't work yet :(

Just to be sure, I will start a new test build for the PR.

TD-er, 2020-10-30:
By the way, you had the integration time set to 200 msec?  (judging on the log entries)

TD-er, 2020-10-30:
Made a new test build:
- [ESPEasy_ESP32_mega-20201022-72-PR_3349.zip](https://www.dropbox.com/s/63ybqbzj7l3ebdh/ESPEasy_ESP32_mega-20201022-72-PR_3349.zip?dl=0)
- [ESPEasy_ESP82xx_mega-20201022-72-PR_3349.zip](https://www.dropbox.com/s/vccoy7f2r5dd3ld/ESPEasy_ESP82xx_mega-20201022-72-PR_3349.zip?dl=0)

And just to be sure about the correct files used for the build, a part of the `git log`:
```
commit 209160059fd76966466f272175a6e015c4c5d40c
Author: TD-er <gijs.noorlander@gmail.com>
Date:   Thu Oct 29 16:32:28 2020 +0100

    [TSL2591] Force disable sensor after timeout + log duration

commit 46ee0ffe191c2d978a7b59cd35a63a2db36dbd54
Author: TD-er <gijs.noorlander@gmail.com>
Date:   Wed Oct 28 17:19:54 2020 +0100

    [TSL2591] Perform async reading of sensor prevent blocking calls
```

heinemannj, 2020-10-30:
> By the way, you had the integration time set to 200 msec? (judging on the log entries)

Jepp you have pushed me into the right direction - after make a deep dive into the logs I've identified rare saturated situations

Below are my actual device settings:

[image]

TD-er, 2020-10-30:
Check!
Please let me know if the plugin is now working as expected/desired.

heinemannj, 2020-10-30:
Is not working

- After running the TaskRun,1 I expect directly the sensor output

[image]

TD-er, 2020-10-30:
And that is now (indeed) not working anymore.
Since I split it to effectively run the PLUGIN_READ to start the reading and schedule it again, it will run PLUGIN_READ twice and on the second run it will create events with the new values.

heinemannj, 2020-10-30:
With my actual rules  this change is breaking my setup ...

Is there any possibility to have nested events in rules?

- turn red LED on
- Wait for the proper sensor reading
- trun red LED off
- Wait for the proper sensor reading
- turn green LED on
- Wait for the proper sensor reading
- trun green LED off
- Wait for the proper sensor reading
- ...

TD-er, 2020-10-30:
A task does send an event.
Something like this:
```
on TSL2591#Lux do
  gpio,[var#1],0 // turn led off
  let,1,[var#1]+1  // Set next GPIO pin to var#1
  if [var#1]<3
    gpio,[var#1],1  // turn led on
    taskrun,1
  endif
endon
```

To initialize it, you can call something like this:
```
let,1,0
taskrun,1
```

TD-er, 2020-10-31:
I was thinking a bit more on this and I think you can even make a lookup table for the GPIO pins.
Just a simple example I made to show you what I mean.

The rules:
```
on init do
  let,1,123
  let,2,234
  let,3,345
  let,4,0
  asyncevent,loop
endon

on run do
  LogEntry,'Values [int#%eventvalue1%]'
endon

on loop do
  if [int#4]<4
    let,4,[int#4]+1
    asyncevent,run=[int#4]
    asyncevent,loop
  endif
endon
```

Calling `event,init` :
```
18102 : Info  : EVENT: init
18114 : Info  : ACT  : let,1,123
18120 : Info  : ACT  : let,2,234
18126 : Info  : ACT  : let,3,345
18131 : Info  : ACT  : let,4,0
18136 : Info  : ACT  : asyncevent,loop
18219 : Info  : EVENT: loop
18257 : Info  : ACT  : let,4,0+1
18265 : Info  : ACT  : asyncevent,run=1
18271 : Info  : ACT  : asyncevent,loop
18285 : Info  : EVENT: run=1
18306 : Info  : ACT  : LogEntry,'Values 123'
18310 : Info  : Values 123
18329 : Info  : EVENT: loop
18347 : Info  : ACT  : let,4,1+1
18354 : Info  : ACT  : asyncevent,run=2
18361 : Info  : ACT  : asyncevent,loop
18447 : Info  : EVENT: run=2
18462 : Info  : ACT  : LogEntry,'Values 234'
18467 : Info  : Values 234
18528 : Info  : EVENT: loop
18547 : Info  : ACT  : let,4,2+1
18554 : Info  : ACT  : asyncevent,run=3
18560 : Info  : ACT  : asyncevent,loop
18634 : Info  : EVENT: run=3
18650 : Info  : ACT  : LogEntry,'Values 345'
18654 : Info  : Values 345
18729 : Info  : EVENT: loop
18748 : Info  : ACT  : let,4,3+1
18755 : Info  : ACT  : asyncevent,run=4
18761 : Info  : ACT  : asyncevent,loop
18836 : Info  : EVENT: run=4
18852 : Info  : ACT  : LogEntry,'Values 4'
18856 : Info  : Values 4
18929 : Info  : EVENT: loop
```

As you can see, it runs quite fast, so the web log will miss a few lines.
Just to give you an impression how you can mix `%eventvalue%` with addressing a variable.
So variable 4 here is used as an index to iterate over the stored values of the other variables.

TD-er, 2020-10-31:
Can we conclude this is now working, or is the not returning a value immediately after calling `taskrun` a dealbreaker?
Not sure though how to fix that part though... 
Tasks running for several 100's of msec (and blocking) is hurting all other tasks and controllers running on the ESP.
Also there are other plugins using the same technique, like the BME280.

heinemannj, 2020-11-01:
> Can we conclude this is now working, or is the not returning a value immediately after calling taskrun a dealbreaker?

Yes - we can conclude your code is working - Thanks!

I have  up and running rules based on your first proposal
- An integrated average for the 2. until the 4. values - the 1. value I have skipped

But it is slow:
- ~10 sec for the 16 measurements in summary (4 LEDs and for each of them 4 measurements)

heinemannj, 2020-11-01:
```
On System#Boot do
  Let,11,12                    // GPIO of Red LED
  Let,12,13                    //         Green
  Let,13,14                    //         Blue
  Let,14,23                    //         IR

  Pulse,[VAR#11],1,1000
  Pulse,[VAR#12],1,1000
  Pulse,[VAR#13],1,1000
endon

on ResetReading#State=1 do
  TaskValueSet,2,1,0           // Reset Intensity of incident light
  TaskValueSet,2,2,0
  TaskValueSet,2,3,0
  TaskValueSet,2,4,0

  TaskValueSet,3,1,0           //       Intensity of transmitted light
  TaskValueSet,3,2,0
  TaskValueSet,3,3,0
  TaskValueSet,3,4,0

  GPIO,27,0
endon

on EmptySample#State=1 do
  TaskValueSet,2,1,0           // Reset Intensity of incident light
  TaskValueSet,2,2,0
  TaskValueSet,2,3,0
  TaskValueSet,2,4,0

  TaskValueSet,3,1,0           //       Intensity of transmitted light
  TaskValueSet,3,2,0
  TaskValueSet,3,3,0
  TaskValueSet,3,4,0

  GPIO,[VAR#11],0              // LEDs off
  GPIO,[VAR#12],0
  GPIO,[VAR#13],0
  GPIO,[VAR#14],0

  GPIO,[VAR#11],1              // Red on

  Let,1,0
  Let,2,0
  Let,3,1
  TaskRun,1                    // Start reading
endon

on Sample#State=1 do
  TaskValueSet,3,1,0           // Reset Intensity of transmitted light
  TaskValueSet,3,2,0
  TaskValueSet,3,3,0
  TaskValueSet,3,4,0

  GPIO,[VAR#11],0              // LEDs off
  GPIO,[VAR#12],0
  GPIO,[VAR#13],0
  GPIO,[VAR#14],0

  GPIO,[VAR#11],1              // Red on

  Let,1,0
  Let,2,0
  Let,3,1
  TaskRun,1                    // Start reading
endon

on TSL2591#Lux do
  if [VAR#3]>0
    Let,1,[VAR#1]+1                      // Loop counter
    if [VAR#1]=2 or [VAR#1]=3            // Skip 1. value
      Let,2,[VAR#2]+[TSL2591#Lux]
    elseif [VAR#1]=4
      Let,2,([VAR#2]+[TSL2591#Lux])/3    // Average for 2. 3. & 4. value
      if [EmptySample#State]=1
        TaskValueSet,2,[VAR#3],[VAR#2]   // Intensity of incident light
      elseif [Sample#State]=1
        TaskValueSet,3,[VAR#3],[VAR#2]   // Intensity of transmitted light
      endif
      if [VAR#3]=1                       // Red
        GPIO,[VAR#11],0
        GPIO,[VAR#12],1
        Let,3,2
      elseif [VAR#3]=2                   // Green
        GPIO,[VAR#12],0
        GPIO,[VAR#13],1
        Let,3,3
      elseif [VAR#3]=3                   // Blue
        GPIO,[VAR#13],0
        GPIO,[VAR#14],1
        Let,3,4
      elseif [VAR#3]=4                   // IR
        if [EmptySample#State]=1
          GPIO,25,0
        elseif [Sample#State]=1
          GPIO,26,0
        endif
        GPIO,[VAR#14],0
        Let,3,0                          // Exit
      endif
      Let,1,0                            // Reset loop counter
      Let,2,0                            //       average value
    endif
    TaskRun,1                            // Loop
  endif
endon
```

TD-er, 2020-11-01:
What is the integration time of the sensor?
And can you check the logs to see how often the measurements are actually performed?
If the integration time is 600 msec, then it simply takes 10 seconds to perform 16 measurements.
16 x 600 msec ~ 10 sec.

TD-er, 2020-11-01:
And just an idea... if you really need to perform lots of measurements, you can also use several sensors.
All I2C plugins have recently been updated to support multiple instances of the same plugin.
If you can't set another I2C address or have only a limited number of addresses, then you can use an I2C multiplexer.

heinemannj, 2020-11-01:
Integration time in GUI was: 200 msec
But if I switch the settings this has not a real impact on the duration

heinemannj, 2020-11-01:
```
1147180: WD   : Uptime 19 ConnectFailures 0 FreeMem 223912 WiFiStatus WL_CONNECTED ESPeasy internal wifi status: Conn. IP Init
1147288: TSL2591: Lux: 4 Full: 1 Visible: 1 IR: 0 duration: 199
1147386: EVENT: TSL2591#Lux=4.00
1147597: EVENT: TSL2591#Full=1.00
1147703: EVENT: TSL2591#Visible=1.00
1147786: EVENT: TSL2591#IR=0.00
1154588: TSL2591: Lux: 4 Full: 1 Visible: 1 IR: 0 duration: 199
1154685: EVENT: TSL2591#Lux=4.00
1154898: EVENT: TSL2591#Full=1.00
1154986: EVENT: TSL2591#Visible=1.00
1155085: EVENT: TSL2591#IR=0.00
1161449: ACT  : TaskValueSet,3,1,0
1161451: ACT  : TaskValueSet,3,2,0
1161453: ACT  : TaskValueSet,3,3,0
1161455: ACT  : TaskValueSet,3,4,0
1161457: ACT  : GPIO,12.00,0
1161458: GPIO : port#12: set to 0
1161459: ACT  : GPIO,13.00,0
1161460: GPIO : port#13: set to 0
1161461: ACT  : GPIO,14.00,0
1161462: GPIO : port#14: set to 0
1161464: ACT  : GPIO,23.00,0
1161465: GPIO : port#23: set to 0
1161466: ACT  : GPIO,12.00,1
1161467: GPIO : port#12: set to 1
1161468: ACT  : Let,1,0
1161470: ACT  : Let,2,0
1161471: ACT  : Let,3,1
1161473: ACT  : TaskRun,1
1161534: SW  : GPIO=12 State=1 Output value=1
1161550: EVENT: Red#State=1.00
1161688: TSL2591: Lux: 17238 Full: 1 Visible: 65471 IR: 66 duration: 154
1161786: EVENT: TSL2591#Lux=17238.00
1161834: ACT  : Let,1,0.00+1
1161984: ACT  : TaskRun,1
1162004: EVENT: TSL2591#Full=1.00
1162086: EVENT: TSL2591#Visible=65471.00
1162188: TSL2591: Lux: 17238 Full: 1 Visible: 65471 IR: 66 duration: 183
1162189: EVENT: TSL2591#IR=66.00
1162286: EVENT: TSL2591#Lux=17238.00
1162392: ACT  : Let,1,1.00+1
1162397: ACT  : Let,2,0.00+17238.0
1162541: ACT  : TaskRun,1
1162562: EVENT: TSL2591#Full=1.00
1162636: EVENT: TSL2591#Visible=65471.00
1162714: TSL2591: Lux: 17238 Full: 1 Visible: 65471 IR: 66 duration: 152
1162715: EVENT: TSL2591#IR=66.00
1162791: EVENT: TSL2591#Lux=17238.00
1162841: ACT  : Let,1,2.00+1
1162846: ACT  : Let,2,17238.00+17238.0
1163027: ACT  : TaskRun,1
1163047: EVENT: TSL2591#Full=1.00
1163124: EVENT: TSL2591#Visible=65471.00
1163200: TSL2591: Lux: 17238 Full: 1 Visible: 65471 IR: 66 duration: 152
1163201: EVENT: TSL2591#IR=66.00
1163286: EVENT: TSL2591#Lux=17238.00
1163338: ACT  : Let,1,3.00+1
1163347: ACT  : Let,2,(34476.00+17238.0)/3
1163384: ACT  : TaskValueSet,2,1.00,17238.00
1163419: ACT  : GPIO,12.00,0
1163420: GPIO : port#12: set to 0
1163422: ACT  : GPIO,13.00,1
1163423: GPIO : port#13: set to 1
1163424: ACT  : Let,3,2
1163495: ACT  : Let,1,0
1163497: ACT  : Let,2,0
1163499: ACT  : TaskRun,1
1163520: SW  : GPIO=12 State=0 Output value=0
1163535: SW  : GPIO=13 State=1 Output value=1
[... 115 lines cut ...]
1170159: EVENT: TSL2591#Lux=4.00
1170209: ACT  : Let,1,3.00+1
1170217: ACT  : Let,2,(4.00+4.0)/3
1170253: ACT  : TaskValueSet,2,4.00,2.67
1170372: ACT  : GPIO,25,0
1170373: GPIO : port#25: set to 0
1170406: ACT  : GPIO,23.00,0
1170407: GPIO : port#23: set to 0
1170408: ACT  : Let,3,0
1170410: ACT  : Let,1,0
1170412: ACT  : Let,2,0
1170414: ACT  : TaskRun,1
1170434: SW  : GPIO=25 State=0 Output value=0
1170450: SW  : GPIO=23 State=0 Output value=0
1170464: EVENT: TSL2591#Full=1.00
1170537: EVENT: TSL2591#Visible=65535.00
1170612: TSL2591: Lux: 4 Full: 1 Visible: 1 IR: 0 duration: 177
1170612: EVENT: TSL2591#IR=2.00
1170702: EVENT: EmptySample#State=0.00
1170773: EVENT: IR#State=0.00
1170858: EVENT: TSL2591#Lux=4.00
1171072: EVENT: TSL2591#Full=1.00
1171158: EVENT: TSL2591#Visible=1.00
1171258: EVENT: TSL2591#IR=0.00
```

heinemannj, 2020-11-01:
Above log was produced with 100ms integration time

```
1161550: EVENT: Red#State=1.00
1161688: TSL2591: Lux: 17238 Full: 1 Visible: 65471 IR: 66 duration: 154
```

heinemannj, 2020-11-01:
Another strange thing:
The measured light intensity is dependant on the build I'm using

Both builds based on the latest zip file you have provided last Friday
a) the test firmware from the bin folder (build by you)
b) a custom firmware compiled by myself

TD-er, 2020-11-01:
Well if you see different behavior between builds, then you probably are using different code.

I will merge it so you can check some reproducible/verifiable code.
I will also have a look at your rules this evening as it seems a bit complex.

PR #3349 (2020-11-01): [TSL2591] Check sensor before reading + speedup
The old implementation did perform 3 readings, which depending on the set integration time could take a while.
But since all output values should be based on the same reading anyway, it is simply wrong to perform separate readings.

Also added a call to `begin()` of the library to make sure the gain and timing are correctly set.
This could cause incorrect readings if the sensor somehow got reset.

Fixes: #3347
commit cfe7d2f2 (2020-11-01): [TSL2591] Check sensor before reading + speedup

The old implementation did perform 3 readings, which depending on the set integration time could take a while.
But since all output values should be based on the same reading anyway, it is simply wrong to perform separate readings.

Also added a call to `begin()` of the library to make sure the gain and timing are correctly set.
This could cause incorrect readings if the sensor somehow got reset.

Fixes: #3347
closed 2020-11-01 by TD-er