letscontrolit/ESPEasy #3350: Plugins: _P168_ThermOLED _P159_Pushbutton,  the display does not show all the symbol.

2007-i, 2020-10-29:
I compiled (Arduino ide) the firmware with your plugin and the new version of ESPEasy_ESP82xx_mega-20201009, the compilation was successful except for the points described here:
[viewtopic.php?f=4&t=7607&p=47529#p47521](https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=7607&p=47529#p47521)

After the corrections indicated by the respected TD-er and Ath, I managed to compile the firmware with plugins:
_P168_ThermOLED
_P159_Pushbutton
but the display does not show all the symbols, there is no full time, signal level and something else.
Unit Name: Thermo
[image]

I have used two plugins:
[ESPEasy+P168_ThermOLED+_P159_Pushbutton.zip](https://github.com/letscontrolit/ESPEasy/files/5459594/ESPEasy%2BP168_ThermOLED%2B_P159_Pushbutton.zip)

as well as the latest version of [ESPEasy_ESP82xx_mega-20201009 and ESPEasy_ESP82xx_mega-20201022](https://github.com/letscontrolit/ESPEasy)

With the old version ESPEasy_Thermo_4M_c241
[ESPEasy_Thermo_4M_c241.zip](https://github.com/letscontrolit/ESPEasy/files/5459645/ESPEasy_Thermo_4M_c241.zip)
, the display had all the symbols.
[image]

tonhuisman, 2020-10-30:
> I also noticed in the latest version of ESPEasy_ESP82xx_mega-20201022 RTTTL (Ring Tone [Text] Transfer Language) did not work

You should create a separate issue for that, it doesn't seem to be related in any way to the issue reported here.
It is possible that this functionality is excluded under specific conditions, related to the flash size, though.

TD-er, 2020-10-30:
... Or related to command handling, which also appeared to be an issue for other commands in the recent build.

2007-i, 2020-10-30:
And on the problem with the display of not all characters on the display, what else can I do?
From me that it is necessary to more?
Maybe something to test?

enesbcs, 2020-11-22:
> And on the problem with the display of not all characters on the display, what else can I do?
You can revert to an older ESPEasy build.

P168 plugin was never ment to be compiled by newer ESPEasy builds.
- I guess 03%3a in the top left is "%systime%" in the code, which is changed in the ESPEasy variables,  %c2% may be the "{D}C" in the code, which also changed. Or the parseTemplate() system function changed...
- Signal bars are displayed by the P168_display_wifibars() function, according to WiFi.RSSI() function result. i guess it is also reworked totally.

tonhuisman, 2022-05-07:
ThermOLED is moved to the mega branch as plugin P109, so this can probably be closed

closed 2022-05-07 by TD-er