letscontrolit/ESPEasy #3351: serialsend is unknown command

TommoT1, 2020-10-30:
In release mega-20201022 serialsend is an unknown command.
Till mega-20201016 it worked....
And I have another problem with serialsend, I have to send HEX Code including 0x00 out of the rules and are not able to manage it. When I do it via the browser (http://ipadress/?cmd=serialsend%FE%FF) it works, except %00 but out of rules is always shows up as ASCII....

Regards Tom

TD-er, 2020-10-30:
Where do you send this command from?

TommoT1, 2020-10-30:
I send it via rules e.g. 

on reset do
  serialsend  cls reset
endon
and 
or via the Browser e.g. http://ipadress/control?cmd=serialsend cls reset

in mega-20201022 I get an "unknown command" message
in mega-20201016 it still worked

TD-er, 2020-10-30:
Do you have the Ser2Net plugin active as a task?
That's the plugin handling the `serialsend` command.

TommoT1, 2020-10-30:
No

TD-er, 2020-10-30:
Please enable a task for that plugin.
In older builds we still had a work-around active that would allow you to send commands handled by not enabled plugins/tasks.
That work around was removed, as it was causing all kind of other issues.

TommoT1, 2020-10-30:
Ok, I disabled  the Serial Port in the Tools -> Advanced to be able to use GPIO 1 for the Serial Server and then activated the task

[image]

When I then put "http://ipadress/control?cmd=serialsend,Hallo"
the Browser says Command unknown: "serialsend,Hallo"

thomastech, 2020-10-30:
Some test results:

1. I installed ESP_Easy_mega_20201022_test_ESP8266_4M1M_VCC on a Wemos. 
2. I created a Communication - Serial Server task. 
3. When I send the serialsend command using _Tools->command_ the response says "OK"
4. When I send it via browser the response says "_SyntaxError: JSON.parse: unexpected character at line 1 column 1 of the JSON data_"
5. I also tried sending a Nextion display command by browser. It gets the _SyntaxError: JSON.parse error mentioned above. So it seems that the html commands are broken.

Using self compiled mega_20201012_test_ESP8266_4M1M_VCC all tests get the friendly "OK" response.

The second problem:
> I send it via rules e.g.
```
serialsend cls reset
```

I suspect the text spaces between the args are confusing the parser. Instead, I'd try something like this:
```
serialsend,"cls reset"
or
serialsend,'cls reset'
```
For details to using quotes with rule commands, see 
https://github.com/letscontrolit/ESPEasy/issues/2724

- Thomas

tonhuisman, 2020-10-30:
> and now also talking to a DFPlayer mini. I guess this mp3 Player was supported earlier by espeasy but now I can't find it in the binaries for 1M of memory anymore.

The DFPlayer is only included in the TESTING (test) builds, and sadly that build no longer fits in the 1 MB devices. You can make a custom build using [PlatformIO](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html) (recommended) or [Arduino IDE](https://espeasy.readthedocs.io/en/latest/Participate/ArduinoIDE.html), instructions are in `Custom-sample.h` in the `src` directory.

TD-er, 2020-10-30:
> * When I send it via browser the response says "_SyntaxError: JSON.parse: unexpected character at line 1 column 1 of the JSON data_"
> * I also tried sending a Nextion display command by browser. It gets the _SyntaxError: JSON.parse error mentioned above. So it seems that the html commands are broken.

Which browser?
I don't recognize it as an ESPEasy error.
Not sure why it should be processed by ArduinoJSON, or something else that may parse JSON, so right now I don't have a clue where that error may come from.

Can you give the exact URL to test here?

thomastech, 2020-10-30:
Browser command is 
`http://192.168.1.5/control?cmd=serialsend,Hallo`

Firefox reports the "SyntaxError: JSON.parse" error.
Chrome and Edge reports the "Command unknown" error.

- Thomas

TommoT1, 2020-10-30:
Hi all together!
I think I mixed two problems together.....
First of all I had no problems sending serial ASCII data till the mega-20201022 build with the syntax I posted already...

And now with Halloween tomorrow I got the need for sending HEX data to this mp3 player :-)

TD-er, 2020-10-30:
Just name the plugins you need and I will build a version for you (can even be based on an older build if you need to get it running in the next 24h, so we can have a look at the issues later)

thomastech, 2020-10-30:
I found the reason to the firefox error. Here's a screenshot:

[image]
The output defaulted to JSON. But when I select "Raw Data" the error reverts to the "Command unknown" So the error is consistent across the three browsers.

> And now with Halloween tomorrow I got the need for sending HEX data to this mp3 player :-)
If you intended to use raw hex data to control the DFPlayer then I suggest using the MP3 Player plugin instead. Otherwise you'll probably get ghoulish tricks instead of tasty treats.

- Thomas

TommoT1, 2020-10-30:
:-)
Very kind TD-er! Thank you very much!
For tomorrow I solved it.....the ESP talks via serial ASCII to an Arduino and when it sends e.g. %00 the Arduino sends on his serial port Hex 00 to the DFPlayer... so the mp3 part works.
So all in all there is time for the issues.

TD-er, 2020-10-30:
> Maybe you'll find someday some time to invent something like serialsendhex....

Like [this](https://espeasy.readthedocs.io/en/latest/Reference/SystemVariable.html?highlight=hex#standard-conversions)?

TommoT1, 2020-10-30:
Yes, I now those conversions in ESPeasy....but how to tell ESPeasy to send for example 0xFE 0x00 0xFF via the serial port out of the rules. Thats the thing I don' t manage to do...

thomastech, 2020-10-30:
For reference, an example of this is found with Tasmota's SerialSend4 (binary), SerialSend5 (Hex), and SerialSend6 (decimal) commands.

But ESPEasy is intended to make things easy; So you don't have to work so hard. If your intention is to send raw data bytes to control the DFPlayer then why not use the plugin? It handles all the messy details related to playing the audio files. No need to care about raw data bytes. Less tricks, more treats.

- Thomas

TommoT1, 2020-11-02:
Hi,
with todays release I still get the "unknown command" message when using serialsend via browser but out of the rules it works again (with enabled Ser2Net plugin active as a task)

Tom

TD-er, 2020-11-02:
Where do you enter the command?
Just as an URL, or via the tools->command field?
Can you give the exact URL in the browser?

TommoT1, 2020-11-16:
Sorry, there was no time the last days.....
When I use the tools->command field it works an I get an "OK" as command output,
also of course http://xxx.xxx.xxx.xxx/tools?cmd=serialsend,Hallo
But when I use http://xxx.xxx.xxx.xxx/control?cmd=serialsend,Hallo    it works (the serialsend is sent) but I get the error 
message: Command unknown: "serialsend,Hallo"
before the latest Releases also on the URL http://xxx.xxx.xxx.xxx/control?cmd=ser.... the message was just "OK"

Hope I made it clearly...

TD-er, 2020-11-16:
@TommoT1 
Thanks that does help me a lot.
I notice there is something strange here for this workflow to send commands.

TD-er, 2020-11-16:
I made an issue to fix this: https://github.com/letscontrolit/ESPEasy/issues/3374

TommoT1, 2021-12-07:
Hi there,
I've to bring bring up this post again.....
Back then I manged my serial communication issues with your help and the Serial Server plugin.
Now I found out that from release 20210223 to release 20210503 something was changed in that plugin. From then on you have to specify the serial port to be used.
Unfortunately there is no selectable preset for the "raw" esp8266 TXD = GPIO1  and RXD = GPIO3.
Any selectable preset doesn't allow communication via those pins.

[image]

'Till that update I used to send serial commands via the rules and the serialsend command e.g. 

on reset do
  serialsend  cls reset
endon
How do I get this functionality back or which command do I have to use instead?

Regard
Thomas

uzi18, 2021-12-07:
You can try to use SoftwareSerial.

TommoT1, 2021-12-07:
I tried already---no help.
Do you know if it is intentional to block serial com via the (chipdesign) designated ports ? Till 20210223 it worked flawless for me.

TD-er, 2021-12-07:
Is "Enable Serial" checked on the tools->advanded page?
If so, then it must be unchecked.

TommoT1, 2021-12-08:
I unchecked "Enable Serial port:" but still no help.
Regardless which Serial Port config I select on  the Task Settings (the ones on the picture above) the log output shows:
"Command unknown: serialsend ....."

tonhuisman, 2021-12-08:
That command is handled by the "Serial Server" plugin, but will only work if the plugin is enabled. Do you have it enabled?

TommoT1, 2021-12-08:
6 posts above I explained my history ... Yes "serial server" is enabled but after 20210223 some settings for this plugin changed and from then on serialsend again is an unknown command

tonhuisman, 2021-12-08:
According to the code history, this command has been restored ~10 months ago, but I don't know when exactly that change was merged into the release branch, so it may have been left out of that build from last May you are using.
What if you test with the latest release? there it _should_ be included.

TD-er, 2021-12-08:
Just writing down some searched commits I already looked at:

In the release notes of `mega-20210503` it said the following:
- [P020 Ser2Net] Restore serialsend command & add ser2netclientsend (commit: https://github.com/letscontrolit/ESPEasy/commit/76c60d138e55435188e1c3bcd96b41e441936989#diff-b55a8bbcde00b6c4c12f44a2353b9152cd46867edcce8bb3d61999005fb29b3e )

Possible related pull request: https://github.com/letscontrolit/ESPEasy/pull/3483
And the last commit before 20210503: https://github.com/letscontrolit/ESPEasy/commit/b4253d6922d66bda40567affa028b8c3fe479d7a#diff-b55a8bbcde00b6c4c12f44a2353b9152cd46867edcce8bb3d61999005fb29b3e

I think I don't really understand what the true problem is here.

So let's make sure we get the really issue clear.
- when trying to run the command `serialsend` you get the reply `unknown command`
- Enable Serial is unchecked on tools->Advanced page
- `esp8266 TXD = GPIO1 and RXD = GPIO3` is "HW Serial0" as in the combo box screenshot you showed, so I don't understand what the problem is in selecting this port
- The changes since Feb 2021 require you to have the task with this plugin to be enabled. If not enabled and initialized, the command will not be executed.
- If server port is 0 or baudrate is 0, the task will not be considered as initialized and thus not execute a command.

Can you please check where you might miss something in this task config?
And perhaps show a screenshot of the task settings?

TommoT1, 2021-12-08:
Hi TD-er!
With your help I found the two steps wich made serialsend not an unknown command anymore...
Disable Serial in Advanced and give the SerialServer a port number !
I think I missinterpreted the combo box...its from the other device point of view (connect TX to GPIO3 (because its the esp's RXD))
Now my ESP is serialsending again BUT not clear ASCII anymore..... where else is my mistake or missunderstanding ?
I'm trying to send just plain text like "serialsend  cls reset" 
After your help in November 2020 it worked till I updated the firmware lately.
 Here is a picture of the SerialServer page

[image]

TD-er, 2021-12-08:
Hmm that's strange as it doesn't do much with the text you try to add.
It just sends whatever is past the 11th character of the command string.  (not the most elegant way of doing it, but that should be an indication that part of the code has not changed in a long time)

If you connect to it using a terminal program like Putty (or whatever terminal program you like) with the correct baud rate, you can see the text you're trying to send via your serialsend command.

How do you give the command? From the rules or the command text field on the tools page?

TommoT1, 2021-12-08:
no, with actual builds I'm not able to see the text, it just produces some unreadable ASCII code
I give the commands via the rules....
[image]

this is the minicom screen from the timer controlled " Hallo echo 123"

[image]

TD-er, 2021-12-08:
What baudrate did you set your terminal to ? (minicom)

TommoT1, 2021-12-08:
also 9600, like the serial server

TD-er, 2021-12-08:
Can you test with both set to 115200?
Just to make sure the command may work. I got the feeling the baud rate may be set to another baudrate somehow.

Another check can be to disable the task and enable it again (can also be done in the rules) without reboot.
With baudrate still set to 9600, it might be working then if the baud rate is re-set somewhere else (what I suspect here)

TommoT1, 2021-12-08:
No help with 115200, just slightly different characters.
With 115200 it starts to show some part of the boot message and then stops.
In rules I disable the task upon boot, wait 2 secs, enable it again and then I start to serialsend....

TommoT1, 2021-12-08:
[image]

TommoT1, 2022-01-22:
Has anybody a hint for me? Still with firmware 20211224 no change. Has anybody a working config? I just need some clear ascii text to be send via the serial port.....

TommoT1, 2022-01-22:
This is my configuration: 

[image]

[image]

And the rule which shall produce some serial output:

[image]

The settings of minicom:
[image]

and the unreadable output:

[image]

Until the version 20210223 the setup worked as expected and then something changed and now my serial devices are not controllable anymore via espeasy :-(

ghtester, 2022-01-23:
I can confirm the Serial Server plugin does not work for me as well. Tried with the custom firmware, compiled using Vagrant yesterday from latest sources. Tried to play with plugin settings but couldn't find a working setup so far. 
I am only able to receive one nonsense character when I send a long string through the serialsend command.

TD-er, 2022-01-23:
I do know that there was a problem with the Serial Server plugin.
A long time ago (> 2 years) I transformed all plugins using serial to use the new serial wrapper so that all plugins could either use SW serial, HW serial and later the I2C to UART bridge from NXP.
However there were 2 plugins that appears not to work well with this change, so I reverted it for those 2.
Later someone (was it @thomastech  ?) picked up where I left off and fixed what I missed and that got merged.

Could be that later change was roughly a year ago.
Have to check the Git Blame to be sure.

thomastech, 2022-01-23:
> Later someone (was it @thomastech ?) picked up where I left off and fixed what I missed and that got merged.
It wasn't me. Too bad, since if I had messed this up then maybe I could help solve the problem. But at the moment I don't have any suggestions.

- Thomas

TommoT1, 2022-01-23:
@ghtester  Thanks for your confirmation, for now I stay on those nodes on Version 20210223

@TD-er  If it helps your investigation.. I think it occured from Release 20210223 to Release 20210503 so your pretty exact with one year :-)

ghtester, 2022-08-03:
FYI in current firmware releases the Serial Server seems to be working fine and I believe this issue can be closed.

TommoT1, 2022-08-03:
Yes, you're right, forgot to close it....

closed 2022-08-03 by TommoT1