letscontrolit/ESPEasy #3352: Oled 1306 row of pixels

PrutzerNL, 2020-10-30:
Hi,

I have a D1 mini with ESP_Easy_mega_20200929_normal_ESP8266_4M1M_VCC Build 20111 - Mega with a 1.3 inch Oled connected.

On the right side of the screen there is a row of pixels. I am wondering if the screen is a doa, or did I something wrong?
[image]

Device settings:
Display - OLED SSD1306
0X3C (60) - Default
Display size 128x64

I also tried device: Display - OLED SSD1306/SH1106 Framed.
Then the whole display is a pixel madness.

Am I doing something wrong, or is this display dead on arrival?

TD-er, 2020-10-30:
Are you sure this is an SSD1306 display with 128x64 pixels?
It does look like a SH1106 with 130x64 pixels, which is only (fully) supported in the Framed OLED plugin.

PrutzerNL, 2020-10-30:
Thanks! Selecting Display - OLED SSD1306/SH1106 Framed and Controller SH1106 works great! Thanks!

TD-er, 2020-10-30:
You're welcome :)

closed 2020-10-30 by TD-er