letscontrolit/ESPEasy #3353: Control of gpio via mqtt not working anymore on 20201022

ebogaard, 2020-10-30:
### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

After upgrading to ESP_Easy_mega_20201022_normal_ESP8266_4M1M from ESP_Easy_mega_20201016_normal_ESP8266_4M1M, mqtt control of gpio switches doesn't seem to work anymore.
When I publish a value of 1 or 0 to the mqtt topic, nothing happens.

### Expected behavior

Up to and including 20201016, I could switch gpio using mqtt messages publishing a value of 0 or 1 to the topic.

### Actual behavior

After upgrading, nothing seems to be happening: I don't see the state of the device changing.
Using another mqtt client, I checked that the mqtt message is published to the correct topic.

### Steps to reproduce

1. Enable mqtt controller and let the settings default
2. Enable switch @ gpio 14 and enable the mqtt controller
3. Publish value of 1 to mqtt topic "%sysname%//gpio/14"
4. The switch should turn on

### System configuration

Hardware: Wemos D1 mini

ESP Easy version: ESP_Easy_mega_20201022_normal_ESP8266_4M1M

ESP Easy settings/screenshots:

TD-er, 2020-10-30:
It is probably already fixed: https://github.com/letscontrolit/ESPEasy/pull/3346

For a different PR, I made a test build, which has this merged PR also included.
See: https://github.com/letscontrolit/ESPEasy/issues/3347#issuecomment-719592855
Can you test this test build to see if this does fix your issue too?

ebogaard, 2020-10-30:
Yes, that build fixes it!

Missed the PR and comment about this in the other issue.

TD-er, 2020-10-30:
Great 👍 
Please also test the WiFi a bit, as that's also a (big) part of the changes pending for the next nightly build.

ebogaard, 2020-10-30:
I'll keep an eye on stability of the WiFi connection.
So far no problems :-)

I think you can close this issue, though.

closed 2020-10-30 by TD-er