letscontrolit/ESPEasy #3355: SGP30 not detected as supported device

immertroll, 2020-11-02:
I try to build a eCO2 sensor with above sensor but the device is not detected.

Affected sensor: GY-SGP30 (https://www.ebay.de/itm/GY-SGP30-TVOC-eCO2-Kohlendioxid-Formaldehyd-Detektor-Luftqualitat-Sensormodul/164006978002)
Used board/hardware: Lolin NodeMCU v3
Used plugin: https://espeasy.readthedocs.io/en/latest/Plugin/P083.html

behavior: The sensor is detected by the i2cScanner at address 0x58. The column "Supported devices" says that there is no supported device. The plugin produces the message "SGP30: Sensor not found" at the log.

expected behavior: SGP30 is shown as supported.

sysinfo:

> Build:⋄
> 20111  - Mega
> System Libraries:⋄
> ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
> Git Build:⋄
> 
> Plugin Count:⋄
> 82 [Normal] [Testing]
> Build Time:⋄
> Nov  2 2020 13:19:48
> Binary Filename:⋄
> ESP_Easy_mega_20201102_test_ESP8266_4M1M_VCC
> Build Platform:⋄
> Linux-4.19.104-microsoft-standard-x86_64-with-glibc2.29
> Git HEAD:⋄
> mega-20201102_ca0be3f

TD-er, 2020-11-03:
Has the sensor worked on a previous build before?

thomastech, 2020-11-03:
Also try flashing with mega-20200328:
https://github.com/letscontrolit/ESPEasy/releases/tag/mega-20200328

This release seems to be one that was tested by TD-er on his SGP30, so I assume it works. Perhaps success with it can be used to identify the problem is with the more recent releases.

- Thomas

immertroll, 2020-11-04:
Thank you for your hints. I reduced the clock frequeny to 50k and it starts working.
The strange thing is, that it works also at 100k now. :see_no_evil: 

The i2cScanner detects the device as not supported at all but under devices it shows some values.
Thank you

thomastech, 2020-11-04:
> The i2cScanner detects the device as not supported at all

There's a i2c scanner function called getKnownI2Cdevice() that reports the names of the various I2C devices. I peeked at the code and saw that the "SGP30" device name is present. So it is odd you're not seeing it in the scanner listing.

Now that the SGP30 is working, try running the I2C scanner again. It should report _Addresses in Use_ is 0x58 and the _Supported devices_ column should list the SGP30. What do you see?

And what ESPEasy release version are you using?

- Thomas

TD-er, 2020-11-04:
If you need to run it at such a low speed, can you give more information on how it is connected?
How long are the cables?
Do you have pull-up resistors connected to the I2C pins and 3v3? (4k7 to clock from 3v3 and 4k7 to  data from 3v3)

Is it the only I2C device connected?

tonhuisman, 2020-11-08:
> There's a i2c scanner function called getKnownI2Cdevice() that reports the names of the various I2C devices. I peeked at the code and saw that the "SGP30" device name is present. So it is odd you're not seeing it in the scanner listing.

That code is wrapped by ``#ifndef LIMIT_BUILD_SIZE``, and that define is set for the TESTING build that is in use here, so, the addresses are shown, but no device names, unfortunately.

thomastech, 2020-11-09:
> That code is wrapped by #ifndef LIMIT_BUILD_SIZE, and that define is set for the TESTING build that is in use here, so, the addresses are shown, but no device names, unfortunately.

Good eye, I missed that. To reduce user confusion, I'd suggest that the LIMIT_BUILD_SIZE should also be used to suppress the _Supported Devices_ name column shown in the web interface. That is to say, suppress the column when the names are missing due to the LIMIT_BUILD_SIZE define.

The answers to TD-er's queries about pullup resistors and cable length may help explain what is going on in the user's installation. Software tends to be blamed, but sometimes the problems with wiggly bits is caused by the hardware.

- Thomas

TD-er, 2020-11-09:
Oh and about my suggestion to lower the I2C clock speed.
I found a bug (and fixed it, but that's not yet included in the nightly builds) which caused a change in I2C clock frequency to be only applied after a reboot.
So for now, please perform a reboot when you change the I2C clock frequency, or add a (dummy) task with just any I2C plugin with the "Low Speed" checkbox set as a work-around.

Otherwise testing with different speeds may not show any differences if it is indeed related to the I2C clock frequency.

tonhuisman, 2023-09-01:
I2C Scanner is now run at low I2C speed, so most sensors should be detected if low I2C speed hasn't been set to an unreasonable/unusable value.
Issue can be closed.

closed 2023-09-01 by TD-er