letscontrolit/ESPEasy #3356: [ESP32][Mini Dashboard] Not working

heinemannj, 2020-11-03:
I try to open the Mini Dashboard esp page on an ESPEasy ESP32 build.

```
http://192.168.178.126/dashboard.esp

URI: /dashboard.esp
Method: GET
Arguments: 0
```

```
Build:⋄ | 20111 - Mega32
System Libraries:⋄ | ESP32 SDK v3.2.3-14-gd3e562907
Git Build:⋄ |  
Plugin Count:⋄ | 81 [Normal] [TEST ESP32]
Build Origin: | Travis
Build Time:⋄ | Nov 2 2020 13:17:49
Binary Filename:⋄ | ESP_Easy_mega_20201102_test_ESP32_4M316k
Build Platform:⋄ | Linux-4.19.104-microsoft-standard-x86_64-with-glibc2.29
Git HEAD:⋄ | mega-20201102_ca0be3f
```

Is working fine on the actual ESP82866 build.

TD-er, 2020-11-03:
Hmm that sounds like it may be missing an extra slash when accessing the file system on the ESP32.

heinemannj, 2020-11-03:
My check was successful.
Tested with ESP_Easy_mega_20201103_custom_ESP32 build.

TD-er, 2020-11-03:
Thanks for testing :)

PR #3357 (2020-11-03): [ESP32][Mini Dashboard] Not working (#3356)
Fixes: #3356
commit 5b83c7a2 (2020-11-03): [ESP32][Mini Dashboard] Not working (#3356)

Fixes: #3356
closed 2020-11-03 by TD-er