letscontrolit/ESPEasy #3360: Bluetooth data connection instead of WiFi

tonydee2377, 2020-11-03:
Hi,
I am doing a mobile project, and I wanted to use the bluetooth connection to a mobile phone to take advantage of the network and send data through one of the controls, using an ESP32.
Is this BLE function enabled in the latest version?
How can I add this feature?

Thanks

TD-er, 2020-11-03:
We don't have Bluetooth enabled in ESPEasy right now.

Also there simply isn't something as "a Bluetooth connection".
Bluetooth uses profiles to connect between devices.
One such profile can be like "recording audio" or "output audio" or "serial data", etc.

So if your phone supports some kind of bluetooth link + tethering then we still need to know how that Bluetooth link is made.
Is it a serial connection and if so, how to communicate? AT commands? I honestly don't know.

What does work is when you switch on WiFi tethering on your phone and connect your ESP via WiFi to your phone.
That does work very well.

tonydee2377, 2020-11-04:
Yes WiFi tethering works for me. 
My bracelet is fine, but it consumes more battery power, and my mobile WiFi is busy. 
Bluetooth would be a good solution as is the case with fitness devices.
Thank you very much, then I will try to do some tests separately and then I will try to merge the file into the project, in the meantime that a solution on EspEasy comes out.
Bye

TD-er, 2020-11-04:
You're welcome to submit a pull request if you get it to work.

tonydee2377, 2020-11-04:
ok

enesbcs, 2020-11-22:
I've added "ESPEasy P2P"-like BLE direct communication to RPIEasy. (task sync and command sending to units)
https://github.com/enesbcs/rpieasy/blob/master/_C021_BLEDirect.py
According my experiments it works, although as BLE range is very limited it was not worth the fatigue to code it. But if you want to connect to a phone, a mobile app has to be developed on that side i guess..

tonydee2377, 2020-11-23:
OK, thanks
I'll try tomorrow!