letscontrolit/ESPEasy #3361: Release mega-20201102 Gpio PWM not work

giobbe70, 2020-11-06:
PWM not work

tonhuisman, 2020-11-06:
Because of moving the GPIO, and with that the PWM and PULSE, handling out of the core, you will need to add a Switch Device and enable that, that should re-enable PWM handling.
What exact .bin file did you upload?

giobbe70, 2020-11-06:
ESP_Easy_mega_20201102_dev_ESP8266_4M1M.bin

TD-er, 2020-11-06:
Can you give more information (and also try the suggestions made by @tonhuisman ) ?
- Do you have an enabled task with the Switch plugin assigned?
- From where do you give the command? (recently introduced source filtering for commands)

giobbe70, 2020-11-06:
ok now I try and let you know

DittelHome, 2020-11-06:
I have done a short test, without the '"Switch input - Switch".
The PWM command is unkown.
After defining a task  "Switch input - Switch" the PWM command is known again.  
It is a little bit confusing to define a switch input to control a output signal..., my opinion...

TD-er, 2020-11-06:
> It is a little bit confusing to define a switch input to control a output signal..., my opinion...

Could not agree more.
That's also the reason why we did start to split up all of the numerous functionalities of the switch plugin.
Splitting the GPIO command to become an internal command is the initial step.
PWM will be a second one, to be either become an internal command or a plugin (not decided yet) and there are more.

To me, the Switch plugin has grown to become an unmaintainable Frankenstein of a plugin, which is totally non-intuitive, error prone, etc.
At least it is good to know that we're now in some kind of limbo state where things that used to work before no longer work and are not immediately clear to why they stopped working.
So I will try to think of a way to warn users about it, or make it a higher priority to move the PWM command (and some others) to become internal commands so they start working again as they were in the previous builds.

henkiejan1, 2020-11-10:
I also have problems with PWM GPIO via the url directly. When i open the url: http://192.168.2.5/control?cmd=PWM,17,1024,512 i got the error in the browser (Firefox) SyntaxError: JSON.parse: unexpected character at line 1 column 1 of the JSON data. I don't use the switch plugin. I have actions on the url. :)

giig1967g, 2020-11-10:
@henkiejan1 
just add a task with a switch plugin.
You don't need to use it.
Since a few changes ago, the plugins are not loaded, if they are not used, to save memory.

@TD-er: I might think of moving the PWM command in the CORE. What do you think?

henkiejan1, 2020-11-10:
> @henkiejan1
> just add a task with a switch plugin.
> You don't need to use it.
> Since a few changes ago, the plugins are not loaded, if they are not used, to save memory.

Sometime it's good to ask things. This also. I have tried a lot and how the switch plugin works is also a mystery for me... But it works now again. So i have light back in the soldering room. ;) thnx!

TD-er, 2020-11-10:
> @TD-er: I might think of moving the PWM command in the CORE. What do you think?

Yep, that's needed for sure.
Right now I am working on the Dallas stuff, so we won't be working on the same code if you move it to the core like the GPIO code was.
I guess it is a rather simple move to do right now.

DittelHome, 2020-11-11:
Dont forget LONGPULSE, please ...

giig1967g, 2020-11-11:
> Dont forget LONGPULSE, please ...

Is there something wrong with LONGPULSE? 
Can you post an example, please?

TD-er, 2020-11-11:
Hmm I looked an it appears those also moved to be internal commands.
So maybe there are some cases now where they no longer can be used like it also was with the GPIO commands?  (no idea what command source though)

giig1967g, 2020-11-11:
> Hmm I looked an it appears those also moved to be internal commands.
> So maybe there are some cases now where they no longer can be used like it also was with the GPIO commands? (no idea what command source though)

LONGPULSE is an internal command. There shouldn't be any problem using it without the plugin, unless there is a bug :)

TD-er, 2020-11-11:
> unless there is a bug :)

So no problems here...  ;)

DittelHome, 2020-11-11:
`LONGPULSE is an internal command. There shouldn't be any problem using it without the plugin, unless there is a bug :)`

Sorry, for confusing. I thought, we have the same problem with LONGPULSE, as we have with the PWM Command.

giig1967g, 2020-11-11:
> > @TD-er: I might think of moving the PWM command in the CORE. What do you think?
> 
> Yep, that's needed for sure.
> Right now I am working on the Dallas stuff, so we won't be working on the same code if you move it to the core like the GPIO code was.
> I guess it is a rather simple move to do right now.

@TD-er 
I could move the following commands into CORE:
- pwm
- tone
- rtttl
- servo

what do you think?

TD-er, 2020-11-11:
Start with PWM.
I think the rest should become a separate plugin so you can exclude it from a build if not needed.

giig1967g, 2020-11-11:
ok

thomastech, 2020-11-13:
It's been discovered that the P001 commands sent by HTTP are not working in Mega-20201102. Specifically commands _pwm, servo, rtttl, tone_ report a _Command unknown_ error in the browser.

However, use of these commands in rules and Tools->Advanced is OK (no issues).

See this forum discussion: https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=8080&p=48104 

\-Thomas

TD-er, 2020-11-21:
@thomastech 
Made a [test build](https://github.com/letscontrolit/ESPEasy/pull/3383#issuecomment-731652123) for the things you mentioned.
Moved to be internal command:
- pwm
- rtttl
- tone

Still to do:
- Servo

closed 2020-11-23 by TD-er

thomastech, 2020-11-24:
Thanks for fixing this. I will forward the good news to the forum.
\ - Thomas

PR #3383 (2020-11-23): Standardise command execution via web interface (#3374)
Fixes: #3374
Fixes: #3361

Moved PWM command to be an internal command and no longer part of P001.
Also added support for setting a PWM frequency for ESP32.