letscontrolit/ESPEasy #3368: [P095] ILI9341 plugin not in test builds

henkiejan1, 2020-11-11:
Hi,

I i am correct should the P095 ILI9341 plugins availeble in the test builds of the releases of ESPEasy. But i have tried a few but i saw never this plugin in the list. What do i wrong??

Kind regards!

TD-er, 2020-11-11:
Due to size limitations it is not included in the current nightly build.
What other plugins and what build size and ESP type you need?
Then I can make you a build.

henkiejan1, 2020-11-11:
Oh, i can build it with platformio i think. But i can't remember where the plugin definitions are stored. If you could say that should be very helpfull :)

TD-er, 2020-11-11:
See [here](https://github.com/letscontrolit/ESPEasy/tree/mega/tools/pio) the Python files used in PIO to generate "Custom" builds (one for ESP8266 and one for ESP32)

Or copy [Custom-sample.h](https://github.com/letscontrolit/ESPEasy/blob/mega/src/Custom-sample.h) to a new file `Custom.h` in the same dir, edit it and build one of the `Custom` PIO builds (looks for the existence of `Custom.h` file)

Or go a bit more in-depth and have a look at these files: [src/src/CustomBuild](https://github.com/letscontrolit/ESPEasy/tree/mega/src/src/CustomBuild)

tonhuisman, 2020-11-12:
@henkiejan1 You might be interested in pulling in PR #3246 if you own such display with a touchscreen mounted 😃

henkiejan1, 2020-12-27:
> @henkiejan1 You might be interested in pulling in PR #3246 if you own such display with a touchscreen mounted smiley

You know what i want indeed. I have the [Wemos TFT 2.4](https://www.wemos.cc/en/latest/d1_mini_shield/tft_2_4.html) with touch indeed. Very nice display for a very good price. I go play with it!

TD-er, 2020-12-27:
I just uploaded a new nightly build.
See the https://github.com/letscontrolit/ESPEasy/releases
In there are 2 new build sets. One "energy" and one "display", which include the "normal" plugins plus resp. the "energy" or "display" plugins.

tonhuisman, 2020-12-27:
@henkiejan1 now that this is solved, could you close this issue please?

closed 2020-12-27 by TD-er