letscontrolit/ESPEasy #3369: Impossible to connect ESPeasy to my Access Point

Sorcier, 2020-11-13:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [x ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [x ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [x ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ x] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

I flashed my ESP8266 Wemos D1 mini (4MB flash) with 

-ESP_Easy_mega_20201102_normal_ESP8266_4M1M_VCC.bin
-ESP_Easy_mega_20201102_normal_ESP8266_4M1M.bin
-ESP_Easy_mega_20201102_normal_alt_wifi_ESP8266_4M1M_VCC.bin
-ESPeasy R120

Flash without problem i see the SSID of ESPeasy, i see the wizard to connect the wifi and the board can see 10 WiFi network in my flat.
I select my SSID then input the key but there is no connection
My wifi key has space inside and also ' 
is it the problem ? I have ESP32cam flashed to use with MotionEye and there is no problem with WiFi Key

Hardware : 
esp8266mod D1 mini

I think i tried everything, i have no idea why it doesn't work.

Log through the serial connection : 

6904 : Info   : WIFI : Disconnected! Reason: '(2) Auth expire' Connected for 4087 ms
7801 : Info   : WIFI  : Start network scan
9987 : Info   : WIFI : Connecting LiveHome attempt #2
10002 : Info   : WIFI  : Scan finished, found: 12
10004 : Info   : WIFI  : Selected: LiveHome xx:xx:xx:xx:xx:xx Ch:11 (-51dBm) WPA2/PSK
14193 : Info   : WIFI : Disconnected! Reason: '(2) Auth expire' Connected for 4088 ms
15092 : Info   : WIFI  : Start network scan
17277 : Info   : WIFI : Connecting LiveHome attempt #3
17291 : Info   : WIFI  : Scan finished, found: 13
17293 : Info   : WIFI  : Selected: LiveHome xx:xx:xx:xx:xx:xx Ch:11 (-53dBm) WPA2/PSK
20464 : Info   : WIFI : Set WiFi to AP+STA
21455 : Info   : WIFI : AP Mode ssid will be Test-ESP-1 with address 192.168.4.1
21557 : Info   : WIFI : Disconnected! Reason: '(202) Auth fail' Connected for 3087 ms
21576 : Info   : WIFI  : Start network scan

tonhuisman, 2020-11-13:
1) This is more like a support request, that could better be handled at the forum (link above)
2) Can you try one of the alt_wifi builds to see if that is able to connect?

TD-er, 2020-11-13:
I think the `'` and the space are indeed the issue here.
Can you try to set the wifikey using a command via serial?

```
wifikey,"my secret' key"
```

So make sure to wrap the key in quotes on the command line.
If that works, then I guess it is indeed a bug in the web server pages handling.
If that doesn't work, then it is a bug in the WiFi key handling when reading from the settings.

tonhuisman, 2022-04-18:
This didn't get any response from the OP, so is probably solved, and can be closed. Unless there is something left to investigate?

closed 2022-04-18 by TD-er