letscontrolit/ESPEasy #3370: P091 - what to select in ESP_Easy_mega_20201102_normal_ESP8266_4M1M.bin

AloisKlingler, 2020-11-14:
Hi,

I wanted to try P091 (https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P091_SerSwitch.ino ) but I cannot find what I need to select as Device.

Did I select the wrong image?

Thanks!
Best regards
Alois

tonhuisman, 2020-11-14:
1) This is a support question, that could better be handled in the [forum](https://www.letscontrolit.com/forum), this medium is more meant for reporting bug or feature requests
2) You need a TESTING build to have the SerSwitch included, pick one with `test` in the name. You can upload it via the Update Firmware feature on the Tools tab

AloisKlingler, 2020-11-14:
thank you @tonhuisman - this helped.
maybe add this information to https://espeasy.readthedocs.io/en/latest/Plugin/_Plugin.html#list-of-official-plugins or https://www.letscontrolit.com/wiki/index.php/Official_plugin_list at the description of the different colors. :-)

I have at least one relay unit in use, will equip it tomorrow with the test firmware. maybe add me anywhere to a testerlist, but ofc if I can see issues I will report back anyhow. :-)

Best regards
Alois

closed 2020-11-14 by AloisKlingler