letscontrolit/ESPEasy #3374: Sending commands via /control?cmd= returns "unknown command"

TD-er, 2020-11-16:
As described here: https://github.com/letscontrolit/ESPEasy/issues/3351#issuecomment-727903003

We currently have several routes to execute commands.
- tools?cmd=
- control?cmd=
- other sources like MQTT etc.

The `control?cmd=` route does try to split executing commands, into several attempts for internal/plugin/remoteconfig.
However each attempt may result in a reply like "Unknown command".

This should be done more uniform and also it should be made clear what should be done with a reply from executing a command.

TD-er, 2020-11-19:
Places from where to send commands:

- `/?cmd=...` Root page, `VALUE_SOURCE_HTTP`
- `/control?cmd=...` Control page, `VALUE_SOURCE_HTTP`
- `/tools?cmd=...` Tools page, `VALUE_SOURCE_WEB_FRONTEND`

Root page does handle 2 special commands:
- `wifidisconnect`
- `reset` - Factory reset

Those special commands will be scheduled to be run from the main loop.
Any other command is executed immediately and the result is plotted in the web page.

Control page does not output anything other then the command output.
If it should be considered JSON, the output is a JSON stream, otherwise a normal web stream.
The output can be:
- "OK"
- the content of `printWebString` either via a JSON stream or as web stream. (depending on `printToWebJSON` )
- "Unknown or restricted command!"

Tools page does show the output of the handled command in a special text form.

TD-er, 2020-11-21:
Made a [test build](https://github.com/letscontrolit/ESPEasy/pull/3383#issuecomment-731652123) 

On top of making the command handling more generic, some commands are also moved from P001-Switch to be internal commands:
- pwm
- rtttl
- tone

Still to do:
- Servo

AloisKlingler, 2020-11-22:
for plugin P091 I can confirm that this now does not response again with "unknown command" but with OK.

Thank you. :-)

Best regards
Alois

PR #3383 (2020-11-23): Standardise command execution via web interface (#3374)
Fixes: #3374
Fixes: #3361

Moved PWM command to be an internal command and no longer part of P001.
Also added support for setting a PWM frequency for ESP32.
commit 82296051 (2020-11-23): Standardise command execution via web interface (#3374)

Fixes: #3374
closed 2020-11-23 by TD-er