letscontrolit/ESPEasy #3376: "GPIO boot state - Output low" is not applied if a "Switch input -Switch" is configured

AloisKlingler, 2020-11-18:
Hi,

I am using ESP_Easy_mega_20201102_normal_ESP8266_4M1M on a wemos D1.

I am using GPIO-2 (D4) as output for HIGH / LOW states.

I have configured in Hardware -> GPIO boot states this to init as "Output Low".
As soon as I add a "Switch input - Switch" Device and attach it to GPIO-2, the pin remains at HIGH after a reboot.

/control?cmd=status,GPIO,2 shows me:
{
"log": "",
"plugin": 1,
"pin": 2,
"mode": "input",
"state": 0
}

the device has the value "0". quite misleading, because the pin stays HIGH.

after switching one time with /control?cmd=GPIO,2,1 and then /control?cmd=GPIO,2,0 it shows me correct output and state 0

if I remove the Switch input device again, it behaves correctly, on reboot the pin is pulled low after boot and set to output.

Best regards
Alois

giig1967g, 2020-11-18:
@AloisKlingler 
What is happening is not a bug.

First of all, it's important to explain that a pin can be configured to be either an INPUT **or** an OUTPUT pin. It's type will affect it's operations.

This is what you are doing:
a) You are setting the GPIO2 to be an OUTPUT pin (value=LOW) in Hardware Tab.
b) Then you are setting the same pin to be an INPUT SWITCH.
c) Finally you are sending a GPIO command to pin 2, forcing him to be an OUTPIN pin.

Some theory here:
A single PIN cannot be an INPUT and a OUTPUT at the same time.
The pin type can change according to the commands that are sent to the pin itself.

So in a) you are setting the pin to be an OUTPUT pin. In b) as you are creating a SWITCH INPUT, you are changing the type to INPUT and the value becomes the current status of the pin. And in c) you are changing again the type to OUTPUT. 

The order of processing of the pin type is: 
1) Hardware tab
2) Plugin INIT
3) Rules

So, you have two alternative solutions to your problem:
1) do not configure a SWITCH INPUT for GPIO2, as it's an output pin
OR
2) if you need to define GPIO2 in the task list, then add the following in the rules:
```
on System#Boot do
 gpio,2,0
endon
```

Hope this helps.

AloisKlingler, 2020-11-18:
@giig1967g thanks for clarification, I thought already this is maybe wanted, as I define a "Input"-switch.

maybe it would make sense to give the internal GPIO plugin handling a configureable device which makes it updating and pushing it's state. I think it is not an uncommon way to drive relays or (in my case) transistors with the HIGH / LOW of GPIO pins. :-)

Best regards
Alois

giig1967g, 2020-11-18:
> @giig1967g thanks for clarification, I thought already this is maybe wanted, as I define a "Input"-switch.
>
> maybe it would make sense to give the internal GPIO plugin handling a configureable device [...]

good.
can you close the issue?
thanks

closed 2020-11-18 by TD-er