letscontrolit/ESPEasy #3377: Emoncms controller not sending data...

ratmole, 2020-11-18:
As it seems, from the release mega-20200829 and on-wards, there is a problem in the Emoncms controller.
(i think there is a issue on all the controllers)

in this if statement :
https://github.com/letscontrolit/ESPEasy/blob/985b05c732436acbab333398a035ef92043c4fe0/src/_C007.ino#L48
the C007_DelayHandler is always equal to nullptr!

Using the  ESP_Easy_mega_20200812_normal_ESP8266_4M1M.bin file from the mega-20200812 release works just fine.

thanks

TD-er, 2020-11-18:
Found the bug...

```c++
    case CPlugin::Function::CPLUGIN_INIT:
      {
        success = init_c004_delay_queue(event->ControllerIndex);
        break;
      }

    case CPlugin::Function::CPLUGIN_EXIT:
      {
        exit_c007_delay_queue();
        break;
      }
```

`init_c004_delay_queue` must be `init_c007_delay_queue`

ratmole, 2020-11-18:
nice one! thanks again

PR #3379 (2020-11-18): [EmonCMS] Fix copy/paste bug (#3377)
Fixes: #3377
commit 8fb51360 (2020-11-18): [EmonCMS] Fix copy/paste bug (#3377)

Fixes: #3377
closed 2020-11-18 by TD-er