letscontrolit/ESPEasy #3381: [ESP32][Initial firmware]

heinemannj, 2020-11-19:
The 'esptool.exe' part of the actual ESP32 releases is not working for ESP32 flashing.
Initial flashing procedure for ESP32 is missing or at least confusing.

Initial images needed for initial setup are also not included too:
- boot_app0.bin
- bootloader.bin
- partitions2.bin

Attached zip file is addressing above including a little bit more precise procedure for initial flashing of ESP32 (hopefully) ...
This topic was already discussed in previous issues - I didn't reinvent the wheel ...

Many thanks for your support - Cheers

[esp32tool-initial-flashing.zip](https://github.com/letscontrolit/ESPEasy/files/5566286/esp32tool-initial-flashing.zip)

TD-er, 2020-11-19:
For initial flashing, those files are included with the correct offsets in the files with "factory" in their name.
It should also be mentioned in the including README file and in the [documentation](https://espeasy.readthedocs.io/en/latest/Participate/ProjectStructure.html?highlight=factory#esp32-builds)

ironataerial, 2020-11-19:
The screenshots in this threat help a lot :

[https://letscontrolit.com/forum/viewtopic.php?f=5&t=7997&p=47444&hilit=esp32#p47444](url)

TD-er, 2020-11-19:
[image]
[image]
[image]
[image]
[image]
Screenshots like these are clear enough?

ironataerial, 2020-11-19:
First ESP32 download tool image correctly shows the factory.bin @ 0x0
Second ESP32 download tool image should show the current build @ 0x10000, no ?

TD-er, 2020-11-19:
Nope, when flashing via serial, you must use the "factory" image, starting at 0.
An update via OTA (via web interface, you must use the image without "factory" in the name.

So no need to place any offset, selecting multiple files for the flash tool, etc.

- Via serial => "Factory" image
- other upgrade options => image without "factory" in the name.

ironataerial, 2020-11-19:
For initial setup when using the "ESP32 download tool" you first have to flash the factory image at 0x0 as step 1. 
As step 2 you have to flash the current .bin at 0x10000.  (or is this not necessary ?)
This is the "not so clear" step for first timers that try to convert a "fresh" ESP32 to ESPEasy.
After that OTA as usual with a new release .bin

TD-er, 2020-11-19:
There is no step 2 in flashing the factory image.

Just flash the factory image starting at offset 0.
Like I said, all other bin files mentioned in lots of other firmware tutorials are included in the "Factory" image, at the correct offsets.

It is just 1 file, no offset.

ironataerial, 2020-11-19:
The instructions in the initial forum link : 

"...The Factory firmware needs to be flashed to 0x0 first. Once that's flashed, flash the actual firmware to 0x10000...."

That is what I followed initially. 

Second step is not required, this is clear now.

heinemannj, 2020-11-20:
> When flashing **via serial**, you must **use the "factory" image, starting at 0**.
> An **update via OTA (via web interface)**, you must **use the image without "factory"** in the name.
> 
> **All other bin files mentioned in lots of other firmware tutorials are included in the "factory" image**, at the correct offsets.
> So no need to place any offset, selecting multiple files for the flash tool, etc.

Conclusion for ESP32 flashing:
> Via serial => "factory" image
> Other upgrade options => image without "factory" in the name.

By adding above screenshots into the ESP32 part of documentation it will be water proofed ...

heinemannj, 2020-11-20:
My proposal is to split the README into 2 different files:
a) Instructions for ESP82xx
b) Instructions for ESP32

heinemannj, 2020-11-20:
Based on above clarifications I have reworked the 'esp32tool-initial-flashing.cmd' using a compiled version of Espressif's esptool.py

esp32tool-initial-flashing.cmd ...
... is a flashing script for ESPEasy_ESP32.
You need to run it in elevated mode (as admin) to fetch the COM ports correctly.
You have to customize the default COM settings first (based on your WIN 10 COM settings)!

REM erase
esp32tool.exe --chip esp32 --port COM8 --baud 256000 --before default_reset --after hard_reset erase_flash

REM flash ESP_Easy_mega_20201102_test_ESP32_4M316k-factory.bin
esp32tool.exe --chip esp32 --port COM8 --baud 256000 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x0 ./bin/ESP_Easy_mega_20201102_test_ESP32_4M316k-factory.bin

For flashing ESP32 you can also use Espressif's own Flash Download Tools.
The latest version can be downloaded from: https://www.espressif.com/en/support/download/other-tools

heinemannj, 2020-11-20:
> Also I am not sure if we can include the Espressif binaries in our downloads, due to licensing issues.

As we have already clarified that no additional Espressif firmware bin files are required for flashing with ESPEasy firmware, the licensing topic is reducing to usage of  Espressif's 'esptool.py'.

https://github.com/espressif/esptool/blob/master/LICENSE
> espressif/esptool is licensed under the
> 
> GNU General Public License v2.0
> The GNU GPL is the most widely used free software license and has a strong copyleft requirement.
> When distributing derived works, the source code of the work must be made available under the same license.
> There are multiple variants of the GNU GPL, each with different requirements.

heinemannj, 2020-11-20:
Adding the following note into ESPEasy license file will avoid  any potential non compliancy:

```
/*****************************************************************************************************************************\
 * Arduino project "ESP Easy" © Copyright www.esp8266.nu
 *
 * This program is free software: you can redistribute it and/or modify it under the terms of the GNU General Public License
 * as published by the Free Software Foundation, either version 3 of the License, or (at your option) any later version.
 * This program is distributed in the hope that it will be useful, but WITHOUT ANY WARRANTY; without even the implied warranty
 * of MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU General Public License for more details.
 * You received a copy of the GNU General Public License along with this program in file 'License.txt'.
 *
 * Additional information about licensing can be found at : http://www.gnu.org/licenses
\*****************************************************************************************************************************/

// This file incorporates work covered by the following copyright and permission notice:

/*****************************************************************************************************************************\
* Arduino project "Nodo" © Copyright 2010..2015 Paul Tonkes
* 
* This program is free software: you can redistribute it and/or modify it under the terms of the GNU General Public License 
* as published by the Free Software Foundation, either version 3 of the License, or (at your option) any later version.
* This program is distributed in the hope that it will be useful, but WITHOUT ANY WARRANTY; without even the implied warranty 
* of MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU General Public License for more details.
* You received a copy of the GNU General Public License along with this program in file 'License.txt'.
*
* Voor toelichting op de licentievoorwaarden zie    : http://www.gnu.org/licenses
* Uitgebreide documentatie is te vinden op          : http://www.nodo-domotica.nl
* bugs kunnen worden gelogd op                      : https://code.google.com/p/arduino-nodo/
* Compiler voor deze programmacode te downloaden op : http://arduino.cc
* Voor vragen of suggesties, mail naar              : p.k.tonkes@gmail.com
\*****************************************************************************************************************************/

/*****************************************************************************************************************************\
 * espressif/esptool is licensed under the
 *
 * GNU General Public License v2.0
 * The GNU GPL is the most widely used free software license and has a strong copyleft requirement.
 * When distributing derived works, the source code of the work must be made available under the same license.
 * There are multiple variants of the GNU GPL, each with different requirements.
 *
 * Additional information about licensing can be found at : https://github.com/espressif/esptool/blob/master/LICENSE
\*****************************************************************************************************************************/
```

heinemannj, 2020-11-20:
https://github.com/espressif/esp-at/blob/master/LICENSE

```
ESPRESSIF MIT License

Copyright (c) 2018 <ESPRESSIF SYSTEMS (SHANGHAI) PTE LTD>

Permission is hereby granted for use on all ESPRESSIF SYSTEMS products, in which case,
it is free of charge, to any person obtaining a copy of this software and associated
documentation files (the "Software"), to deal in the Software without restriction, including
without limitation the rights to use, copy, modify, merge, publish, distribute, sublicense,
and/or sell copies of the Software, and to permit persons to whom the Software is furnished
to do so, subject to the following conditions:

The above copyright notice and this permission notice shall be included in all copies or
substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY, FITNESS
FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR
COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER
IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN
CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.

```

TD-er, 2020-11-20:
I will have a good look at the license and when possible include the Espressif flasher in the ESP32 ZIP file.

Maybe also a good idea to check all libraries we use if we have all the required license files present.
Usually I copy all files of the libraries just for this purpose of making sure to include them.
But I am not 100% sure it was done like this for libraries we already use for years without the need to update them.

DutchmanNL, 2020-12-10:
After a day of searching found this issue and used the steps, Plus 1 works like a charm !

TD-er, 2020-12-10:
Thanks for confirming it and remembering me to update the docs :)

DutchmanNL, 2020-12-10:
> and remembering me to update the docs :)

good idea 👍 and nice job !

heinemannj, 2020-12-12:
> I will have a good look at the license and when possible include the Espressif flasher in the ESP32 ZIP file.

Including of Espressif flasher in the ESP32 ZIP file is compliant to both:
a) ESPEasy licensing model
b) espressif/esptool licensing model

Please see pull request #3402 - Add copyrights and permission notices

TD-er, 2020-12-15:
I added documentation for it and the Espressif Download Tool

DutchmanNL, 2020-12-15:
there should be no need to include the software, just link were you can get it and the licensing part is done.
Why should you get the software with every download of e new version on your PC. that makes not really sense :)

heinemannj, 2020-12-15:
To be frankly:

- Attached esptool seems to be a specificly patched version which has been provided by ESPEasy folks during previous ESPEasy issues
- The uncompressed binary is only ~3,3 MB
- No need for installation/setup - just only one exe file and a simple batch file

TD-er, 2020-12-15:
> there should be no need to include the software, just link were you can get it and the licensing part is done.
> Why should you get the software with every download of e new version on your PC. that makes not really sense :)

I was thinking about change the build scripts to leave out the flash tools and offer them as a separate download.
We have had the download link for the flasher in the readme, but still lots of users ask about the flashing procedure.

So I really don't know what is the best way to handle this.
I will make sure to have a version at hand somewhere just in case the Espressif site is down, links change etc.
Wouldn't be the first time depending on some site for tools appeared to be a bad idea.

heinemannj, 2020-12-15:
> Wouldn't be the first time depending on some site for tools appeared to be a bad idea

100% confirmation because of following potential disasters:

a) Distribution web site is getting broken - low risk
b) Changes on origion source code and binary results in incompatibilities for ESPEasy setup procedures - high risk
c) Distribution web site will be terminated for ever - high risk

Do not trust somebody else - Do it yourself ...

closed 2022-11-09 by TD-er