letscontrolit/ESPEasy #3385: plugin  P091 Serial MCU controlled switch: relaylonglulse command not pulsing but switching

AloisKlingler, 2020-11-22:
hello,

I am using ESP_Easy_mega_20201102_normal_ESP8266_4M1M
I have also tried the test build from here: https://github.com/letscontrolit/ESPEasy/issues/3374#issuecomment-731652447 (ESP_Easy_mega_20201121_test_ESP8266_4M1M_VCC)

I have configured a LC tech 4ch relay module. I use http://<IP>/control?cmd=relaylongpulse,2,1,1 and would expect switching the relay off after 1second again.

This does not work, it switches the relay on, but never off again.

here is a log of such an action:

```
138625: WD : Uptime 2 ConnectFailures 0 FreeMem 20600 WiFiStatus WL_CONNECTED ESPeasy internal wifi status: Conn. IP Init
147646: HTTP: relaylongpulse,2,1,1
147653: SerSW : SetSwitchPulse r2:1 Pulse for 1 sec
147712: EVENT: WohnzimmerRelays#Relay0=0
147811: EVENT: WohnzimmerRelays#Relay1=0
147818: ACT : timerset,1,1
147911: EVENT: WohnzimmerRelays#Relay2=1
147922: ACT : timerset,1,1
148011: EVENT: WohnzimmerRelays#Relay3=0
148923: EVENT: Rules#Timer=1,1
148951: ACT : SendToHTTP,192.168.0.250,80,/apcucollector.php?light=WohnzimmerDimmer&ipv4=192.168.10.244&uptime=2&load=6.94&freeram=201
148972: ACT : timerSet,1,30
168626: WD : Uptime 3 ConnectFailures 0 FreeMem 20472 WiFiStatus WL_CONNECTED ESPeasy internal wifi status: Conn. IP Init
178974: EVENT: Rules#Timer=1,1
179002: ACT : SendToHTTP,192.168.0.250,80,/apcucollector.php?light=WohnzimmerDimmer&ipv4=192.168.10.244&uptime=3&load=5.48&freeram=200
179021: ACT : timerSet,1,30
```

in the custom built firmware from forum this worked.

Would be glad if we can get it working here too. 👍 

Thank you!

Best regards
Alois

AloisKlingler, 2020-12-07:
nobody?

TD-er, 2020-12-08:
Sorry, had this one still marked again as "unread", meaning I had to take a longer look at it, but haven't gotten to it yet.

AloisKlingler, 2022-03-27:
just wanted to ask charily if there is a realistic chance to view at this one? :-)

AloisKlingler, 2022-11-09:
as I read in release notes 20221105:
```
[LongPulse] Stop scheduled pulse when setting GPIO + add waveform call
[Longpulse] Cleanup and document
[Longpulse] Fix missing includes
```

I have retried and those fixes did not influence the behavior here: it does not work.

BR
Alois

TD-er, 2022-11-09:
Those longpulse command changes I made were for the internal GPIO pins.
The serial MCU controlled switch from P091 is not a normal internal GPIO pin.

Anyway, I will have a quick look at it.

TD-er, 2022-11-09:
OK, this plugin code really needs to have some cleaning up to do.
For example the relaypulse command does use a delay() call, which is really bad.

The longpulse does try to reschedule a call to turn it off, but it is using the completely wrong parameters.
```c++
            Scheduler.setPluginTaskTimer(timer, PLUGIN_ID_091, event->TaskIndex, rnum, !rcmd);
```
Where there is `PLUGIN_ID_091` the function expects a taskindex.
So this is never going to be scheduled.

TD-er, 2022-11-09:
Can you test [this test build](https://www.dropbox.com/s/mjxox20h1adr9u5/ESP_Easy_mega_20221109_collection_C_ESP8266_4M1M_VCC.bin?dl=0) ?

AloisKlingler, 2022-11-09:
awesome @TD-er - works on both devices now flawless. :-)

Thank you! :-)

TD-er, 2022-11-09:
Sorry it took so long for me to have a proper look at it...

PR #4348 (2022-11-09): [LCtech switch] Fix longpulse (#3385)
Fixes: #3385
commit bade047f (2022-11-09): [LCtech switch] Fix longpulse (#3385)

Fixes: #3385
closed 2022-11-09 by TD-er