letscontrolit/ESPEasy #3389: Heatpump IR - Mitsubishi MSZGA35VA

gorfo66, 2020-11-24:
I'm trying control my Heatpump and using the HeatPump plugin present now in the repository.
Problem: there is no mitsubishi_msz brand/model available in the code.
I was using mitsubishi_fe and mitsubishi_fd which was somehow ok.

But I realized that contrary to orders provided by my remote, the fan is absolutely never stopping, even when we are reaching the expected temperature. Said differently, with my stock remote, I'm asking : heat, fan = 1, temp = 16 in the bed rooms.
When we are reaching 16deg, the fan is stopping.

With Arduino, the fan is not stopping and we can get temperature up to 24deg.

This is quite annoying in fact and I would like to know how I can fix this problem, or if anyone has it.
Best
Romain

TD-er, 2020-11-24:
@jimmys01 has probably the most knowledge about the IR specific part, although I'm not entirely sure if the Heatpump IR plugin maybe also does something different.

Just for helping any debugging attempts, can you provide the exact build you use?

gorfo66, 2020-11-24:
Unfortunately, I'm using a very old build, before the merge of the libray into official ESPEasy build.
Testing a new version is tedious for me because I don't have spare IR transmitter.

However, I made a diff between the latest level of code of ESPEasy and my old fork of the library and i don't see any change in the files located in /lib/HeatpumpIR/. Therefore I assume that you can reproduce with any build number.

The framework itself works well because IR codes are sent.
I suspect something wrong in the values being sent.

More info: the reference of the standard remote control is : KM05B

jimmys01, 2020-11-24:
Unfortunately, I do not have experience with the HeatpumpIR library. 
I would suggest to use the more powerful IRremoteESP8266 library.

gorfo66, 2020-11-24:
Found the issue while debugging the lib. 
Stupidly, the temperature must be strictly higher than 16 degre (can equals using standard remote). If not, it sends default value of 23degre.

gorfo66, 2020-11-24:
Cause found. I will ensure temperature remains above 17 degre

closed 2020-11-24 by gorfo66