letscontrolit/ESPEasy #3391: Since release ESP_Easy_mega_20201022 buzzer doesn't work anymore (rtttl, tone)

hcremer, 2020-11-24:
Since release ESP_Easy_mega_20201022 buzzer doesn't work anymore (rtttl, tone)
Only getting "Command unknown:"
On release ESP_Easy_mega_20201016 it still works fine.

TD-er, 2020-11-24:
It has already been fixed, but not yet in a new build.

See this merged PR: https://github.com/letscontrolit/ESPEasy/pull/3383
There are some test builds linked in that PR (Servo was not yet fixed in that test build)

hcremer, 2020-11-24:
Great! Thank you very much! I am looking forward to the new build :-)

TD-er, 2020-11-24:
> Great! Thank you very much! I am looking forward to the new build :-)

Have you tested the test builds I linked in that PR?

hcremer, 2020-11-26:
Just tested it:
Command: "tone,14,1300,200" works.
but
Command: "rtttl,14:d=10,o=6,b=180,c,e,g" doesnt't work and shows the following error:
Too many arguments, try using quotes!Command unknown: rtttl,14:d=10,o=6,b=180,c,e,g

TD-er, 2020-11-26:
Can you place quotes on the "argument"?

`rtttl,"14:d=10,o=6,b=180,c,e,g"`

TD-er, 2020-11-26:
Ah found the bug..
As I also made another error initializing the nr. of decimals, I will make a test build for these two and later a new build.

TD-er, 2020-11-26:
Can you test if this fixes it?
- [ESPEasy_ESP32_mega-20201125-4-PR_3395.zip](https://www.dropbox.com/s/ir0b7mo4bbqvtum/ESPEasy_ESP32_mega-20201125-4-PR_3395.zip?dl=0)
- [ESPEasy_ESP82xx_mega-20201125-4-PR_3395.zip](https://www.dropbox.com/s/91ln6f8up993atr/ESPEasy_ESP82xx_mega-20201125-4-PR_3395.zip?dl=0)

hcremer, 2020-11-26:
Just tested it. Now, tone and rtttl both are working :-) Great job!

hcremer, 2020-11-26:
Only tested on ESP8266 with ESP_Easy_mega_20201126_normal_ESP8266_4M1M

TD-er, 2020-11-26:
The bug was in code that's present in all builds, so for this bug testing only a single binary is enough.

Thanks for testing.

closed 2020-11-29 by TD-er

netson01, 2022-02-06:
It seems that latest build "_mega-20211224_f162ebf_" does not support rtttl "_39185912: RTTTL : command not included in build_". Which build should i use to make rtttl work?

tonhuisman, 2022-02-06:
The TEST and MINIMAL builds won't have rtttl enabled, to reduce the .bin size. Any NORMAL build should have it enabled, though.

netson01, 2022-02-06:
I am using _ESP_Easy_mega_20211224_normal_ESP8266_1M_ and rtttl is disabled here.

tonhuisman, 2022-02-06:
Ah, indeed, in the 1M builds it is also excluded, same issue, not enough space to fit it in. You could create a Custom build that has only the plugins you need, then it should be possible to not exclude the rtttl feature.

rgr0b50n, 2023-03-09:
Hello,
I am using ESP_Easy_mega_20230306_normal_IRext_no_rx_ESP8266_4M2M and rtttl is not working. According to the logs "RTTTL : command not included in build".
Can you please tell me which build should I use or what do I have to do in order to be able to use rtttl?
Thank you

tonhuisman, 2023-03-09:
Because of the construction of the IR library it uses quite a large amount of a specific type of memory, requiring it to be compiled with the `LIMIT_BUILD_SIZE` flag to build successfully, but that also causes the RTTTL feature to be excluded from the build.
Looking at the name of the build you use, you are most likely using the IR transmit plugin. Then you would have to self-build a Custom build, but there is currently no configuration for that combo: Custom, IR and a 2M file-system. 🤔 

When _not_ using the IR transmit plugin, you should be able to install a normal_ESP8266 build, but that has the limitation that there is no 4M2M flash layout available, so it would require to save your current config from the Tools page, flash a 4M1M build and restore the config.dat file (after renaming/copying the previously saved config file).

tonhuisman, 2023-03-09:
@rgr0b50n
I've opened PR #4541 to enable the `RTTTL` command for this specific build, and allow that to be used for other builds as well.
You can test by downloading from [this Actions build](https://github.com/letscontrolit/ESPEasy/actions/runs/4378518374) once it's finished.

rgr0b50n, 2023-03-10:
However there is some distortion in the sound, not being so clear as before.
I just upgraded another one that had a 2019 build on it with buzzer sounding good but now it has this annoying distortion.
Do you have any idea why is this happening?

tonhuisman, 2023-03-10:
> but now it has this annoying distortion.

Do you also have commands like `pwm` running, or timers active?
There have been changes in the way timers and schedules are handled, since 2019, and both `pwm` and `tone` use timer functions, so that could be related.

TD-er, 2023-03-10:
N.B. the network activity LED is also using this timer.

rgr0b50n, 2023-03-10:
No pwm but I have active timers.
I removed everything from rules leaving only the rtttl to run at boot but no change.
@TD-er I'm not sure I understand what you mean by network activity LED.
I have power led, relay led, charging led (not used) and WiFi led.

TD-er, 2023-03-11:
> WiFi led.

That one...
The "WiFi Status LED"

rgr0b50n, 2023-03-11:
Ah, ok. I thought that this is just indicating the status (connected or not) but now that you said I noticed that it's pulsating when there is activity.
The distortion is still present though I disabled the WiFi led and removed the rules.

commit 432652d7 (2020-11-29): [RTTTL] Fix check argument count for RTTTL command (#3391)

Fixes: #3391

RTTTL argument uses a lot of commas (`,`), so don't check for nr of arguments