letscontrolit/ESPEasy #3392: [P087] Very often reset caused by hardware watchdog

uwekaditz, 2020-11-24:
I use the P087 plug in to read data from my TS125 temperature control.
The data are transmitted once a second as a 7 byte data field @1200 baud.
Therfore I changed the code of the [P087] to use 1200 baud.
I never send any data and use the `SoftwareSerial` setting.
[image]
I can read the data but it happens very often (at least once a day) that the ESP8266 is reset by the hardware watchdog.
**Is this issue known and is there a way to avoid it?**

TD-er, 2020-11-24:
Software serial should not be used at such a slow baudrate.
Please try using a hardware serial for this.

uwekaditz, 2020-11-24:
What does hardware serial means? Is this a special IC that has to be used?

TD-er, 2020-11-24:
In the "Serial port" selector you now have selected "Software Serial"
If you switch that to something like:
- HW Serial0: GPIO-3 (D9) <- TX/GPIO-1 -> RX
- HW Serial0 swap: GPIO-13 <- TX / GPIO-15 -> RX

[image]

Then you are using a hardware (HW) serial port, meaning the ESP does handle the serial communication in hardware and not in software (SW).

Please be aware that if you connect something to GPIO-15, you need to add a PNP transistor to make sure GPIO-15 is not pulled up during boot.
But if you only read from the sensor, there is no need to connect the ESP's TX to the sensor's RX.
So just use GPIO-13 as ESP's RX.

N.B. as it is "Serial0" you must make sure not not have "serial port enabled" checked on the Tools->Advanced page, or else ESPEasy will try to read incomming data on the serial port as commands.

uwekaditz, 2020-11-25:
I changed to `HW serial0 swap` and I can receive the data. Lets see how stable it works.
I tried the `HW serial` already while starting this project, but it didn't work because I did not turn off the serial port on the Tools->Advanced page.
By the way what is the problem with the `SoftwareSerial` using low baud rates?
I thought it is using ISR or timers to receive and transmit data, that should not trigger the hardware watchdog.

TD-er, 2020-11-25:
I'm not 100% sure why, but I assume the ISR call for SWserial does try to handle a few bits at once (e.g. a byte), which essentially blocks everything else.
A single byte at 1200 baud roughly takes 10 msec.
So receiving 10 bytes already blocks almost anything for 100 msec (with only very short intervals inbetween)
This does affect WiFi as packets are no longer received (also acting on interrupts which are disabled when handling incoming SW serial data).
Thus it may cause network state to become in somewhat limbo state which may cause all kinds of timeouts.
Timeouts on the network are sadly not always handled well. => watchdog reset.

uwekaditz, 2020-11-29:
Update after changing to `HW serial0 swap` @1200Baud:
The ESP8266 runs now for 4 days, 360000 messages received, 0 errors and resets.
This looks quite stable.

TD-er, 2020-11-29:
Thanks for getting back on this.
Glad it is working well now.

closed 2020-11-29 by TD-er