letscontrolit/ESPEasy #3394: Rotary encoder counter doesn't change value via MQTT

max1666, 2020-11-25:
Hi. Im making IOT AC dimmer and I want use rotary encoder to save compatibility. The encoder works fine. the value changes as it should, but I cant change encoder counter via MQTT. This is a bug or feature?

TD-er, 2020-11-25:
There is a command in the P059_Encoder plugin to set a value.
The command is `encwrite` with a single parameter.

Controllers C002, C005, C006 and C014 do support receiving commands via MQTT.
Each in its own particular format, so you may need to look in the platform you're using how to format the commands.

max1666, 2020-11-25:
I've already tried it. I see` 2646782: QEI: encwrite 30` in the log, but I don't see the change on the Device page, nothing is published to the broker either.

max1666, 2020-11-25:
Encoder changes published to 'ESP8266/dimmer/Counter'. In the same topic I'm trying to publish my changes. This is wrong?

TD-er, 2020-11-25:
Depends on what controller you're using.
If you search through the code for `VALUE_SOURCE_MQTT` you'll see this flag is used only on a single line in the controllers I mentioned.
There you can see how the topic must be formatted for it to be recognized as a command.
See for example this implementation in C005:

https://github.com/letscontrolit/ESPEasy/blob/8fdf230991bc3ff1d7e9779b02742fa5391c7f19/src/_C005.ino#L67-L122

TD-er, 2020-11-25:
I took another look at the P059_Encoder code:

https://github.com/letscontrolit/ESPEasy/blob/8fdf230991bc3ff1d7e9779b02742fa5391c7f19/src/_P059_Encoder.ino#L177-L194

This one does update the values in the internals of the encoder, but it should then immediately call a PLUGIN_READ to update the output values.
PLUGIN_READ is called at the set interval of a task, or when you reschedule it.

A simple change could be:

```c++
            if (command == F("encwrite"))
            {
              if (event->Par1 >= 0)
              {
                log = String(F("QEI  : ")) + string;
                addLog(LOG_LEVEL_INFO, log);
                P_059_sensordefs[event->TaskIndex]->write(event->Par1);
                Scheduler.schedule_task_device_timer(event->TaskIndex, millis());
              }
              success = true; // Command is handled.
            }
```
To schedule a new call to PLUGIN_READ

max1666, 2020-11-25:
Thanks! Im using Home Assistant (openHAB) MQTT controller. should i use the `ESP8266/dimmer/cmd` topic and data `encwrite,50`? I understand correctly?

TD-er, 2020-11-25:
It looks like it.
If you see the "QEI" log entry, then the command is executed from wherever you send it.

max1666, 2020-11-25:
> Thanks! Im using Home Assistant (openHAB) MQTT controller. should i use the `ESP8266/dimmer/cmd` topic and data `encwrite,50`? I understand correctly?
  I uploaded the latest release, everything works like this. Thank you very much!

closed 2020-11-25 by max1666