letscontrolit/ESPEasy #3398: P102_PZEM004Tv3.ino prebuild 2 errors: safe_strncpy() & parseString()

ipa64, 2020-11-27:
Hi,
I built a custom version with P102_PZEM004Tv3 plugin.
On MacOs Vscode+platformio:
- Rename (_)Custom.h and (_)P102_PZEM004Tv3.ino
- ESPEasy_common.h uncomment #define USE_CUSTOM_H
- Custom.h add #define USES_P102
- Custom.h comment some elements I will not use like Domoticz, GPS,...
While editing (before build), there is 2 errors in the file P102_PZEM004Tv3.ino
safe_strncpy() & parseString() need to update include path 

I can build the project with success, but I would like to solve this wrong behavior on editing
Thank you

TD-er, 2020-11-27:
> ESPEasy_common.h uncomment #define USE_CUSTOM_H

This part should not be needed if you use PlatformIO.
PlatformIO runs a pre-build script I wrote to check if a file named Custom.h exists and then defines `USE_CUSTOM_H`

> Rename ()Custom.h and ()P102_PZEM004Tv3.ino

Rename of the `_P102_PZEM004Tv3.ino` is also not needed.
Rename the `Custom-sample.h` to `Custom.h` should be the only part that's needed, as P102 already is present in the main repo.

> Custom.h add #define USES_P102
> Custom.h comment some elements I will not use like Domoticz, GPS,..

Yep, that's needed indeed.

> While editing (before build), there is 2 errors in the file P102_PZEM004Tv3.ino
safe_strncpy() & parseString() need to update include path

That sounds like the only issue??

TD-er, 2020-11-27:
I just made a test build here and it builds just fine if I only rename Custom-sample.h to Custom.h and add `#define USES_P102`

I did notice the Custom-sample.h file is not up-to date with the latest possible defines.

The file _P102_PZEM004Tv3.ino includes this:
```c++
#include "_Plugin_Helper.h"
```
Which includes "src/Helpers/StringConverter.h", so the  functions you mention are also known when compiling.

ipa64, 2020-11-28:
So, I made a Custom.h from  Custom-sample.h (instead of _Custom.h) and rename back _P102_PZEM004Tv3.ino
No all is building without any error 
Thank you TD-er

closed 2020-11-28 by ipa64