letscontrolit/ESPEasy #3399: Release mega-20201125, DS18b20 don't work, release before is working

fly74, 2020-11-27:
Hi, DS18b20 shows no values since this release.

TD-er, 2020-11-27:
Are you using parasitic power for the Dallas sensors?

fly74, 2020-11-27:
What is it? Back to Release mega-20201102 works.

TD-er, 2020-11-27:
Dallas sensors can be powered in 2 ways:
- Voltage applied to Vdd pin
- Short Vdd to data pin and thus power it via the data pin.

That last one is "parasitic powered" but it requires a different way of measuring.
I did not implement that way yet as I was under the impression parasitic powered was not working well until now.
I'm still working on that part.

If you have the sensors connected by powering them via the Vdd pin, then I really would like to know what sensors you are using as I have a setup here with 6 Dallas sensors connected to a single pin and that's working very stable here.

fly74, 2020-11-27:
These https://www.amazon.de/dp/B01MZG48OE/ref=pe_3044161_185740101_TE_item?th=1 are the sensors

[image]

powered like here https://espeasy.readthedocs.io/en/latest/Plugin/P004.html

TD-er, 2020-11-27:
Hmm I never tested it with GPIO 16.
If you look in the "Device Address" combo box (as shown in your screenshot), does it show more info?
Like this:
[image]

What I find strange is that your screenshot doesn't show the task to which the address is assigned.

tonhuisman, 2020-11-27:
> What I find strange is that your screenshot doesn't show the task to which the address is assigned.

The 'Number Output Values' combo is not there either, so that screenshot is showing an older ESPEasy build I assume.

fly74, 2020-11-27:
New Release shows this
[image]

The old one only the address.

fly74, 2020-11-27:
Second flash and submit from device fixed it without changing any values. Strange.

closed 2020-11-27 by fly74

TD-er, 2020-11-27:
Maybe the settings had to be written again?
I will have a look at it as another user also reported something similar on the forum.
Maybe I can check the settings when init the task and patch missing or not valid values.

TD-er, 2020-11-27:
Ah found the issue...
I added a selector to set the nr of output values.
But that one wasn't present before and thus it was set to 0 which equals "none"
I will add a check for "simple" selectors that only allow "single", "double", "Triple" and "Quad" to not accept "none".