letscontrolit/ESPEasy #3400: switch using optocoupler switching state issue

TungstenE2, 2020-11-27:
Hi all,

I have an issue with an optocoupler connected to my door bell and configured as a switch.

Tested with Release mega-20201125 and ESP_Easy_mega_20200721.

The issue is that the state is send to controller FHEM.

In FHEM the state is switching very fast between on/off if the door bell is pressed.

Below is the wiring schema:

[image]

I tried to play with the advanced event management, but did not succeed to get a stable state while the door button is pressed.

[image]

I assume this is related to door bell operating on 13V AC, but the wireing using a diode should solve this, so the optocoupler should not switch state while triggered.

Can anybody advice on how to solve this?

[image]

[image]

Thx

tonhuisman, 2020-11-27:
Doesn't that hardware setup 'press' your button 50 times per second (as the AC frequency is 50Hz, thus trigger 100 times per second) while the bell-button is pressed?

TungstenE2, 2020-11-27:
This is what I also assume, but I understood from the documentation that this is the reason for the diode to eliminate this.

https://www.reichelt.de/magazin/how-to/smarte-tuerklingel/

Is there a way in ESPeasy to eliminate this by advanced event managment settings?

tonhuisman, 2020-11-27:
Well, I would expect this kind of output:
[image]

tonhuisman, 2020-11-27:
You might get better results if you set the 'De-bounce' value to ~100 msec.

tonhuisman, 2020-11-27:
For those interested (or don't understand German) [here](https://www.reichelt.com/magazin/en/build-smart-doorbell-arduino/) is the English version of that Reichelt project.

TungstenE2, 2020-11-27:
Important to notice that the linked project is working. ESP is waking up from deep sleep and phone is ringing, but there is a delay.

So I changed the setup, no deep sleep, no battery, and using D6 as switch instead of RST to wake from sleep.

Problem now is the AC frequency.

TungstenE2, 2020-11-27:
@tonhuisman I was wondering if the diode should be changed to inline from parallel like in your image

tonhuisman, 2020-11-27:
> @tonhuisman I was wondering if the diode should be changed to inline from parallel like in your image

The diode in the schematic is actually the Led inside the optocoupler...
You could add a 4-diode equalizer and elco to flatten the output to stay on while the ac is applied, but too large a capacitor will delay the release of the signal.

Did the de-bounce setting improve the result? You can up that to max 250 msec if needed.

TungstenE2, 2020-11-27:
yes, with 250 msec it is a bit better, but still slowly switching while button is pressed. But I think I can work on this.

TD-er, 2020-11-27:
You better add a small capacitor over GND & D6 in your schematic.
I think the capacitor can be quite small as there is only a very small charging current via the internal pull-up resistor in the ESP.

Maybe experiment with 100nF - 330nF
Given an internal pull up resistor of 50k and 200 nF you have an RC time of roughly 10 msec.
But since it is oscillating quite a bit it is hard for me to estimate what would be the best value for the capacitor.

tonhuisman, 2020-11-27:
That is one of the reasons why I built my doorbell system with the buttons (bell and door-open contact) connected to the ESP (with pull-ups), and a relay switching the bells (hall and attic), and also 'broadcast' the state to other ESP's (Sonoff S20's mostly) that either flash their second LED or (/and) sound a buzzer. Now that the 'broadcast' is using SendToHTTP (tcp) instead of SendTo (udp) it is quite stable (the 'door is closed' event sometimes got lost/missing using udp). Especially handy now I'm working from home and wearing a headset for music and calls most of the day 😄 (a Sonoff with a buzzer is in my workroom)

TungstenE2, 2020-11-27:
@TD-er  thx, nF = uF?

Got some capasitors here. Would try with a 220uF

TungstenE2, 2020-11-27:
@tonhuisman my door bell is 40 years old and I have to use the existing infrastructure trying to make it smart. But using the linked project worked in a way, but it was battery powered with deep sleep. This cause some delay to have the phone ringing. But every 12 weeks I had to recharge the battery. Now I have a small 200v -> 5v adapter powering the D1 mini and my plan was to have the D1 mini active all time and to connect the door button directly in order to reduce the wake up delay.

TD-er, 2020-11-27:
> @TD-er thx,
> 
> Got some capacitors here. Would try with a 0,22uF (220nF) , that should do, right?

Just try it.
I really don't know what value will be best here.
220 nF is just a rough estimate here, but it all depends on the value of the pull-up resistor used.
The internal pull up resistor is roughly 50 - 80k and for such a value the 220 nF is probably a good first attempt.

TungstenE2, 2020-11-27:
why do I get the value with 2 decimal places for a switch device?

[image]

tonhuisman, 2020-11-27:
> why do I get the value with 2 decimal places for a switch device?

That has already been fixed by TD-er, but isn't in the release yet

TD-er, 2020-11-27:
Yep, it will be fixed for new added switch plugins and for those plugins where you cannot set the number of decimals in a plugin.
In the next nightly it will be fixed and I think I will make a new build soon as I fixed quite a lot of loose ends since the last build. ;(

tonhuisman, 2023-09-02:
This seems to be solved, and can be closed.

closed 2023-09-02 by TD-er