letscontrolit/ESPEasy #3410: MQTT Import: CPU 100% if broker is down

giig1967g, 2020-12-02:
Hi,
during the upgrade of my Raspberry (that hosts Mosquitto), I just found out that if the MQTT broker is down, then the MQTT Import Plugin uses 100% of the CPU.

[image]

[image]

Is there a way to reduce CPU usage if the broker is down?
Maybe to inform the Plugin that the broker is down, so it doesn't wait for an answer?

tonhuisman, 2020-12-03:
I'm working on the MQTT Import plugin (adding value-mapping and json parsing & filtering), so I'll see what I can do here. I might need a few pointers, as I'm not really experienced in using MQTT yet.

TD-er, 2020-12-03:
This also happens when only using a MQTT controller, so I guess we should have a look at the timing interval for the MQTT timer.
This one has a variable interval (`timermqtt_interval`) which can be set at a much higher interval if there is no connection.

It is now at max 30 sec:
```c++
  if (!MQTTclient_connected) {
    // As suggested here: https://github.com/letscontrolit/ESPEasy/issues/1356
    if (timermqtt_interval < 30000) {
      timermqtt_interval += 5000;
    }
  } else {
    timermqtt_interval = 250;
  }
  Scheduler.setIntervalTimer(ESPEasy_Scheduler::IntervalTimer_e::TIMER_MQTT);
```

But maybe it should be allowed to go higher, as timeouts during connection attempt are blocking.

One other thing that may be useful to test right now is to see if it already helps to lower the controller setting for timeout.
Often MQTT broker is local, so no need to have it at a very long timeout.

TD-er, 2020-12-03:
Found a possible reason for the high load...
`processMQTTdelayQueue()` does not check if there is a connection to the broker, so it does prepare all data for publishing it over and over again.

TD-er, 2020-12-03:
Can you test this test build?
- [ESPEasy_ESP32_mega-20201130-26-PR_3411.zip](https://www.dropbox.com/s/j86vrk29qb3qh5v/ESPEasy_ESP32_mega-20201130-26-PR_3411.zip?dl=0)
- [ESPEasy_ESP82xx_mega-20201130-26-PR_3411.zip](https://www.dropbox.com/s/5wutz1ridsauwg2/ESPEasy_ESP82xx_mega-20201130-26-PR_3411.zip?dl=0)

giig1967g, 2020-12-03:
With the new version, the unit still responds slowIy, CPU usage 100%
1 per second loop is slightly better from previous version, but stll high.

[image]

TD-er, 2020-12-03:
OK, thanks for testing... will dive into it.

TD-er, 2020-12-04:
> I'm working on the MQTT Import plugin (adding value-mapping and json parsing & filtering), so I'll see what I can do here. I might need a few pointers, as I'm not really experienced in using MQTT yet.

Just FYI:
I'm now looking into the P037_MQTTimport plugin and I just decided to go for the needed change that's bugging me for a long time with this plugin.
It is creating its own MQTT client for just the import plugin while still using the same credentials as the MQTT controller that has to be active anyway.
So I will change this plugin to no longer have its own MQTT client (taking way over a kB of RAM) and start sharing the existing MQTT client.
I will try not to touch the part of the PLUGIN_IMPORT, which is probably where you're poking around.

tonhuisman, 2020-12-04:
> I will try not to touch the part of the PLUGIN_IMPORT, which is probably where you're poking around.

Ok, but I've already added `P037_data_struc` stuff like other plugins, so maybe you can go easy on that?

I have also re-done WEBFORM_LOAD and WEBFORM_SAVE (moved most of it to the plugin struct)

TD-er, 2020-12-04:
Not sure if that is needed anymore.
I will create a commit so you can see what's changed.
The import plugin has been stripped of a lot of code.

TD-er, 2020-12-04:
Just pushed a commit to this PR for the changes in that plugin.
It is not yet tested, just "it compiles".

TD-er, 2020-12-04:
As you may see, there are no longer task specific variables needed.
Or do you store the decoded topics in a P037_data_struct?
If so, then I guess it is still relatively easy to merge as I hardly touched those parts of the code.

tonhuisman, 2020-12-04:
Looked at the code quickly, looking just fine I must say, but it is best if I wait for that PR to be merged before I paste in my changes. 👍

TD-er, 2020-12-04:
Yep I think so.
Before I merge it, I will add a slight modification as it is now receiving all published messages from the broker, which may already be subscribed to a topic, so you are flooded with error messages.

TD-er, 2020-12-04:
Test build:
- [ESPEasy_ESP32_mega-20201130-36-PR_3411.zip](https://www.dropbox.com/s/nq323zxwh9r58ub/ESPEasy_ESP32_mega-20201130-36-PR_3411.zip?dl=0)
- [ESPEasy_ESP82xx_mega-20201130-36-PR_3411.zip](https://www.dropbox.com/s/tm0woy845x4g2fi/ESPEasy_ESP82xx_mega-20201130-36-PR_3411.zip?dl=0)

There are still loose ends, but would be nice to know if the high CPU load is resolved by this.

giig1967g, 2020-12-04:
this PR solves the issue:
[image]

TD-er, 2020-12-04:
Thanks for testing :)

TD-er, 2020-12-04:
> Looked at the code quickly, looking just fine I must say, but it is best if I wait for that PR to be merged before I paste in my changes. 👍

Your turn :)

closed 2020-12-04 by TD-er

giig1967g, 2020-12-05:
Hi @TD-er 
I just noticed a side effect of your latest test-build:
System variables %isntp% and %ismqttimp% are always 0 despite the fact that NPT is fine and MQTT is connected.
 
[image]

TD-er, 2020-12-05:
%ismqttimp% is indeed something I can confirm.
But %isntp% is something I cannot see why that may have been changed by this.
Maybe it was broken by another change?

giig1967g, 2020-12-05:
I looked more deeply, and also in 20201130 there is the %isntp% problem:

[image]

TD-er, 2020-12-06:
Found the bug in the `%isntp%`
```c++
case ISNTP:             value = String(statusNTPInitialized ? 0 : 1); break;
```

@tonhuisman I will add a fix for the `%ismqtt%`, in such a way you can easlily copy it into your work.
Or did you already fix that part?

tonhuisman, 2020-12-06:
> @tonhuisman I will add a fix for the `%ismqttimp%`, in such a way you can easlily copy it into your work.
> Or did you already fix that part?

I'm still working on the main features, and haven't merged it in yet, though I had that planned for today. I can postpone that until you merge these fixes I guess, lots of stuff to test here 😆

TD-er, 2020-12-06:
It has been tested and merged :)

norstbox, 2022-08-07:
> This also happens when only using a MQTT controller
@TD-er is this PR https://github.com/letscontrolit/ESPEasy/pull/3411 was supposed to fix MQTT controller along with MQTT Import?
With "Home Assistant (openHAB) MQTT" controller I see 100% Load if broker is down, on the latest release mega-20220616.

TD-er, 2022-08-07:
Can you test also using one of the latest GitHub Action builds, [like this one](https://github.com/letscontrolit/ESPEasy/actions/runs/2809509699)

There have been a lot of other fixes lately (still waiting on some confirmations to make a new build) and detecting connected state is one of them.
The thing merged before that should reduce the 100% CPU load was related to increasing retry intervals when a reconnect was failing.
But it will still lead to high CPU load if the ESP is constantly disconnecting and reconnecting to the WiFi network.

N.B. the CPU load can also appear higher than it actually is if you have "ECO" mode enabled.

norstbox, 2022-08-07:
> Can you test also using one of the latest GitHub Action builds

Just tested on mega_91c2719 from your link, still 100% CPU load. Maybe this is just "apparent" load, because UI is responsive.

It is easy to reproduce: on newly flashed firmware (default settings) add:
-any MQTT controller pointed to the offline broker
-dummy device, with sending to the above controller

> But it will still lead to high CPU load if the ESP is constantly disconnecting and reconnecting to the WiFi network.
In the scenario with 100% Load described above, connection to WiFi is stable

> N.B. the CPU load can also appear higher than it actually is if you have "ECO" mode enabled.
ECO mode disabled (default)

PR #3411 (2020-12-04): [MQTT] Prevent high CPU load when not connected to MQTT broker (#3410)
Fixes: #3410
commit 14492786 (2020-12-04): [MQTT] Prevent high CPU load when not connected to MQTT broker (#3410)

Fixes: #3410