letscontrolit/ESPEasy #3414: Shelly 1  add addon for 18B20+Power meter of 1PM

KnatteAnka, 2020-12-03:
Is it possible to add 18B20 to possible plugins for Shelly 1?
because you can connect them by the Temperature addon
https://shop.shelly.cloud/temperature-sensor-addon-for-shelly-1-1pm-wifi-smart-home-automation#312
also know if its possible to add the powermeter circuit in PM1?

tonhuisman, 2020-12-03:
I installed the `hard_Shelly_PLUG_S_2M256` .bin on my Shelly 1PM, and it should work just fine on a Shelly 1, AFAICS.
This bin doesn't have the DS18B20 plugin though, but maybe we could add that to the definition for this build. That way we could possibly eliminate the specific Shelly 1 build to save some space in the release files, one size fits all in this case 😄 
A rename to `hard_Shelly_ALL_2M256` would then be appropriate, as they all have the same memory layout, allowing easy updates via OTA because of the 2MB Flash.

TD-er, 2020-12-03:
For that add-on module you also need this PR: https://github.com/letscontrolit/ESPEasy/pull/3408
As soon as it is merged, I will also add it to the Shelly build.

CurlyMoo, 2020-12-03:
@KnatteAnka Can you test this build i made based on my PR.

[ESP_Easy_mega_20201203_hard_Shelly_1_2M256.bin.gz](https://github.com/letscontrolit/ESPEasy/files/5639067/ESP_Easy_mega_20201203_hard_Shelly_1_2M256.bin.gz)

KnatteAnka, 2020-12-04:
dont have a Temperature addon at home yeat waiting on one from blackfriday will tell when i do have one. but i do have spare Wemoses and 18b20 at home if any other tests is needed

TD-er, 2020-12-04:
The Shelly module for the Dallas sensor uses an isolator circuit (for safety reasons).
Therefore the Dallas plugin needed some changes which are included in the build made by @CurlyMoo 
So I suggest to wait for the official one and don't connect a Dallas sensor directly to the Shelly as those isolators are there for a reason.

KnatteAnka, 2020-12-04:
That i know :) but maybe other dont but thougt if needed to test if liberary still work with one wire and not trought the isolation.

TD-er, 2021-01-08:
Is implemented

closed 2021-01-08 by TD-er