letscontrolit/ESPEasy #3415: GOSUND P1

fly74, 2020-12-03:
Hi there I would give https://templates.blakadder.com/gosund_P1.html a try with espeasy.

I tried ESP_Easy_mega_20201130_hard_other_POW_ESP8285_1M but there is no analog plugin for the relais switches. Is there combination with ESP8285 SW/Analog/CSE7766 ?

It OTA with 1M not possible anymore - 2wayflash?

Firmware
--
Build:⋄ | 20111 - Mega
System Libraries:⋄ | ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ |  
Plugin Count:⋄ | 7 [Minimal 1M OTA]
Build Origin: | Travis
Build Time:⋄ | Nov 30 2020 00:31:30
Binary Filename:⋄ | ESP_Easy_mega_20201130_hard_other_POW_ESP8285_1M
Build Platform:⋄ | Linux-4.19.104-microsoft-standard-x86_64-with-glibc2.29
Git HEAD:⋄ | mega-20201130_f4f960c
System Status
Syslog Log Level: | None
Serial Log Level: | None
Web Log Level: | None
ESP Board
ESP Chip ID: | 15839117 (0xF1AF8D)
ESP Chip Frequency: | 80 MHz
ESP Chip Model: | ESP8266
ESP Chip Cores: | 1
ESP Board Name: | PLATFORMIO_ESP01_1M
Storage
Flash Chip ID: | Vendor: 0x51 Device: 0x4014
Flash Chip Real Size: | 1024 kB
Flash IDE Size: | 1024 kB
Flash IDE Speed: | 40 MHz
Flash IDE Mode: | DOUT
Flash Writes: | 0 daily / 0 boot
Sketch Size: | 636 kB (236 kB free)
Max. OTA Sketch Size: | 599 kB (614384 bytes)
OTA possible: | false
OTA 2-step Needed: | true
SPIFFS Size: | 110 kB (28 kB free)
Page size: | 256
Block size: | 4096
Number of blocks: | 27
Maximum open files: | 5
Maximum path length: | 32

TD-er, 2020-12-03:
1M modules are still a problem as you may have seen.
This build you show is 36k too large for 2-step OTA.

fly74, 2020-12-04:
But I think that's a little probem, since no 1M builds for simple 1M plus-switches are included with OTA. I liked it.

TD-er, 2020-12-04:
I know and I am trying to get the build sizes down.
Another thing that's bugging me is that the core 2.4.2 builds no longer work now with PlatformIO 5 and I honestly have no idea why.

My last option is to start using compression on the OTA bin files.
But that requires some more changes as we need to keep the current settings stored somewhere and I cannot include a full settings reader into the 2-step OTA image as it then gets too big.
Good thing is that I will then also keep the host name during OTA, which was also mentioned in an earlier issue by someone ;)

fly74, 2020-12-04:
:) If you need a tester, you can address me.

fly74, 2020-12-27:
ESP_Easy_mega_20201227_minimal_core_274_sdk3_ESP8266_1M_OTA_FHEM_HA can handle it.

fly74, 2021-01-03:
it uses AD for the plug input buttons https://templates.blakadder.com/gosund_P1.html can we include a AD plugin without exceeding the 1M OTA Limit ?

closed 2021-04-05 by fly74