letscontrolit/ESPEasy #3416: Repelling issue template

adamdyga, 2020-12-05:
Honestly I wanted to report an issue with MQTT connectivity, but once I saw the template I have to fill, I decided to report this one:

The issue template is like 1000% too long. 
Explanation: your user are your testers. If you expect them to fill a 68-lines long template - just think how manu of them will just give it up? And I guess they all want the project to  make better.

tonhuisman, 2020-12-05:
>  I wanted to report an issue with MQTT connectivity, 

Well, then you must be really pleased that a couple of hours before you wrote this, a fix for MQTT connectivity has been incorporated. 😄

TD-er, 2020-12-05:
The template was once based on numerous questions we had to ask over and over again and somewhat based on what is included in the template of other repositories like the esp8266/Arduino.

But I guess since hardly anyone fills in the template, it may be too long I guess.
Do you have any suggestions of things that are complete overkill in the template?

tonhuisman, 2023-09-02:
No more responses, so can be closed.

closed 2023-09-02 by TD-er

TD-er, 2023-09-02:
Fixed in: https://github.com/letscontrolit/ESPEasy/pull/4782