letscontrolit/ESPEasy #3419: Why show decimals for switch output?

malwikl, 2020-12-07:
Firmware
Build:⋄ | 20111 - Mega
System Libraries:⋄ | ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ |  
Plugin Count:⋄ | 46 [Normal]
Build Origin: | Travis
Build Time:⋄ | Nov 25 2020 10:16:12
Binary Filename:⋄ | ESP_Easy_mega_20201125_normal_ESP8266_4M1M
Build Platform:⋄ | Linux-4.19.104-microsoft-standard-x86_64-with-glibc2.29
Git HEAD:⋄ | mega-20201125_5e64afa
 

### Summarize of the problem/feature request
I use two switch output connected to my ESPeasy, after updating to the latest release (see above), the state of the output is not longer 0 or 1, but 0.00 or 1.00. I cannot change this in the configuration. Could you please advice is this behaviour is intended?

[image]

This makes additional remapping of values in my controller necessary (not undoable, but unless this  behaviour is really intended, I would like to avoid it).

### Expected behavior

Statue should be either 0 or 1

### Actual behavior

State is either 0.00 or 1.00

TD-er, 2020-12-07:
This is already been fixed in the latest build.

malwikl, 2020-12-09:
can confirm this. After updating to the latest version it is back as expected.

Thanks a lot!

closed 2020-12-09 by malwikl