letscontrolit/ESPEasy #3422: Max7219 led matrix

Rixtyan, 2020-12-08:
Hi. I'm just try to make clock with max dot matrix display. But occasionally espeasy doesn't support it, however 7dot display module try to show something.

Besides i can't build any firmware, every time i dont have a lot of plugins, so i even cant try to do it.

So i ask: if somebody got this plugin - pls share it, maybe we could finally move it to production.

tonhuisman, 2021-01-01:
Max7219 is supported by the `Display - 7-segment display` plugin, that's included in the `normal` and `test` builds (for at least 2 years), and supports to display as a clock in either 12h or 24h format, so I'm not sure what your request actually is...
Have you tried a recent build of ESPEasy (download from [here](https://github.com/letscontrolit/ESPEasy/releases))?

Grovkillen, 2021-01-01:
So the request is to add support for the dot matrix display. Valid request.

tonhuisman, 2021-01-01:
Aha, 'great' to have 2 different hardware items use the same type# 😠

Grovkillen, 2021-01-01:
Or expand the plugin.

tonhuisman, 2021-01-02:
> Or expand the plugin.

I've ordered such Max7219 matrix display, so I'll have a look at it once it arrives (ordered from Aliexpress so may take a few weeks).

Rixtyan, 2021-01-04:
I've tried to code some plugin, as source i used p073 plugin. It works with max7219 and even dot displays, but only as test. Actually problem is using more than 1 display (vertical/horisontal).

I'm trying to code something, but its a little bit complicated, code, i've been copied from GyverString for example, needs to be updated for espeasy. And my c++ skill (one week after work) needs time for this)

Grovkillen, 2021-01-05:
We don't have support for the dot matrix yet. So your max7219 is turning on what it believe is a 7-segment LED.

You need to wait for it to be released.

doc221, 2021-01-05:
> We don't have support for the dot matrix yet. So your max7219 is turning on what it believe is a 7-segment LED.
> 
> You need to wait for it to be released.

That makes sense. I see now they are not the same. 
Thanks.

Epik77, 2021-01-27:
> You need to wait for it to be released.

do you know if this plugin has been implemented and when it will be present in the beta version? 
Thanks.

tonhuisman, 2021-01-27:
Not yet, I have received the hardware, but haven't started working in it yet, though am planning to start soon.

Epik77, 2021-01-27:
I wanted to make a design for a parking sensor in my garage with SR-04 and merge multiple matrix displays.
If you prepare the plugins I can help you test it
thank you very much
I am very grateful to your community, I find this software very useful for the realization of my projects :-)

JojoS62, 2021-01-29:
for the MAX7219, there is great library MD_Parola, using MD_MAX72XX. It has a lot of display options and scrolling features. But it makes more sense to create a new plugin for matrix displays. I have used this lib in an ESP8266 project and it works fine, also long-term in an app running 24*7.
https://github.com/MajicDesigns/MD_Parola
There are also some different matrix modules, the MD libs supports different types. And it seems that there are already many counterfeits of the MAX7219, I have ordered three times somes modules and all were different (MAX and also the LED matrix).
Hope this helps.

tonhuisman, 2021-01-29:
> for the MAX7219, there is a great library MD_Parola, using MD_MAX72XX.

That is the library I'm intending to use, as it is a) quite popular, and b) in active development.

Thanks for reporting positive results with it 👍

Rixtyan, 2021-03-13:
Hi guys.

I've tried to do this f..in plugin few times with p073 sources, with max72xx panel sources (GyverString) and even with parola sources. Last time was a little bit successfull - i can build my version of espeasy firmware without any error. But it isnt working on my nodemcu. Espeasy can not initialize my plugin and i have no idea why. So i gave up, too difficult for me.

sources and libs:

https://github.com/Rixtyan/p248_espeasy_dotmatrix

I've a little bit changed MD_Parola lib, because Atom couldn't compile it. I've removed two strings:

ON = 1
OFF = 0

Plugin has been copied as example of MD_Parola Double High Clock (test matrix - 4x1) and... actually i don't know if it works (

If anyone can and will help - you can do it. If i found any ideas - i'll try to make in again, but i don't promise

oh yeah, forgot to text this: firmware is too big for nodemcu with MD_Parola and MD_Max72xx libraries.

tonhuisman, 2021-05-01:
> Last time was a little bit successfull - i can build my version of espeasy firmware without any error. But it isnt working on my nodemcu. Espeasy can not initialize my plugin and i have no idea why. [...]

I've taken your stuff and got it working by adding a couple of `delay(0);` calls in the `displayZoneText()` and `displayAnimate()` members of MD_Parola to keep the watchdog fed, so the ESP doesn't restart every 30 seconds.

> I've a little bit changed MD_Parola lib, because Atom couldn't compile it. I've removed two strings:
> 
> ON = 1
> OFF = 0

I reverted that change, and renamed the enum values to MD_ON/MD_OFF to finally get rid of the compiler errors.

If you don't mind, I'm trying to pour this into a generic use plugin, with most of the options of the MD_Parola library available. I've put most of my time into the separation of settings from the plugin, and that part is already kind of working.
A sample screenshot, with 4 out of a maximum of 10 zones configurable:

[image]
(The combo boxes supply all available options in readable form. Ditto for Hardware type)

The code and memory requirements of these libraries are quite large, so I'm using, f.e., `std::vector` in trying to keep runtime memory usage to a minimum. Possibly it will need a separate build for this plugin, though currently it fits in the 'custom' builds. Haven't tried the 'display' builds yet, will do that later.

Actually displaying any text on the display I didn't get to yet... 🤔

KaiHuHu, 2021-05-16:
@tonhuisman 
great work you did there.
i am an absolute beginner on the subject.

I have an ESP8266 CP2102, JSN-SR04T and MAX7219 8x32 4 in 1 Dot Matrix LED here.
I would display the measured distance directly on the matrix LED and send it later via MQTT.

how can I use the Display - MAX7219 dot matrix [DEVELOPMEN]?

many thanks

tonhuisman, 2021-06-11:
This project has been on the back-burner for a while, I'll try to revive it during the weekend, but it is going to be very nice weather, so there may be distractions... 😎

TD-er, 2021-06-11:
> This project has been on the back-burner for a while, I'll try to revive it during the weekend, but it is going to be very nice weather, so there may be distractions... 😎

Also very "dangerous" to do it with such nice weather.
Due to the amount of sunlight, you might not see the LEDs light up and spend  a lot of time debugging something that's already working :)

tonhuisman, 2021-06-13:
> you might not see the LEDs light up

Eventually I could see the LED's light up, and display some text as configured, so I tidied up the code and made a PR. Still quite a few things to do, but it's testable, I guess. 😃

exkape, 2021-06-14:
really many thanks for your work

tonhuisman, 2021-06-14:
It still is a work in progress, and I will do many improvements soon, I expect, but any testing will be helpful.
Can you obtain the required .bin file (it is in the DISPLAY and MAX builds, currently), or should I provide it here?

exkape, 2021-06-14:
well, I'll start testing immediately.

exkape, 2021-06-14:
Hi, I think I have tested all testable.
ok the actuator part. 
Zone configuration: ok for the size and text but the working contents are text (with also the system variables) and clock (size 4) the others do not work.
ok font, layout, Special Effect and Offeset but brightness and Repeat are not working but most importantly I could not make other zones work besides the 0 (zero)
I hope to be proved helpful.
Thanks again very much for your wonderful work

tonhuisman, 2021-06-14:
> Hi, I think I have tested all testable.

Thank you for your effort.

> Zone configuration: ok for the size and text but the working contents are text (with also the system variables) and clock (size 4) the others do not work.

That is part of what's not completed yet, so: known issue.

> but brightness and Repeat are not working 

About brightness, I thought already that wasn't working as intended, but haven't looked into yet
I wrote in the PR that I have not yet implemented 'Repeat', but I wanted to get the setting in place before initial availability

> but most importantly I could not make other zones work besides the 0 (zero)

I'm investigating that now

> I hope to be proved helpful.

Feedback (any) is very helpful, thank you again.

tonhuisman, 2021-06-15:
I fixed a few bugs, so here are new builds:
[ESPEasy_display_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/6653765/ESPEasy_display_ESP32_4M316k.zip)
[ESPEasy_display_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6653767/ESPEasy_display_ESP8266_4M1M.zip)

Bugs fixed:
- Multiple zones are now working
- Brightness setting per zone is applied

Still lots of work to do, though.

Epik77, 2021-06-16:
273 / 5000
Risultati della traduzione
i tried ESPEasy_display_ESP8266_4M1M.zip
I confirm the operation with size 12 Hardware type IC Station.
I look forward to being able to integrate it with MQTT domoticz controller and maybe use a different content type bar graph (in order to integrate it with HC-04).
Thanks for support

tonhuisman, 2021-06-16:
> use a different content type bar graph

That is one of the options I've been thinking about, but I have not done any investigation on that yet. It's on my local (unshared) todo list 😉

tonhuisman, 2021-06-19:
I've implemented the Repeat timer function, and fixed quite a couple of bugs.
(Took a bit of fiddling to squeeze it in the ESP8266 DISPLAY build, but eventually it gave in.)

[ESPEasy_display_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/6681194/ESPEasy_display_ESP32_4M316k.zip)
[ESPEasy_display_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6681195/ESPEasy_display_ESP8266_4M1M.zip)

Anyone that has MAX7219 modules that have the `Generic` or `Parola` hardware configuration (I currently own only FC16 hardware, ordered some more, but don't know what hardware config that will be), can you please test if the Double-height font works as intended or needs U/D flipping or rotation to present normally?

tonhuisman, 2021-06-28:
_(Copied from the associated PR)_
Implemented the `direction` (0 = right to left and 1 = left to right) and `barType` (0 = solid, 1 = solid, single line bar, 2 = alternating dotted bar if at least 2 pixels wide, else solid) options for the Bar-graph feature.
Also fixed many bugs, mostly off-by-one computation errors (that was to be expected 😛)

A set of `DISPLAY` builds from Github Actions build service, and a local `CUSTOM` build for ESP8266 having all available options compiled in, as the ESP8266 `DISPLAY` build exceeds the .bin size if all options are enabled. (That build _does_ have the Bar-graph feature though).

[ESPEasy_display_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/6722232/ESPEasy_display_ESP32_4M316k.zip)
[ESPEasy_display_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6722233/ESPEasy_display_ESP8266_4M1M.zip)
[ESP_Easy_mega_20210627_custom_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6722236/ESP_Easy_mega_20210627_custom_ESP8266_4M1M.zip)

All based on the latest `mega` branch.

tonhuisman, 2021-06-29:
Yesterday, a couple of new dotmatrix displays where delivered by the mailman, so I was able connect them to an ESP8266 (they are all of the same hardware type 'FC16' so they can all be on the same instance). It is working quite nice, IMHO 😎 

[image]

From top to bottom display contents shown:
- Date (DD-MM) with animation
- Date (YYYY-MM-DD) with animation (scroll left, in progress)
- Clock (24h), updated every minute, no animation, no flashing colon
- Date/Time (YYYY-MM-DD HH:mm) with animation (scroll left, in progress)
- Bar-graph with 3 bars, type 2 (single line) mimicking temperatures (-14, +4, +26) on a +40 to -20 (left to right) scale and a zero-point around 1/3rd from the right (the left module reflected weirdly in the photo, but it does work)
- Text (ESPEasy is great fun!) with animation (scroll left, in progress)

Top-right is the 'ample power supply', required to keep it all running, between the ESP and the first display unit is the required level converter for 3.3V <-> 5V signal conversion.

TD-er, 2021-06-29:
Great :)
A really nice addition, this plugin.
How much current do these modules draw? (based on nr. of pixels active of course)

tonhuisman, 2021-06-30:
> How much current do these modules draw? (based on nr. of pixels active of course)

I did not yet have the time to do some measurements, but I'll do that soon.

BTW, the units I connected are, from top to bottom:
- 3x 1x4 modules (1 zone each)
- 1x 2x4 modules (2 zones)
- 1x 1x8 modules (1 zone)

And there is this kind of 'practical' issue:
The zone definitions (1..6 as show here) are actually from bottom to top, because of:
1. how the MD_Parola library works
2. the 2x4 unit modules are connected in a zig-zag way, from lower-right to top-left

This is something I'm still thinking about how to 'manage', either by clear documentation or maybe put the zones presentation in the Device configuration 'upside down' (high to low) so it will look like the actual physical setup.
Comments and suggestions on this are appreciated.

TD-er, 2021-06-30:
Why are they connected like this? Is it because text scrolling is often from right to left?
Or is there another reason?

If there is no reason why these must be connected like this, then I would suggest to rotate the "display" to overcome practical issues like changing zone numbers if you plan to add another module.

tonhuisman, 2021-06-30:
I don't know the original reason, besides the hardware being wired that way, but the entire MD_Parola library (and its base, MD_MAX72XX) are built like that, and it feels counter-intuitive to try and reverse the direction.

I'll experiment to see if it is easily reversed though. 👍

TD-er, 2021-06-30:
If you then also need to rotate function calls like "left", "right" etc. then leave it as it is.
Then I guess addressing must be based on "module nr" where you start counting from where you connect it and document it.

tonhuisman, 2021-06-30:
> How much current do these modules draw? (based on nr. of pixels active of course)

Did some measurements, using an 8-module display, though I only measured the LEDs of the first 4 modules, as it is all about an average current consumption, and the multimeter used has a max. range of 2000 mA 😉
Basic current per module is 7.5 mA, so 60 mA with all LEDs off, then it heavily depends on the brightness used:
Brightness 1: 0.8 mA/LED -> 256 LEDs = 270 mA really soft light, but still very usable
Brightness 3: 1.8mA/LED -> 256 LEDs = 530 mA the brightness used in the photo above, nice during the evening
Brightness 7: 3.6 mA/LED -> 256 LEDs = 995 mA usable brightness in normal daylight
Brightness 15: 5.5 mA/LED -> 256 LEDs = 1485 mA this is kind of a 'blinding light' brightness

tonhuisman, 2021-06-30:
> If you then also need to rotate function calls like "left", "right" etc. then leave it as it is.

I did have a go at it, but it does indeed involve reversing left and right, most likely only possible by rewriting the MD_Parola/MD_MAX72XX libraries, and that's not something I aspire.

Presenting the zones in reverted order _does_ seam usable though, I'll experiment further with that.

TD-er, 2021-06-30:
If it makes things hard to use, you can consider adding a button to insert a module at "position" N.

tonhuisman, 2021-07-03:
I added an Action column to each zone (can be disabled if .bin size gets tight) to add a zone above or below the current zone, or delete a zone, that should make it quite manageable. Also added a Zone ordering options, fixed a bunch of bugs and did some code improvements.

Downloads:
[ESP_Easy_mega_20210703_custom_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6758973/ESP_Easy_mega_20210703_custom_ESP8266_4M1M.zip)
(Display builds will be added once the Github Actions build is completed)

tonhuisman, 2021-07-08:
Fixed a few bugs (found a local tester, yay! 😄) so new builds for download:

[ESP_Easy_mega_20210708_custom_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6786852/ESP_Easy_mega_20210708_custom_ESP8266_4M1M.zip)
[ESPEasy_display_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/6786903/ESPEasy_display_ESP32_4M316k.zip)
[ESPEasy_display_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6786904/ESPEasy_display_ESP8266_4M1M.zip) (Limited fonts and options available, use Custom or ESP32 build to test all options)

tonhuisman, 2021-07-11:
Did some more fixes and improvements, and worked on documentation.

[ESP_Easy_mega_20210711_custom_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6797490/ESP_Easy_mega_20210711_custom_ESP8266_4M1M.zip)
[ESPEasy_display_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/6797494/ESPEasy_display_ESP32_4M316k.zip)
[ESPEasy_display_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6797495/ESPEasy_display_ESP8266_4M1M.zip) (Limited fonts and options available, use Custom or ESP32 build to test all options)

Exported the documentation for this plugin into a PDF, so any comments on that is also welcome.
[Display - MAX7219 dot matrix — ESP Easy 2.1-beta1 documentation.pdf](https://github.com/letscontrolit/ESPEasy/files/6797496/Display.-.MAX7219.dot.matrix.ESP.Easy.2.1-beta1.documentation.pdf)

TD-er, 2021-07-11:
So, that's how documentation has to be :)

tonhuisman, 2021-07-14:
I've done some more testing, so found a couple of bugs, like selecting a font didn't work properly, that I promptly fixed 😆
The 'Numeric, double height' font is now always included in a Custom or MAX build, even if the 'Full, double height' font is included.
Added a new `Text reverse` Content type, that reverses any text just before displaying it, making the Vertical font more usable f.e. with the Scroll left/right options.
**Attention** May have 'shifted' current Content settings after update, as the new option was inserted, and other options renumbered.

[ESP_Easy_mega_20210714_custom_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6818662/ESP_Easy_mega_20210714_custom_ESP8266_4M1M.zip)
[ESPEasy_display_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/6818729/ESPEasy_display_ESP32_4M316k.zip)
[ESPEasy_display_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6818730/ESPEasy_display_ESP8266_4M1M.zip) (Limited fonts and options available, use Custom or ESP32 build to test all options)

The updated documentation: (No links to the font mapping tables, though)
[Display - MAX7219 dot matrix — ESP Easy 2.1-beta1 documentation.pdf](https://github.com/letscontrolit/ESPEasy/files/6818735/Display.-.MAX7219.dot.matrix.ESP.Easy.2.1-beta1.documentation.pdf)

TD-er, 2021-07-14:
What is seen as a "module" ?
Those units with 4x 8x8 pixels are 4 "modules" ?
Still have to make a setup of my modules.

tonhuisman, 2021-07-14:
> What is seen as a "module" ?
> Those units with 4x 8x8 pixels are 4 "modules" ?

Yes, 1 module is a max7219 chip with an 8x8 led display.

> Still have to make a setup of my modules.

Configuration is done in zones, where each zone has 1 to 255 modules, but all zones combined max. 255 modules.
Usually, these 4 module units can be mounted exactly next to another unit, so it can form a seamless display, and by combining 2 rows of 3 4-module units, you can even configure a double-height display, using the double height fonts and alternating lower/upper layout.

Just trying it probably says more than me writing a thousand words 😉

TD-er, 2021-07-14:
Yep, I have to try it... I know ;)

tonhuisman, 2021-07-14:
The screenshot in the linked PR is an (older) example of how I set up for the photo above

TD-er, 2021-07-14:
The biggest challenge for me is to find desk space as those modules are quite big ;)

tonhuisman, 2021-08-09:
I've reworked the settings-load and -save routines to no longer require too much stack memory, so the numerous crashes that where seen before should not happen anymore.
Also a couple of bugs have been squashed.
I have a few builds available for those that want to test:

An 'official' build from the Github Actions:
[ESPEasy_display_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/6957049/ESPEasy_display_ESP32_4M316k.zip)
And 2 local builds: (GH Actions doesn't build the ESP8266 display edition, for some reason)
[ESP_Easy_mega_20210809_display_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6957057/ESP_Easy_mega_20210809_display_ESP8266_4M1M.zip) (Only basic features available because of .bin size-limit)
[ESP_Easy_mega_20210809_custom_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6957058/ESP_Easy_mega_20210809_custom_ESP8266_4M1M.zip) (Includes all available features)

tonhuisman, 2021-08-10:
Did some small improvements, like avoiding to max-out the daily Flash writes, and reduce the build size for ESP8266 Display build (to include some more features)

From the Github Actions build service:
[ESPEasy_display_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/6960506/ESPEasy_display_ESP32_4M316k.zip)
[ESPEasy_display_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6960508/ESPEasy_display_ESP8266_4M1M.zip) (Some fonts are left out, other features enabled)

And 1 local build with all options & fonts enabled:
[ESP_Easy_mega_20210810_custom_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6960514/ESP_Easy_mega_20210810_custom_ESP8266_4M1M.zip)

martcs, 2021-09-09:
i have strange thing, cant get chinese module 4x 8x8 work as one. all time showing text or clock only on individual 8x8 display. "1234" show only 12 (2 is partialy) on ewery 8x8 display not full text on all as one. use hardware type fc16.

tonhuisman, 2021-09-10:
> cant get chinese module 4x 8x8 work as one

Have you set the Size column to 4? As the issue you are describing is exactly what I've seen when that setting is set to the wrong number of modules, or when I have connected more modules then I have configured, then the output is _repeated_ from the first module after the amount I've set.

martcs, 2021-09-10:
exactly, i dont touch size, because thinking its size of fonts 🤣

tonhuisman, 2021-09-10:
No, we have only different fonts, not different font-sizes, and then that size would have been next to the Font column 😉

martcs, 2021-09-10:
can i send trought mqtt some text to this matrix display (domoticz or hamqtt)? i try mqtt import, but display dont have this option. maybe trought rules ?

tonhuisman, 2021-09-10:
Any way to send a command to the unit will let you put a message on the display, next to the default available content options per zone (1 line in the configuration).
So that can be from serial, rules, via url or a MQTT controller that handles the `/cmd` ending of a topic like HA/OpenHAB MQTT.

martcs, 2021-09-10:
my fault, all work great. i write wrong device id in topic. thanks. question, esp8266-12f is minimal requirements? little slow and sometime if activate another plugin show low memory. best choice will be esp32?

tonhuisman, 2021-09-10:
IMHO, it works as well, or sometimes even better, on an ESP8266 than on an ESP32... and usually an ESP8266 is cheaper and smaller 😃 
I have not tried to squeeze it into an ESP01 or 1MB build, because of the trouble of updating those modules, but it should run the same on these, even smaller, units, as they use the same ESP8266 chips, but the number of GPIO pins available is very limited, though it should be just enough, I think, when disabling serial.

martcs, 2021-09-10:
yep, just im reflashing few times. reason was periodic reboot and freeze (specialy when change zones). i dont now why. i disable i2c and now work nice, i use esp8266 12 4mb, and that will be only for display, seams to be ok, thanks. 👍

tonhuisman, 2021-09-11:
I've made some improvements to the documentation, fixed a few inconsistencies, did minor code improvements, and changed the title of the 'Size' column to 'Modules' for clarity.

Fresh builds, updated to the current master branch:
[ESPEasy_display_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/7147830/ESPEasy_display_ESP32_4M316k.zip)
[ESPEasy_display_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/7147831/ESPEasy_display_ESP8266_4M1M.zip)

Updated .pdf export of the documentation:
[Display - MAX7219 dot matrix — ESP Easy 2.1-beta1 documentation.pdf](https://github.com/letscontrolit/ESPEasy/files/7147832/Display.-.MAX7219.dot.matrix.ESP.Easy.2.1-beta1.documentation.pdf)

[13 more replies]

PR #3688 (2021-10-19): [P104] New plugin: MAX7219 dot matrix display
Adds support for the MAX7219 (& MAX7221) 8x8 dot matrix display modules (SPI connected), available in 1, 2, 4, 8, 2x2, 2x4 & 2x8 module configurations, that can be daisy chained up to 255 units.

Configuration possible up to 8 (ESP8266) or 16 (ESP32) zones, that can each display independent texts/content.

Using a modified version (to allow proper working with ESPEasy's watchdog) of MajicDesign's [MD_Parola](https://github.com/MajicDesigns/MD_Parola) & [MD_MAX72XX](https://github.com/MajicDesigns/MD_MAX72XX) libraries, supporting all available effe [... cut]
closed 2021-10-19 by TD-er