letscontrolit/ESPEasy #3423: Compile OTA source from today with PlatformIO ends in error

Micha-he, 2020-12-10:
If I compile the source from today with my PlatformIO-IDE, the compile of the OTA versions (e.g. minimal_core_242_ESP8285_1M_OTA) ends with the following error:

`Tool Manager: Installing framework-arduinoespressif8266 @ 2.20402.4-puya
Tool Manager: Warning! More than one package has been found by framework-arduinoespressif8266 @ 2.20402.4-puya requirements:
 - platformio/framework-arduinoespressif8266 @ 3.20704.0
 - tasmota/framework-arduinoespressif8266 @ 2.7.4.9
Tool Manager: Please specify detailed REQUIREMENTS using package owner and version (showed above) to avoid name conflicts
Error: Could not find the package with 'framework-arduinoespressif8266 @ 2.20402.4-puya' requirements for your system 'windows_amd64'
Der Terminalprozess "C:\Users\Micha\.platformio\penv\Scripts\pio.exe 'run', '--environment', 'minimal_core_242_ESP8285_1M_OTA'" wurde mit folgendem Exitcode beendet: 1.`

The compile-process from the normal 1M and normal 4M1M works fine.

Any idea what I can do ?

TD-er, 2020-12-10:
Nope, core 2.4.2 cannot be compiled since PIO 5.
I have no idea how to convince PIO to allow the same platform code version 2.4.x we've been using for ages.
Since the Puya patch was not backported to the official 2.4.x code, we were using a version I made myself.

Micha-he, 2020-12-10:
Ok, then I'll have to wait for a working version...

closed 2020-12-10 by Micha-he