letscontrolit/ESPEasy #3425: PZEM-004T multiple works very well

ipa64, 2020-12-12:
Hi,
Not an issue, just to say THANK YOU !
PZEM-004T multi works very well, I think you can remove [TESTING].
[image]

TD-er, 2020-12-12:
Good to know.
Right now the "testing" is also misused to determine what build contains which plugins.

closed 2020-12-12 by ipa64

Shumde, 2021-06-07:
Hi ipa64, I have 2 PZEM-004t and configured in ESPEasy as SoftwareSerial. But all of them shows the same results.
If coil #1 is connected and #2 not, both PZEM shows the same results. How did you configure ESPEasy? Adress, GPIO?

Shumde

uzi18, 2021-06-07:
@Shumde you should connect them one by one and change address in first task configuration.

TD-er, 2021-06-07:
Yep, modbus devices must have their own unique address if used with >1 on the same bus.

hunterelectros, 2023-12-12:
@TD-er , I have a similar problem, all three PZEMs are connected via SW-serial but each to their own different pins, and still they all show the same data. In this case, is it also necessary to change the addresses for the PZEMs so that each of them has its own unique address?
[image]