letscontrolit/ESPEasy #3434: PWM fade down problem

DominikBegere, 2020-12-21:
When I fade a GPIO with PWM up (PWM,0,800,1000) everything works great.  
If I fade a GPIO with PWM down (PWM,0,10,1000). It does not work. The pin goes up maybe 5 times and then it goes to the new value but that is not a fade.  
I also tried different GPIO but the result was the same.

Hardware: Wemos D1 mini
Firmeware ESP-Easy_mega_20201130_normal_ESP8266_1m.bin

tonhuisman, 2020-12-21:
This would have better fitted on the forum, I think, as it needs a little investigation.

Can you share the rule that is using this?
Do you have timers running at the same time the pwm command runs?

DominikBegere, 2020-12-22:
I tried it at the beginning with rules. But that did not work either. Then I reset easyesp to factory settings and tried it only with the http command. http://192.168.1.184/control?cmd=PWM,14,100,1000  But the result was the same. Also with the internal led gpio 2 was the problem only that the led is switched against vcc. PWM 0 = led is on.

henkiejan1, 2020-12-22:
Ooh, that also the problem i have with a ESP12E on a ceiling light. Fade on works perfect. But fade down gives a strobe effect. Strange enough does it work ok with a ESP32 with the same light. I use the PWM pins of the ESP12E.

Grovkillen, 2020-12-22:
PWM on the ESP32 is most likely hardware based but on the ESP-12 software
based.

TD-er, 2020-12-22:
Yep it is hardware based on ESP32, but I don't use the hardware fade function.
So I guess it may just be something strange in the step computation which causes the pulse generation to stop on ESP8266

TD-er, 2020-12-24:
I just looked at the code handling this and to my knowledge the code to compute the steps hasn't changed for a long time.
However I do see a number of issues which may be related to signed/unsigned int types and I really wonder why it is working on ESP32.
Maybe the ESP32 does apply some bitmask to only use the N least significant bits where the ESP82xx implementation does maybe only apply the value if it is within range of usable duty cycle.

Anyway, I will make a fix for it.

TD-er, 2020-12-24:
Could you test this test build?
I'm not near an ESP device to test right now.

- [ESPEasy_ESP32_mega-20201130-277-PR_3433.zip](https://www.dropbox.com/s/ixisn0uf510u77u/ESPEasy_ESP32_mega-20201130-277-PR_3433.zip?dl=0)
- [ESPEasy_ESP82xx_mega-20201130-277-PR_3433.zip](https://www.dropbox.com/s/hslr5t9riqifq7v/ESPEasy_ESP82xx_mega-20201130-277-PR_3433.zip?dl=0)

henkiejan1, 2020-12-25:
> Could you test this test build?
> I'm not near an ESP device to test right now.
>
>     * [...]

I will try! :) And post the results!

TD-er, 2020-12-25:
Well I finally got to a PC with an ESP and logic analyser and I can tell you it isn't working as it should.
Fade up works fine as you said, but fade down isn't....
[image]
It even turns to this strange 31 Hz signal, which isn't specified.

Tomorrow evening I can have another look.

TD-er, 2020-12-26:
OK, strike that....
The file wasn't actually flashed to the ESP node, and when it was it was behaving perfectly :)
So I guess I'll happily wait your findings too, but now I know I can go to sleep.

closed 2020-12-27 by TD-er

henkiejan1, 2020-12-27:
Wow you are quick  TD-er!! But it works indeed perfect as before. On the ESP32 as on the ESP12E. Both with the LSC ceiling light from Action. No strobe when i just wake up and put the light off. ;)

TD-er, 2020-12-27:
Thanks for reporting back :)

PR #3433 (2020-12-27): [Cleanup] Reduce build size to allow minimal_OTA builds for 1M
It links large blobs of data to online services like the Github CDN servers.
The CSS and JS files stored there will be served with a TTL of 1 year, so they should remain in the browser cache even if you're not connected to the internet, but once were.

Also added some more optimizations to finally make the minimal OTA builds again small enough for 2-step OTA updates.

Fixes: #3434
commit c343c17e (2020-12-27): [PWM] Fix fade down (#3434)

Fixes: #3434