letscontrolit/ESPEasy #3436: Lack of support for AHT1x sensor

dragonfly69, 2020-12-22:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

dragonfly69, 2020-12-22:
Hi, 
I've just get some bunch of sensor and unfortunately noticed that one of them is not supported.
I'm building some project for my rekuperator and need a lot of sensors - so looking for one with different addresses.

TD-er, 2020-12-22:
For I2C sensors which have an address conflict, you can use an I2C multiplexer.
Those are supported by ESPEasy.
Using such a multiplexer gives you essentially a separate bus per port on the multiplexer, so you can have sensors with the same address on another I2C channel of such a multiplexer.

See the [documentation](https://espeasy.readthedocs.io/en/latest/Hardware/Hardware.html?highlight=multiplexer#i2c-multiplexer)

tonhuisman, 2020-12-22:
> project for my rekuperator and need a lot of sensors

What do you define 'a lot', 5, 15, 50?
ESPEasy on ESP8266 supports 12 devices out of the box, on an ESP32 unit, 32 devices are supported.

Adafruit has an Arduino compatible library for the AHT10 and AHT20, and those devices do support 2 I2C addresses (0x38 & 0x39), so please feel invited to create a plugin and make a PR for that 😄

dragonfly69, 2020-12-22:
Currently using 2xBME280, 1xSi7021, 1xDHT22,  4xDS18b20 and cheap SharpDust
sensor.
Still need two or 3 more and just because get AHT10 before already ordered
multiplexers just asked about support for it.
I understood that no - so I'll wait for ordered multiplexer and will handle
it,

Thx for quick response,
Regards
Rafal

TD-er, 2020-12-22:
The recent changes for the Dallas sensor may also be useful for you.
I recently added support for having 4 Dallas sensors in a single task.
One extra nice feature this offers is that the sensors in a single task will all be read at the same time.

As an example a test setup on my breadboard:
[image]

[image]

dragonfly69, 2020-12-23:
Thanks,
I'll check them out as today it appears that my DHT22 sensor just died :(.
It was meant to measure temp/humidity of used air removed from home which
has humidity very close to 100% and very low temp in the same time ca 3-6
degree.
I'll try to use a weatherproof Dallas sensor as I already know how, from a
thermodynamic point of view, heat exchange in Reku looks like.

Thx again and best wishes for Christmas and New Year,
Rafal

tonhuisman, 2021-08-01:
@dragonfly69 Do you have any AHT sensor(s) and can you please test the new plugin I migrated from the Playground to the main ESPEasy repo? I can provide a .bin file from the Github Actions build if you want, just name the current build you are using.

dragonfly69, 2021-08-02:
Hi Ton,
I have some AHT sensors and some spare Wemos D1 mini boards.  If you
provide me with ESP_Easy_mega_XXXXXX_normal_ESP8266_4M1M.bin I'll be
happy to test it.
Thx
Rafal

tonhuisman, 2021-08-02:
Hi Rafal,
Thank you for your support in this! 👍 

Please verify this, if possible for all AHT sensor models supported (AHT10/AHT15, AHT20, AHT21) that you have available:
- Do they keep working for prolonged time? (> 1 hour, possibly > 1 day) without `AHTnn : reset` messages in the log, indicating time-outs)
- Do they work combined with other I2C devices, also for longer time?
- An indication of the accuracy, taking into account the average tolerance is ~2%

[ESPEasy_normal_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6915531/ESPEasy_normal_ESP8266_4M1M.zip)

Regards, Ton.

dragonfly69, 2021-08-02:
Hi Ton,
I did flash my ESP with attached software but still do not have AHT sensors
on the device list.
As for now I only have a few AHT10 sensors ( as none of the software
supports them currently ) but if it will work with them I may buy some
others.
I may use whatever version you have for ESP8266 - I will build some test
environment to check it out :).

Thx
Rafal

tonhuisman, 2021-08-02:
I just tried with above zip file, and it does have the `Environment - AHT10/20/21 [TESTING]` plugin. I changed it from normal to test_A about an hour ago, so maybe you had the previous normal build that doesn't have the plugin included?

dragonfly69, 2021-08-02:
Will check and let you know.
Thx Rafał

tonhuisman, 2021-08-02:
You can see the current build name and date in the footer of the web interface (and on the Tools/Info page)
[image]

dragonfly69, 2021-08-02:
Hi Ton,
The build test_A contain AHTXX sensor on the device list but .. when I;ve scan I2C bus it do not recognize sensor - see below image
[image]
So in the log i see continuslly below error:
Error  : AHT10 : unable to initialize
, and no readings on device page :(

Normal build you've sent me hasn't got it on device list but during scaning I2C buss it was correctly recognized as AHT10,
Ther's something missing in this build. 

Thx
Rafal

dragonfly69, 2021-08-02:
It started to work ?? :) 
I'll will make some summary tomorrow,

Thx
Rafal

TD-er, 2021-08-02:
It is a "testing" build, which doesn't include the descriptions for I2C scanned devices.
So that's the difference between "normal" and "testing" builds

dragonfly69, 2021-08-03:
Hi Ton,
I'm runing this test since  yesterday 5 pm CET, and:
- all of sensors (AHT10/SHT30/BH1750) are working fine without any erros,
- no issue with combining with other I2C devices,
- I would say that both sensor accuracy is very good - tolerance ~1% for both sensors // AHT10 better than SHT30,
Below charts from all of connected sensors:
[image]

My environment is still running so I may take a look for a longer time if you need ?

Plz. consider to move both AHTx/SHT3x to normal build as both seems to work OK.

Thx
Rafal

tonhuisman, 2021-08-03:
Thank you very much for the detailed report 👍 
I'm really happy it all works so very well!

These plugins probably won't be moved to the 'normal' builds, as over time the labels, like normal, test, etc., no longer say much about the quality or stability of the plugin, but is now only a way of grouping them. It is no longer possible to combine all plugins in a single ESP8266 or ESP32 4MB build. All plugins (that work on ESP32, ~98%) _are_ available in the ESP32 MAX builds, but that requires an ESP32 with 16MB flash memory. (Unless you need to store many/large files, f.e. using the Cache controller, the 16M1M configuration is the suggested choice.)

Next best thing will be the removal of the [TESTING] moniker from the name, but that requires some more, independent, positive implementation/usage reports.

dragonfly69, 2021-08-03:
Fully understood.
I'll continue observation of both sensor stability - and if I notice any
abnormal behavior will let you know.

Thx
Rafal

PR #3724 (2021-09-05): [P105] Move AHT10/20/21 plugin from Playground
Move plugin for AHT10/20/21 I2C environment sensor from the Playground.

- Some adjustments to the code, reformatted using Uncrustify
- Re-used Plugin ID P105
- Documentation

TODO:
- [x] Complete documentation
- [ ] Testing (to be reported)
  - [x] AHT10 (confirmed by ghtester, and dragonfly69)
  - [ ] AHT20
  - [x] AHT21

Resolves #3604 
Resolves #3466
Resolves #3436 
Resolves #3550
closed 2021-09-05 by TD-er