letscontrolit/ESPEasy #3438: Error when sending BME680 data emoncms : Unknown sensortype or too many sensor values

fluppie, 2020-12-23:
https://github.com/letscontrolit/ESPEasy/blob/c530b082f59291b67869856a5c36470b76c857c2/src/_C007.ino#L62

Any idea why the max values per task is set to 3? I see no reason to send more data to Emoncms?
https://emoncms.org/site/api#input

In my case a BME680 sends out 4 values, so maybe we can change the code to 5?
`        if (valueCount == 0 || valueCount > 5) {`

TD-er, 2020-12-23:
I have no idea why there is a limit on the Emon CMS controller.
I think I messed up once in the valuecount check.
I guess it should be something like this:

```c++
if (valueCount > VARS_PER_TASK) {
```

fluppie, 2020-12-23:
Jup, that's working. Modified the code and did a test build :-).
[image]

TD-er, 2020-12-23:
Great, I will make a PR for it :)

TD-er, 2020-12-23:
By the way, by looking at  the code, I noticed another bug for C007 , which I will fix along the way.

commit 334893ee (2020-12-27): [EmonCMS]  Allow sending 4 values per task (#3438)

Fixes: #3438

Also stores the values in the event queue element to make sure not only the last value of a task is being sent if sending values got delayed somehow.
closed 2020-12-27 by TD-er