letscontrolit/ESPEasy #3444: Shelly1 does not report status correctly

Bruka, 2020-12-29:
MQTT status from shelly1 looks like this when I send shelly1/cmd/gpio,4.0 Topic answer: "shelly1/% tskname%/% valname%" 
tskname and valname? Should be "shelly1/status"

Then there is the difference in the answer as well,
message:
{
"log": "port # 4: set to 0", (formerly: "log": "GPIO 4 Set to 0")
"plugin": 1,
"pin": 4,
"mode": "output",
"state": 0
}

Then gpio5 is hard-coded, which means that it is not possible to set the button for double-clicking, but maybe there is another version that works for Shelly1 2mb memory with OTA?

tonhuisman, 2020-12-29:
AFAICS, GPIO-5 isn't used on a Shelly 1 (well, on my Shelly 1PM's it is used for the PM part).
Also, I don't think any GPIO is hardcoded in the software, all needs to be setup on the Devices page.

Indeed in a recent release the log line in the message has changed, and also the superfluous `Ok` at the end is no longer there (as it was invalid, and undesired, for the json output).
If you really want to know the pin and state you should check those attributes, better not try to parse the log text.

Bruka, 2020-12-29:
Look at this picture on Shellys site:  https://shelly.cloud/wp-content/uploads/2020/06/shelly1_pinout.jpg
I use ESP_Easy_mega_20201227_hard_Shelly_1_2M256 and there it is not possible to select gpio5 on the Devices page.

ironataerial, 2020-12-29:
> Look at this picture on Shellys site: https://shelly.cloud/wp-content/uploads/2020/06/shelly1_pinout.jpg
> I use ESP_Easy_mega_20201227_hard_Shelly_1_2M256 and there it is not possible to select gpio5 on the Devices page.

GPIOs 4 and 5 on ESPEasy are by default assigned to I2C. Once you can release them from I2C in the Hardware page then will show up on the Devices page.

Bruka, 2020-12-29:
Ahh, i2c was the problem, thanks.
But this remains: Topic answer: "shelly1/% tskname%/% valname%"

tonhuisman, 2020-12-29:
> But this remains: Topic answer: "shelly1/% tskname%/% valname%"

You may need to check what is setup at your MQTT Controller on the controllers tab. Those spaces after the `%` should not be there.

Bruka, 2020-12-29:
Sorry but the space, it was my fault when i copy.
I have double checked the settings for my controller.
It looks like this in my viewer:
[image]

tonhuisman, 2020-12-29:
What MQTT Controller have you configured?

Bruka, 2020-12-29:
Homeassistant (Openhab) MQTT.

tonhuisman, 2020-12-29:
Looked at the sourcecode of that controller, but there are no obvious flaws visible (which would have been highly unlikely, tbh), so a few of the more obvious questions:
- What release of ESPEasy are you running?
- Can you provide a screenshot of the Device page where GPIO-4 is used?
- Can you provide a screenshot of the Controller page for Homeassistant MQTT?

(Screenshots are easier to provide then all separate settings 😉)

Bruka, 2020-12-29:
ESP_Easy_mega_20201227_hard_Shelly_1_2M256

GPIO-4 is an output and has no device so for it I send shelly14/cmd and message gpio,4.1

[image]

tonhuisman, 2020-12-29:
> message gpio,4.1

I expect you are sending `gpio,4,1` with a comma instead op a period?

Ah, as you have no device defined for gpio-4, there is no taskname and variablename, thus nothing to replace those variables with. 
Suggestions to solve this (only 1 should be applied!):
- Define a task with a Switch device for gpio-4, so the MQTT Controller can use that for the variable replacement
- Remove the %tskname%/%valname% part from the Controller Publish field
- Define an event in the rules that publishes the current state of the gpio, and f.e. use cmd `event,reportgpio=4`

Grovkillen, 2020-12-30:
If you look at his screenshot you see that the unit is broadcasting on the shelly14/%tskname%/%valname% (i.e. it doesn't have any task set up for GPIO 4). This is a design flaw since you shouldn't have to have a task set up for GPIO outputs.

So my suggestion is to look into adding a check before sending the an response:

If no task is set ==> broadcast response on a generic topic (<unit name>/log or something like that).

tonhuisman, 2020-12-30:
> If no task is set ==> broadcast response on a generic topic (/log or something like that).

Well, yes, that's a great idea, but it will require to add another setting like `Controller Publish Status-only` or something similar.

I've just tried to give a few possible options to fix this for now, not as a final solution 😄

Grovkillen, 2020-12-30:
Ah, then I understand you :)

dertobias, 2021-01-11:
Hi all!
Got the same %sysname%/**%tskname/%valname%** issue since updating to newest version… (20210110)
Normally it would Publish on ESP/status
now its on: ESP/%tskname%/%valname%

TD-er, 2021-01-11:
OK, this looks like it is an answer of the `gpio` command, right?
The `gpio` command was changed from being a command of the switch plugin, to being an internal command.

Internal commands do not have a task name, nor a task variable name, so those cannot be replaced.

As this is popping up more often, I wonder what should be done for those situations?
Is it specific to this single MQTT controller, or is it a more generic issue?

Simply stated, how should a reference to a task name or task value name be considered if it is not set?

TD-er, 2021-01-11:
Hmm a simple search through the code reveals it is used in C011, C014, C005 and C010.
It is also part of the default publish topic, which makes it potentially an issue for all MQTT controllers.

So there is a number of possible solutions:
- Add an extra fall-back topic to publish when not all `%...%` elements can be replaced in the topic.
- For internal commands try to fill in `%tskname%` and `%valname%` with useful values like "gpio" (the command) and "1" (the GPIO pin)

Is one preferred over the other, or should I implement both?

dertobias, 2021-01-11:
Thanks for your quick reply!
A fall-back topic would be a good idea!

TD-er, 2021-01-11:
Working on it now and I found some other discrepencies.
I will now first remove the things that cannot be replaced in the publish topic and make sure it appends the `/status` to the topic.
So `%sysname%/%tskname/%valname%` will then be turned into `ESP/status` if it sends a status update from an internal command or an unknown command. (`ESP` is the `%sysname%` here)

TD-er, 2021-01-11:
What kind of test build do you need?
4M1M?

dertobias, 2021-01-11:
Exactly 4M1M!
Many thanks!!!

TD-er, 2021-01-11:
[test_issue_3444.zip](https://github.com/letscontrolit/ESPEasy/files/5795564/test_issue_3444.zip)

I included a normal and a dev build.
Please let me know if this is working well.

dertobias, 2021-01-11:
Just tested it on a wemosd1.
Now it publishes  ESP/%valname%/status
Also the only response is "Ok" for example cmd Status,GPIO,13 returns:
ESP/%valname%/status Ok

TD-er, 2021-01-11:
> Now it publishes ESP/%valname%/status

Check, will look at it.

> Also the only response is "Ok" for example cmd Status,GPIO,13 returns:
> ESP/%valname%/status Ok

I will look at it to see if that one is capable of returning a JSON message like the `gpio` command.

dertobias, 2021-01-11:
i use OpenHAB MQTT.
thanks!!

TD-er, 2021-01-11:
[test_issue_3444_2.zip](https://github.com/letscontrolit/ESPEasy/files/5796247/test_issue_3444_2.zip)

A new test build.
I looked into the reply of the status command.
Are you sure that's the only reply it returns?
It should have sent a JSON message too.

dertobias, 2021-01-11:
Thx! Now it publishes on ESP//status still just 'Ok' when i try the Status,GPIO,13 cmd... no JSON message.

U are right, in the latest release (20210110) there is a JSON message, followed by the 'ok' shortly after (~0.07sec)

TD-er, 2021-01-11:
Hmm is this a typo, or does it really publish to a topic with `//` ?
I do have this in the code: 
```c++
pubname.replace(F("//"), F("/")); 
```
So that should not happen.

dertobias, 2021-01-11:
In MQTT Explorer it looks like this:
[image]
when i copy paste it: wemos_conduction//status

TD-er, 2021-01-11:
Hmm that's strange.
Just as if it was a string with some character between the `//` which may have been removed later due to being an nivalid character?

TD-er, 2021-01-11:
[test_issue_3444_3.zip](https://github.com/letscontrolit/ESPEasy/files/5798147/test_issue_3444_3.zip)

Can you test with this build?
I now explicitly replace the texts with an empty string, instead of assuming a string will be cleared if the task index is invalid.

TD-er, 2021-01-11:
See here for a full test build: https://github.com/letscontrolit/ESPEasy/pull/3439#issuecomment-758278897

dertobias, 2021-01-12:
Just tested 3444_3 with the same outcome. really strange indeed.... will now try your full test build.
Where is pubname.replace(F("//"), F("/"));  located?

TD-er, 2021-01-12:
Can you double check the publish topic in your settings by copying it to Notepad++ (or similar advanced editor) and check if there are strange characters in there?
Or dump the settings and email them to me so I can view them in a HEX editor and/or use it for testing here.

dertobias, 2021-01-12:
[image]

nothing changed in notepad++:
[image]

TD-er, 2021-01-12:
Ah found the problem.
The replace continues after the position it just replaced.
So `///` => `//`.

Simple fix:
```c++
    while (pubname.indexOf(F("//")) != -1) {
      pubname.replace(F("//"), F("/")); 
    }
```

PR #3439 (2021-01-14): [Rules] Add operators to convert integer values
For example:

- toBin
- toHex
- bitRead
- bitSet
- bitClear
- bitWrite
- XOR
- AND
- OR
- div
- mod

Fixes: #3444
closed 2021-01-14 by TD-er