letscontrolit/ESPEasy #3446: Gases - CO2 MH-Z19

DangerD1024, 2021-01-01:
After powering on it shows zero values and sends that zeros to controller.
It there any way to avoid sending zeroes to controller?

[image]

TD-er, 2021-01-01:
Is it sent via a controller assigned to the task?
Or do you send it via rules?

DangerD1024, 2021-01-02:
Just added controller here:
[image]

TD-er, 2021-01-02:
And that's the controller you use to send it from the task?
Can you also show the controller setting of your task configuration?

DangerD1024, 2021-01-02:
Controller
[image]
Interval 10sec
[image]

> And the communication statistics of the task?

Where i can get that?

TD-er, 2021-01-02:
> Where i can get that?

You already gave those in the last screenshot.
`Checksum (pass/fail/reset)`

They do look OK, as in there are not a lot of transmission errors.

What does look a bit tricky is the used GPIO pins.
Probably not causing the symptoms you're describing in this issue, but they may prevent the ESP from booting and cause issues when flashing. (as GPIO-0 needs to be low to enter flash mode)

GPIO-0 must be high during boot and GPIO-15 must be low during boot.
Both pins are now connected to a sensor, which may cause boot failured.
So if possible, I suggest to at least use different pins for the MH-Z19.

TD-er, 2021-01-02:
I labelled it as a bug, as it is possible to add a simple filter in this plugin to only consider a value valid if it is above 300-ish ppm (global CO2 level is already > 400 ppm, but it is possible to have a local concentration < 400 ppm)