letscontrolit/ESPEasy #3447: Weired interpretation of Nextion commands

Wiki591, 2021-01-02:
To get the advantage of WOL I updated my old Wemos D1 connected to a Nextion display from dev 4M1M-binary dated on 08.12.2019 to dev 4M1M-binary dated on 02.11.2020.

Looking at the Weblog I noticed:

[image]

Nextion is written fine, but seems to me as if someone else tries to interprete the Nextion commands before the Nextion plugin does?

The next days I will be at another location where I am able to probably play around with another Nextion-attached device if further informations are needed. Over here I won't do that to leave WAF at the level where it is.

TD-er, 2021-01-02:
Is the "Enable Serial port" checked on the Tools->Advanced page?

Wiki591, 2021-01-02:
No, I do need the serial port for my Nextion

Wiki591, 2021-01-02:
I don't have a problem writing to Nextion nor any other problem.

There are just these protocol entries, Nextion is working fine.

[image]

[image]

TD-er, 2021-01-02:
Hmnm strange...
Command parsing should not be done on data received from the serial port, when "Enable Serial Port" is unchecked.

Looks like a bug.... I will look into it.

Wiki591, 2021-01-02:
The protocl entries are generated if I want to write to Nextion from other devices using:
`
  SendToHTTP 192.168.xx,yy,80,'/control?cmd=Di,page1.Bat1t.txt="[Bat1T#°C]°C"'
`

Wiki591, 2021-01-02:
The command arrives per http, is at first interpreted by someone else, afterwards correctly by the nextion display plugin:

```
5638167: HTTP: Di,page2.t19.txt='0W'
5638172: Command: Di
5638172: Command unknown: 'Di,page2.t19.txt='0W''
5638175: Command: Di
5638180: NEXTION075 : WRITE = page2.t19.txt='0W'

```

TD-er, 2021-01-02:
Apart from the fact "Di" is not a command in ESPEasy, there is also another point to take in mind as  I doubt this command will be sent correct.
```
SendToHTTP 192.168.xx,yy,80,'/control?cmd=Di,page1.Bat1t.txt="[Bat1T#°C]°C"'
```

Can you try sending it via the "Command" field on the Tools page?
That will for sure convert the URL to an URL encoded one (in the addressbar of your browser), like this:
```
http://192.168.10.231/tools?cmd=SendToHTTP+192.168.1.2%2C80%2C%27%2Fcontrol%3Fcmd%3DDi%2Cpage1.Bat1t.txt%3D%22%5BBat1T%23%C2%B0C%5D%C2%B0C%22%27
```

Wiki591, 2021-01-02:
The command "Di" is the name of the Nextion display task and works this way since years.

Have a look one post above, seems as if we have cross-posted

TD-er, 2021-01-02:
Hmm I guess we indeed have cross-posted.
Command parsing is done in several stages:
- internal command
- task command
- others (have to check the code for the correct name)

So it seems the first stage is sending confusing error messages.

Wiki591, 2021-01-02:
Jep, a log snippet directly in relation to my example from above:

```
5691267: Command: Di
5691267: Command unknown: 'Di,page1.Bat1t.txt='14.3°C''
5691270: Command: Di
5691271: NEXTION075 : WRITE = page1.Bat1t.txt='14.3°C'
```

TD-er, 2021-01-02:
Marked it as a todo for the upcoming release, as it does look like a simple fix and leaving it in will generate a lot of new issues.