letscontrolit/ESPEasy #3448: Mixed up configuration page of DS18b20 with build newer than 02.11.2020

Wiki591, 2021-01-02:
The configutation page of DS18b20 devices using a build newer than ESP_Easy_mega_20201227_normal_ESP8266_4M1M.bin looks like this:

[image]

instead of this:

[image]

TD-er, 2021-01-02:
What is the problem?
I added the option to read > 1 sensor in a task.
That allows to have a lot more Dallas sensors on a node, but not all controllers (e.g. Domoticz) can't handle it like this.
So the default setting is to have only a single output value (thus 1 Dallas sensor) per task.

For example my breadboard config of a Dallas task:
[image]

The optional 2nd GPIO selection is intended for Shelly boards which do need a separate RX and TX pin.
You can leave the optional one empty, or set it to the same value as the TX if your setup only needs a single pin.

Wiki591, 2021-01-02:
OK, so it is a problem I generated on my own. On my (almost 1 1/2 years old) device I do need the SPI for a thermocouple sensor. In the past (binary of 08.12.2019) there was no problem to use GPIO-13 for my Dallas in parallel to activated SPI. Now after update I had to reconfigure the Dallas but this wasn't possible because of disbaled GPIO-13 by SPI settings. So I didn't see the possibilities of the changes you have posted. Looks great and I promise, I never ever will attach a Dallas to eventually reserved GPIO ports again.

My confusion has been the naming of the ports and the layout of the page, looks like serial configuration instead of 1-wire.

The page in readthedocs for DS18b20 should be revised.

ironataerial, 2021-01-02:
> The optional 2nd GPIO selection is intended for Shelly boards which do need a separate RX and TX pin.

Wouldn't it defeat the objective connecting 2-Wire on a 1-Wire sensor ?
How would that be possible anyway ?

TD-er, 2021-01-02:
> > The optional 2nd GPIO selection is intended for Shelly boards which do need a separate RX and TX pin.
> 
> Wouldn't it defeat the objective connecting 2-Wire on a 1-Wire sensor ?
> How would that be possible anyway ?

They use an isolator, which are unidirectional.
So you need to have 2 pins, one to send to the 1-wire and one to receive data.
Not sure how this will interfere when I add support for "parasitic powered" Dallas sensors, but that's something for when I implement support for it :)

TD-er, 2021-01-02:
@Wiki591 So this issue can now be closed?

closed 2021-01-02 by Wiki591

CurlyMoo, 2021-01-23:
>  Not sure how this will interfere when I add support for "parasitic powered" Dallas sensors, but that's something for when I implement support for it :)

That wouldn't be an issue, because the only difference is the lack of +5V connection. The communication stays the same.

TD-er, 2021-01-23:
> > Not sure how this will interfere when I add support for "parasitic powered" Dallas sensors, but that's something for when I implement support for it :)
> 
> That wouldn't be an issue, because the only difference is the lack of +5V connection. The communication stays the same.

Nope it isn't... or at least it isn't if you're using more than one Dallas sensor.

What I'm doing now:
- Task A running N Dallas sensors starts the reading sequence for all its sensors and schedules itself to collect the data from them.
- Task B does the same
- Task A collects the samples from its sensors
- Task B does the same

Thus there may be several sensors being addressed and collecting a sample at the same time.
When powered via the Vdd pin, that's just fine and does work very reliable.

However, when you power a Dallas sensor via its data pin (parasitic power), you must keep the data pin high during the measurement.
This is something that has to be shared among all instances of the Dallas plugin (also the iButton and the iCounter and whatever Dallas plugins will be written in the future) if they share the same GPIO pin.
When a sensor is being read and powered via parasitic power, you cannot address any other sensor on that bus.