letscontrolit/ESPEasy #3450: Zero is missing

fly74, 2021-01-02:
Happy new year,

[image]

is there a possibility to get a leading zero for degrees between -1° and 0°?

uzi18, 2021-01-02:
Please show your configuration

fly74, 2021-01-02:
the rule:

...
On Rules#Timer=1 do  
  timerSet,2,8
  7dt,[BLK#Temperature]
endon
...
the device:
[image]

TD-er, 2021-01-02:
See [formatting values](https://espeasy.readthedocs.io/en/latest/Rules/Rules.html?#formatting-refered-values)
```
7dt,[BLK#Temperature#D1.1]
```

fly74, 2021-01-02:
I'll try it, but its too cold at the moment :)

TD-er, 2021-01-02:
You can also set a dummy task to test it.

fly74, 2021-01-02:
no don't work with:

...
On Rules#Timer=1 do  
  timerSet,2,8
  7dt,[D1#Dummy#D1.1]
endon
...

12 | ✔ | Generic - Dummy Device | D1 |   |   |   | Dummy:-0.50
-- | -- | -- | -- | -- | -- | -- | --

TD-er, 2021-01-02:
And what if you just send `7dt,"-0.7"` ?

fly74, 2021-01-03:
The same.

fly74, 2021-01-03:
[image]

Old rule engine

TD-er, 2021-01-03:
You should use the "old" engine.
I will mark the option to make clear the "new" one should not be used as it is not working as expected.

fly74, 2021-01-03:
I always use the old engine.

TD-er, 2021-01-03:
> I always use the old engine.

That's what you should do :)

fly74, 2021-01-03:
So what can we do to format the segments value? 🥺

tonhuisman, 2021-01-03:
The trouble is in this code (`_P073_7DGT.ino`), where the temperature is passed, and formatted, as an integer type:
```
  void FillBufferWithTemp(long temperature) {
    ClearBuffer();
    char p073_digit[8];
    sprintf_P(p073_digit, PSTR("%7d"), static_cast<int>(temperature));
```
The decimal sign is set separately, and put between the - and the 7, where it is numerically correct and works fine below -1 and above +1, but UI-wise note very pleasing between +1 and -1.
Don't think this is easy to be fixed 🤔 

Alternative:
`7dn,"[BLK#Temperature#d6.1]^"`
(The 7dn 'numeric' output just outputs the provided text, and the `^` is translated to display a `°` symbol, manually formatting it in the correct width.)
(Caveat: I don't have such display, 'tested' by adding log statements to the plugin code...)

fly74, 2021-01-03:
Great works, but I don't understand the reason/workaround completely.

closed 2021-01-03 by fly74

tonhuisman, 2021-01-03:
> I don't understand the reason/workaround completely.

The temperature is provided as a float value, multiplied by 10 into a long (integer-type so without decimals), decimal position set to 1 from the right for the Max7219 display type, but as the numeric formatting routine doesn't know about the decimal position, it doesn't insert the desired 0 'prefix' before the decimal position.

tonhuisman, 2021-01-03:
Made a PR adding an exception for temperatures between 0.9 and -0.9 degrees as all display types use 1 digit accuracy when between 10 and -10 degrees.

fly74, 2021-01-03:
Fine. Give a sign when it is testable.

tonhuisman, 2021-01-03:
What .bin are you using? I can build one for you and attach it here.

fly74, 2021-01-03:
ESP_Easy_mega_20201227_normal_ESP8266_4M1M

Thx

> What .bin are you using? I can build one for you and attach it here.

tonhuisman, 2021-01-03:
A local build of only that configuration with PR #3452 included: 
[ESP_Easy_mega_20210103_normal_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/5761594/ESP_Easy_mega_20210103_normal_ESP8266_4M1M.zip)

fly74, 2021-01-03:
Test positive 👍

PR #3452 (2021-01-03): [P073] Display a zero-prefix for temperatures between 0.9 and -0.9
To display `0.9` .. `-0.9` instead of `.9` .. `-.9`.

Adds an exception for temperatures between 0.9 and -0.9 degrees as all display types use 1 digit accuracy when between 10 and -10 degrees.

Resolves #3450