letscontrolit/ESPEasy #3456: potentiometer x9c103s

denismsd, 2021-01-04:
Hello everyone, everything good,
my name and Denis, i'm from Brazil, i don't know english, i'm with google translator, i searched on the internet but i didn't find anything to solve my problem and i also don't know programming.
I have domoticz, and some relay. it controls the lighting of the house and works well, the mqtt and local by node red.
i need my esp8266 with ESP_Easy_mega_20201227_normal_ESP8266_4M1M version.
control a potentiometer model x9c103s,
this potentiometer will be connected to my stereo to increase and decrease the volume through the domoticz, but I don't know how to do it or how to start.
thanks.

TD-er, 2021-01-04:
Hmm that's not the most useful digital potentiometer as it only can be controlled via up/down counts.
So you can only increase/decrease a value and not set it to a specific value.

When you boot the ESP, you don't know what the current value is of the potentiometer and thus you cannot set your stereo to a predefined volume.
I can imagine there are lots of other digital potentiometers which can be controlled via some I2C protocol allowing to set a specific value (and read the current value).

denismsd, 2021-01-04:
Thank you very much for your help, I will look for one that has, I2C

TD-er, 2021-01-04:
OK, I will close this issue, as we're not going to implement support for the one you mentioned.
I also think it is best for discussions like these to ask them on the [forum](https://www.letscontrolit.com/forum/).

closed 2021-01-04 by TD-er