letscontrolit/ESPEasy #3457: Possibility to define Custom Properties on Dummy Devices

AloisKlingler, 2021-01-07:
hello,

my project for a central shutter controller for seven shutters showed me a bit of a limitation of dummy devices.
the shutters are operated with GPIOs which are connected to relays. as every shutters needs two relays, it's kind of mess in my rules where I have to mix up GPIOs and the dummy device which reflects the shutters state.

so I had an idea which is maybe worth to think about: give a dummy device besides the "Values" also editable fields where you can define custom options which can be used in the rules.

below here a rule for one room, which has two shutters, is attached.
If I now have to change any GPIOs for this, I need to adapt the rules on all positions.
If I would have fields in the dummydevice where I define which GPIOs are used for this device, then I can access the values in rules (e.g. with [Room3#Prop1] , I can assign the according dummydevice-property-value to a variable and just copy&paste the script for all rooms.

another stumbling block: in rules I have to fetch values from the dummy devices with e.g. "[SZ#S]" - but to write those values I have to use TaskValueSet,1,1,VAR#1 - it would be much more handy if this would be in same format, e.g. TaskValueSet,[SZ#S],VAR#1

Best regards
Alois

dummydevice-config:
[image]

rules:
```
on ControlZ3 do
 //control?cmd=event,ControlZ3=0
 Let,4,%eventvalue%
 if [VAR#4] > 0 and [VAR#4] < 100
  //open/close percent
  Let,5,(20/100)*[VAR#4]
  Let,6,(20/100)*[Z3#S]
  if [VAR#6] > [VAR#5]
   // open a bit
   Let,7,[VAR#6]-[VAR#5]
   logentry,Z3_o_secs=[VAR#7]
   delay,250
   LongPulse,2,0,[VAR#7] //PWR
   LongPulse,26,0,[VAR#7]
   gpio,25,1
   LongPulse,33,0,[VAR#7]
   gpio,32,1
  endif
  if [VAR#6] < [VAR#5]
   //close a bit
   Let,7,[VAR#5]-[VAR#6]
   logentry,Z3_c_secs=[VAR#7]
   delay,250
   LongPulse,2,0,[VAR#7] //PWR
   LongPulse,26,0,[VAR#7]
   LongPulse,25,0,[VAR#7]
   LongPulse,33,0,[VAR#7]
   LongPulse,32,0,[VAR#7]
  endif
 else
  if [VAR#4] = 0
   logentry,Z3_o_comp
   delay,250
   LongPulse,2,0,20 //PWR
   LongPulse,26,0,20
   gpio,25,1
   LongPulse,33,0,20
   gpio,32,1
  endif
  if [VAR#4] = 100
   logentry,Z3_c_comp
   delay,250
   LongPulse,2,0,20 //PWR
   LongPulse,26,0,20
   LongPulse,25,0,20
   LongPulse,33,0,20
   LongPulse,32,0,20
  endif
  if [VAR#4] = 101
   //special state
  endif
  if [VAR#4] = 102
   //special state
  endif
  if [VAR#4] = 103
   //special state
  endif
 endif
 TaskValueSet,4,3,[VAR#4]
endon
```

Grovkillen, 2021-01-07:
If you instead showed me a logical block diagram I could probably help you with cleaning up the rules and make it more maintainable.

TD-er, 2021-01-07:
As was discussed also when this (merged) PR was being developed: https://github.com/letscontrolit/ESPEasy/pull/3198

But to be honest, I don't know why some format like this would not work.
```
TaskValueSet,[SZ#S],[VAR#1]
```

You can store specific values (like GPIO pin or task index) in variables at the system boot event and use them exactly as you tried in this way

AloisKlingler, 2021-01-08:
hello @Grovkillen 
thanks for the offer with the rules - I will come back to it by a private message in forum, I guess it is the better place. :-)

@TD-er: I tried:
   TaskValueSet,[Z3#S],[VAR#4]
   TaskValueSet,4,2,[VAR#4]
   TaskValueSet,4,3,[VAR#4]

but it did not work. [Z3#S] was filled by it's value (was 100), instead of a reference to the device. log:

4447969: ACT : TaskValueSet,100,0.00
4447973: ACT : TaskValueSet,4,2,0.00
4447977: ACT : TaskValueSet,4,3,0.00

I cannot store the GPIO pins to variables, as I need most probably all variables in the rules for dynamic data and calculation. Not in the rules above, but in other rules. 

Best regards
Alois

tonhuisman, 2021-01-08:
> I tried:
> TaskValueSet,[Z3#S],[VAR#4]

After applying PR #3098 it is supposed to be used like `TaskValueSet,Z3,S,[VAR#4]`, so instead of the numbers you can use the task and var names of the Dummy Device.

What TD-er was trying to say was that you can store the variable numbers in other variables, so you can already use
 `TaskValueSet,[Z3#Task],[Z3#Var],[VAR#4]` where Z3#Task stores the task number (1..12 or 1..32 on ESP32) and Z3#Var stores the var number (1..4)
Instead of Dummy Devices you can also use Var's, set by the `Let` command, like `[Var#4]` or `%v4` (that's actually the same variable)
In PR #3439 TD-er recently lifted the limitation of 16 var's, and you can use as many var's as memory allows.

TD-er, 2021-01-08:
> In PR #3439 TD-er recently lifted the limitation of 16 var's, and you can use as many var's as memory allows.

"Recently" must be read as literally as possible, as I did it today.
And believe it or not, it was meant as an optimization and also with your attempts in mind as I already imagined you would need more then 16 variables for this :)

AloisKlingler, 2021-01-09:
guys, you are my heros. 👍 

in #3439 I also saw there is abs() function now. is the 16 var limitation in this build (https://github.com/letscontrolit/ESPEasy/pull/3439#issuecomment-757142804) already in?

Thanks!

Alois

TD-er, 2021-01-09:
Not yet, I'm working on it.
And I will change some of the new introduced functions there to make them more intuitive.

AloisKlingler, 2021-01-28:
> After applying PR #3098 it is supposed to be used like `TaskValueSet,Z3,S,[VAR#4]`, so instead of the numbers you can use the task and var names of the Dummy Device.

Do you have a rough idea when it looks ready for merge? I can see #3098 has a lot commits, partially merged - maybe the needed part is not as big as the complete PR? 😍 

BR
Alois

closed 2021-02-12 by AloisKlingler

tonhuisman, 2021-02-12:
> maybe the needed part is not as big as the complete PR?

The effectively changed code has become smaller and smaller over time because other changes and optimisations, most changes are now in documentation 😆

AloisKlingler, 2021-02-12:
in the meantime, I would like more addressing values to read by number, as handling numbers in rules is possible, strings (more or less) not.
So, a function like "TaskValueRead,1,2" would help more than TaskValueSet,[SZ],[S] as the arguments of the function can be stored and recalled in variables. ;)