letscontrolit/ESPEasy #3458: sendToHttp question...

giig1967g, 2021-01-07:
Hi,
I am struggling with the following issue:
I created a script (called: `commandoff`) in JavaScript that I can execute from Firefox but that I am not able to execute from EspEasy.
From Firefox the script runs ok: 
`http://192.168.88.5:8080/static/commandoff`

But from EspEasy it doesn't do anything:
`sendtohttp,192.168.88.5,8080,/static/commandoff`

Here is the log:
```
1029695514: HTTP: sendtohttp,192.168.88.5,8080,/static/commandoff
1029695515: HTTP after parseTemplate: sendtohttp,192.168.88.5,8080,/static/commandoff
1029695518: Command: sendtohttp
1029695519: sendtohttp,192.168.88.5,8080,/static/commandoff
1029695519: Par1: 192 Par2: 8080 Par3: 0 Par4: 0 Par5: 0
1029695520: SendToHTTP: Host: 192.168.88.5 port: 8080
1029695521: SendToHTTP: Path: /static/commandoff
1029695525: GET /static/commandoff HTTP/1.1^^Host: 192.168.88.5:8080^^Accept: */*;q=0.1^^User-Agent: ESP Easy/20111/Dec 17 2020 19:22:27
1029695526: GET /static/commandoff HTTP/1.1^^Host: 192.168.88.5:8080^^Accept: */*;q=0.1^^User-Agent: ESP Easy/20111/Dec 17 2020 19:22:27
1029695527: HTTP : Command_HTTP_SendToHTTP written to client (150/150)
1029695527: HTTP : Command_HTTP_SendToHTTP closing connection

```

I am running openHab 2.5 where I can create html JavaScript scripts called by an Http GET request.

Here is the simple script:
```
<!DOCTYPE html>
<html>
    <script>    
        var xhr = new XMLHttpRequest();
        xhr.open('POST', "http://192.168.88.5:8080/rest/items/zs10a");
        xhr.setRequestHeader("Content-Type", "text/plain");
        xhr.setRequestHeader("Accept", "application/json");
        xhr.send("OFF");
    </script>
</html>

```
The scope of the script is to send a POST request to the openHAB server to turn off a lamp.

TD-er, 2021-01-07:
Have you also tested with this checked?
[image]

giig1967g, 2021-01-07:
Hi @TD-er 
Yes I checked with that parameter too.
But unfortunately same result.

giig1967g, 2021-01-07:
I checked in the code and the sendToHttp doesn't use the HttpClient class, but sends the request using WiFiClient::print(request).
I found a bit weird... Actually the whole http code is very complex to read.

TD-er, 2021-01-07:
Yep, there is a good reason for it to not use the HttpClient class, as it adds quite a lot to the build size.
But since I've now added it to a few other places, it may no longer be such a large contributor to the bin sizes, so maybe we should have a look at it.

uzi18, 2021-01-07:
Do you have any logs from http server on 192.168.88.5?

giig1967g, 2021-01-08:
> Do you have any logs from http server on 192.168.88.5?

I didn't find any. Are you using openHab?

tonhuisman, 2023-09-01:
This issue doesn't seem valid anymore, and can be closed.

closed 2023-09-01 by TD-er