letscontrolit/ESPEasy #3459: MAX7219 add dot char to parameter "7dtext,<text>"

aivs, 2021-01-09:
Hi, the dot is needed to display fractional numbers.
My display shows the outdoor and the room temperature.
I used text command to display two temperatures, with a dot it would be prettier.
http://192.168.1.33/control?cmd=7dtext,28.0-3.5

[image]

tonhuisman, 2021-01-09:
For displaying temperatures the `7dt` or `7dn` commands should probably be used, but I don't think that'll work with your hardware setup 😞

aivs, 2021-01-09:
Ok, how to display first temperature on the first 4 segments and second temperature display on the next 4 segments?
With text its possible. But need dot char.

tonhuisman, 2021-01-09:
Well, splitting the display with the Max7219 currently isn't possible. That's what the sad face in my previous reply is for 😃 

I recently added/fixed the 0-prefix for temperatures between 1 and -1 degrees in this plugin, so I'll see if I can fix this too.
That would be:
- Periods in `7dtext`
- Dual temperature display in `7dt` command (may need to add a command `7ddt` for **7d** **d**ual **t**emperature).
This would only be applicable for the Max7219 display as other devices don't have enough space.

aivs, 2021-01-09:
What the problem to add dot to 7dtext?  This is a more universal solution and it can be used not only for temperature display.

tonhuisman, 2021-01-09:
@aivs  I've done some changes on `7dtext` already, what kind of .bin file do you need, so I can supply one for testing, as I don't own such display (yet)... 😄

aivs, 2021-01-09:
Wow! quickly! 
I'm using a test build, like this: ESP_Easy_mega_20201227_test_ESP8266_4M1M_VCC.bin

tonhuisman, 2021-01-09:
That is the build I'm working on, just uploaded it to my test-unit and the UI looks OK, as I've added an option to turn this feature on `Test show periods as dot`, default off (backward compatible)

[ESP_Easy_mega_20210109_test_ESP8266_4M1M_VCC.zip](https://github.com/letscontrolit/ESPEasy/files/5791705/ESP_Easy_mega_20210109_test_ESP8266_4M1M_VCC.zip)

I'm quite curious what your findings are, but you'll have  to add an extra space somewhere to get the text-distribution right.
Didn't do the double temperature display yet.

aivs, 2021-01-09:
Now nothing is displayed.
The display does not respond to the test command
http://192.168.1.33/control?cmd=7dtext,12345678
I checked and unchecked the box, the result is the same

tonhuisman, 2021-01-09:
Hm, I'll take a look tomorrow, I'm at UTC+1, meaning I'm not getting much done today anymore 😉

TD-er, 2021-01-09:
> Hm, I'll take a look tomorrow, I'm at UTC+1, meaning I'm not getting much done today anymore 😉

You still got close to 2 hours left today :)

aivs, 2021-01-09:
I can test it at any time. I'm at UTC+4 (Moscow).
I believe that a special checkbox is not needed. In any case, if there is a dot in the text, it should be shown.
Possible variants of the text: "1234.5678.", "0....0....", "-.-.-.-.-.-.-.-"

tonhuisman, 2021-01-10:
Well, after a good night of sleep, and a couple of hours debugging stuff, I think I've got this `7dtext` with periods to dots conversion covered 😎 
Please try today's build:
[ESP_Easy_mega_20210110_test_ESP8266_4M1M_VCC.zip](https://github.com/letscontrolit/ESPEasy/files/5792722/ESP_Easy_mega_20210110_test_ESP8266_4M1M_VCC.zip)

> You still got close to 2 hours left today :)

Well, I really need my beauty-sleep, and sometimes allow myself a couple of alcoholic beverages, so that's what I ~~wasted~~ spent my time on 😋

aivs, 2021-01-10:
I will test today at night.

aivs, 2021-01-10:
This is a success!

[image]
[image]

tonhuisman, 2021-01-10:
That's looking great, and because I used your test cases, of course they work 🤣 

In the meantime I've worked on (and completed) the `7ddt` command, that takes 2 values, comma separated, and presents them as separate temperatures (including the degrees symbol)
When feeding it values > 999 or < -99.9 it will display 4 dashes instead (like the other display types do).

If you would be so kind to test that too, a valid command would be `7ddt,14,-3.5` but `7ddt,21.3` will also work, but just display 0.0 for the second temperature.
In the logging you can now also find the same output for 7ddt and 7dt like 7dtext, I changed the P: to display `t` or `f` instead of 1/0. (This extra logging will be removed in the final builds that are 'memory-challenged' to save some space in the bin file.)

[ESP_Easy_mega_20210110_test_ESP8266_4M1M_VCC.zip](https://github.com/letscontrolit/ESPEasy/files/5793097/ESP_Easy_mega_20210110_test_ESP8266_4M1M_VCC.zip)

tonhuisman, 2021-01-10:
> For full functionality would be nice to add ":", but this is not so important.

I'm not sure if the Max7219 _can_ drive the colon between the digits, but it also would require a very specific text to be displayed 🤔

aivs, 2021-01-11:
First char missing

7ddt,14,-3.5
[image]

7ddt,21.3
[image]

tonhuisman, 2021-01-11:
> First char missing

That's not how I planned that, hmm, I'll have a look in a few hours (after work).

tonhuisman, 2021-01-11:
I've had a look at the code, but because I haven't received by display yet, I can't actually test it. Nothing seems different from the 7dtext command that worked before.

- Can you verify that the 7dtext command still works as expected?
- Can you show the log output (Info level is just fine) where the 7dtext and 7ddt log lines are shown?

aivs, 2021-01-12:
Today it was not possible to test, tomorrow I will check

aivs, 2021-01-13:
http://192.168.1.33/control?cmd=7dtext,25.0%20-10.5
246710 : Info   : HTTP: 7dtext,25.0 -10.5
246713 : Info   : 7DGT : Show Text=25.0 -10.5
246714 : Info   : 7dtext buffer: periods: t 0x2,,0x5,.,0x0,,0xa,,0xb,,0x1,,0x0,.,0x5,
[image]

http://192.168.1.33/control?cmd=7ddt,25.0,-10.5
303591 : Info   : HTTP: 7ddt,25.0,-10.5
303596 : Info   : 7DGT : Show Temperature 1st=25.00 2nd=-10.50
303598 : Info   : 7ddt buffer: periods: t 0x2,,0x5,.,0x0,,0xc,,0xb,,0x1,,0x0,,0xc,
[image]

tonhuisman, 2021-01-13:
> 246713 : Info : 7DGT : Show Text=25.0 -10.5
> 246714 : Info : 7dtext buffer: periods: t 0x2,,0x5,.,0x0,,0xa,,0xb,,0x1,,0x0,.,0x5,

> 303596 : Info : 7DGT : Show Temperature 1st=25.00 2nd=-10.50
> 303598 : Info : 7ddt buffer: periods: t 0x2,,0x5,.,0x0,,0xc,,0xb,,0x1,,0x0,,0xc,

Thanks for your tests.
That is exactly the kind of response I'd expect in the log, and if you compare the buffer contents you can see they are quite similar, and that buffer (there is only 1 buffer where the output is prepared) gets sent to the display with exactly the same function. So I can not understand why the 7ddt command doesn't include the first digit 🤔 I'll have a close look at the code again to see if there is some argument or algorithm difference that I missed until now.

I'd also like this to be tested with another Max7219 7 segment display module, just for cross-reference, do you own another display you can try, or should we invite someone else (@TD-er ?) to try the above build?

TD-er, 2021-01-13:
Maybe also try to use your patch on the code base of PR #3439 ?
I did fix some numerical parsing things in there, maybe it is just a matter of incorrect parsing of parameters?

Since you're essentially only changing 1 file, it is probably quite easy to test it against that code base.

tonhuisman, 2021-01-13:
> maybe it is just a matter of incorrect parsing of parameters?

That would be weird, as the `Show Temperature` log line has the correct values, an also the buffer is containing the correct hex values (0-9 = numbers, 0xa = space, 0xc = degrees, 0xb = dash). The code of the plugin is written very well, IMHO.

aivs, 2021-01-13:
I have the same module from aliexpress. But it will be better if someone else tests it.
[image]
[image]

tonhuisman, 2021-01-13:
It is a somewhat different type of module, so it would be great if you could test it 👍

aivs, 2021-01-13:
This is exactly the same module, on the same board!
I only placed the displays not side by side, but one under the other.

tonhuisman, 2021-01-13:
Pfew, found it. And I'd call it a bug, I'd think.
Temperature, including the degrees symbol, has always been displayed on the left 7 digits of the 8-digit display. Don't know why, but when sending the 8 byte buffer to the display, only the _right_-most 7 bytes are sent to the 7 _left_-most display-digits. 😞 
Had I owned such display then it would have been obvious from the start when trying the `7dt` command 😮 

As I'm inclined to believe this is actually a bug, I've now fixed that, unless someone might think it should be bug-compatible for the single-temperature `7dt` command?

@aivs Please try this fresh build:
[ESP_Easy_mega_20210113_test_ESP8266_4M1M_VCC.zip](https://github.com/letscontrolit/ESPEasy/files/5811247/ESP_Easy_mega_20210113_test_ESP8266_4M1M_VCC.zip)

aivs, 2021-01-13:
http://192.168.1.33/control?cmd=7ddt,30.8,-10.5
If you do not add the degrees symbol, then more information will fit

[image]

tonhuisman, 2021-01-14:
> If you do not add the degrees symbol, then more information will fit

Agreed, but I'd like to keep it as backward compatible as possible. People are using this plugin, I assume, so they should not be surprised by unexpected changes. And I hope your ambient temperature won't exceed 100 °C 😆, but it would be helpful for temperatures below -10 °C, (like my freezer has, between -13 and -18).

I'll add 2 more options to the configuration:
- Hide degrees symbol for Temperature display (7dt, 7ddt) _(I don't think that should be split per temperature for 7ddt)_
- Right-align Temperature display (7dt) _(only applicable for Max7219 displays)_

Both will be off by default.

aivs, 2021-01-14:
This is the right way to add 2 options. Thank you very much for your work!

PR #3460 (2021-02-21): [P073] Improve display of periods in 7dtext
- Optionally display periods in `7dtext` command as dots on display, default disabled (= backward compatible)
- Adds command `7ddt,<left temp>,<right temp>` for dual temperature display (includes degree symbol) (removed in memory-tight configurations, only applicable to MAX7219 8-digit display)
- Minor code optimizations (Disabling debug leaves that code out)
- Adds option for not showing the degrees symbol with `7dt` and `7ddt` commands, allowing higher and lower temperatures to be displayed (>1000° and < -100°)
- Adds option to right-align output f [... cut]
closed 2021-02-21 by TD-er