letscontrolit/ESPEasy #3461: [Notify]Problems with Devices in Dropdown

W4nn4be, 2021-01-12:
Hello there,

i have no inexperienced in terms of ESPEasy and i am sorry for not knowing but i am trying to solve this for hours now, but no success. Current project is playing *.mp3 files from SD-Card (esp8266 x dfplayer mini) via mqtt.

`https://www.letscontrolit.com/wiki/images/e/ea/Mp3-device.PNG`

I cant reproduce this, there is no "Notify" for me in dropdown. I tried every single device from the dropdown, but there is no device template with "volume".

Pretty sure its a beginner question, but i hope someone can help with this.
Thx in advance

tonhuisman, 2021-01-12:
You will need an ESP unit, either ESP32 or ESP8266, with 4 MB of flash, and install a .bin file (get the latest from Releases link) with `test` in the name. That build has the Notify - DFPlayer Mini MP3 plugin available.

W4nn4be, 2021-01-12:
Thank you, sir.

virtual hugs for u :)
totally solved my problem, i used the normal bin. My misstake.

closed 2021-01-12 by W4nn4be