letscontrolit/ESPEasy #3463: When Wifi connection is made the selected SSID is not chosen when multiple SSID with same name are available

wifiminime, 2021-01-14:
### Summarize of the problem/feature request
In my home i have multiple Wifi AP's with the same SSID name. When an ESP device boots, it tries to connect to one of the Wifi AP's. When i look at the serial output, the selected AP is not the one that the ESP device connects to.

### Expected behavior
When multiple Wifi AP's with same SSID are available ESP must connect to the strongest AP.

### Actual behavior
The ESP seems to connect to the AP based on sorted MAC address, which is in my case the weaker AP.
serial log output:
2283 : Info   : WIFI  : Selected: MiniMe24 EC:08:6B:24:C9:50 Ch:11 (-56dBm) WPA2/PSK     **<--- strong AP -56 dBm**
3313 : Info   : WIFI : Connected! AP: MiniMe24 (60:E3:27:C8:5E:BF) Ch: 1 Duration: 1065 ms         **<--- Weak AP -80 dBm**

I think what happens is that the ESP connects to the second MAC address because it will be first when sorted on MAC when both are available.

This happens on all ESP devices that i use and i also use many different versions from ESP R1.47 up to latest Mega builds.
I use ESP8266, 8285 and ESP32 devices

### Steps to reproduce
1. setup 2 access points with identical SSID, make sure that the weaker AP has the 'lowest MAC address'
2. Factory reset, setup ESP to connect to device to strongest SSID, will by default be selected
2. monitor serial output, reset ESP --> log will show that ESP selects correct AP, but connects to AP with "lowest MAC address"

I think many people have a setup with multiple AP, same SSID. This problem is very annoying although my Wifi coverage is good (there is always is strong AP available) when the ESP device 'sees' the AP with the lowest MAC, it will always connect to that depending on the setup and place, this can be the very weak AP. This makes my setup less reliable and leads to unnecessary connection losses.

TD-er, 2021-01-14:
ESPEasy does try to reconnect to the last known BSSID when a connection was lost or after a crash or warm reboot (N.B. not after power cycle)

I recently noticed the selection criteria for what BSSID to connect to are different after you perform a WiFi scan via the web UI.
It was my intention to have it all use the same criteria, but apparently that's not how it is implemented right now.

The selection of the BSSID at cold boot, or when it does switch to a fallback SSID configuration (after 2 failed connection attempts, or when the connection was lost twice in a short time) is based on the logic in the core libraries, so it could be alfabetically, but I doubt it uses alfabetical order as a selection criterium.
I do think the weaker AP may perhaps react a bit faster, or maybe it is based on the scan order (thus lower channel wins).

What I do find a bit strange is that you're showing ESPEasy logs which clearly state which one to select and apparently it does connect to another one.

wifiminime, 2021-01-14:
I think if it would be nice if the scan in the Web UI could 'reselect' the AP to connect to (including providing a password) . Now i have no control over it. I guess after a few connection losses it will eventually connect to the stronger AP (because other one is to weak or not visible). The scan functionality currently is purely a scan.

wifiminime, 2021-01-14:
One other remark, i believe i never saw a initial setup where the weaker was first in line (or one of my neighbors AP's) . The initial setup always seems to have the strongest AP on top, which makes sense especially when many AP's are present.

tonhuisman, 2022-04-18:
This should be resolved by recently merged PR #3983 so can be closed.

closed 2022-04-18 by TD-er