letscontrolit/ESPEasy #3464: log state misrepresented on system info page

Muller59, 2021-01-15:
### Summarize of the problem/feature request

YOUR TEXT GOES HERE
the log state is misrepresented on the  system info page

### Expected behavior

### Actual behavior

at tools-log settings both serial log and web log are set to debug
at tools-info serial log level is debug, web log level is none

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 
ESP_Easy_mega_20210114_normal_ESP8266_4M1M

ESP Easy settings/screenshots: 

[image]

[image]

### Rules or log data

```

```

TD-er, 2021-01-15:
The web log is only active if it is being queried. (and set enabled of course)
That's the reason why I added the status there.

Muller59, 2021-01-15:
Ah, i see. a bit confusing, but i can live with that now i know the reason.
thx

closed 2021-01-15 by Muller59