letscontrolit/ESPEasy #3466: AHT10 sensor

linuxvsmicrosoft, 2021-01-16:
Hello
I would open an issue, because the actual ESPeasy Mega software does not support AHT10 humidity/temp sensors and I have a few of them and would use them with my standard firmware for ESP8266 ... that is  you ESPeasy Mega image.
Thank you in advice

Wookbert, 2021-04-25:
+1

I'd also like to see support for the AHT10 in ESPEasy. The board is significantly smaller than the typical DHT11/22 boards, and accuracy is claimed to be better too.

[image]
[image]

TD-er, 2021-04-25:
Like this: https://github.com/letscontrolit/ESPEasyPluginPlayground/pull/176#issuecomment-826084980 ??

TD-er, 2021-04-25:
As it is now part of the playground, I also added an issue to move it to the main repository.
https://github.com/letscontrolit/ESPEasy/issues/3604

But it does need some big fat warnings as described in that issue as it is a sensor that does not allow to work with any other I2C device on the ESP.
It is not working according to the I2C standard.

TD-er, 2021-04-25:
> I'd also like to see support for the AHT10 in ESPEasy. The board is significantly smaller than the typical DHT11/22 boards, and accuracy is claimed to be better too.

The board looks similar in size to the Bosch BME280/BMP280 boards and those do comply with I2C standards.

Wookbert, 2021-04-25:
How about the [SHT30 or SHT31](https://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20210425034355&SearchText=sht30). (See description e.g. [here](https://www.aliexpress.com/item/32678741657.html)). Even smaller. Looks to be I2C compliant, or am I wrong?

[image]

TD-er, 2021-04-25:
Those SHT ones are also quite good.
The higher the last digit of them, the more expensive (higher claimed accuracy and/or precision)

tonhuisman, 2021-08-01:
@linuxvsmicrosoft The plugin is migrated from the Playground, and can be tested if you want. 
@Wookbert If you have these AHT sensors available, please try this PR build to see if it works as expected.

I can provide a .bin file if needed, as it is not yet included in the regular release of ESPEasy, (and you most likely can't download the PR build files yourself.)

Wookbert, 2021-08-02:
@tonhuisman I think I have a single AHT Wemos shield here which I had the intention to use, but actually never did and dropped it in favor of the BMP/BME280, because of the above discussion. Unfortunately I don't have the time to test AHT10. Giving away the AHT10 shield to whoever is willing to pay postage from Germany.

Wookbert, 2021-08-02:
Uhm ... I know 🙃

tonhuisman, 2021-08-02:
DHT11 _is_ supported by ESPEasy...

Wookbert, 2021-08-02:
Copy 'n paste: Uhm ... I know 🙃. I just don't trust the DHT11.

linuxvsmicrosoft, 2021-08-02:
Hi guys. Thanks for adding support for the aht10 sensor. Could you provide an download link for the specific firmware to check the functionality?
thank you for all this work done !

tonhuisman, 2021-08-02:
On what type of hardware do you want to install it on? (ESP8266 or ESP32, flash size and file system size, f.e. ESP8266: 1M, 4M1M, 4M2M, ESP32: 4M316k)

tonhuisman, 2021-08-02:
I've created local builds for ESP8266 normal 1M and test_A 4M flash, and an ESP32 test_A 4M316k:
[ESP_Easy_mega_20210802_normal_ESP8266_1M.zip](https://github.com/letscontrolit/ESPEasy/files/6918908/ESP_Easy_mega_20210802_normal_ESP8266_1M.zip) (normally *doesn't* have the AHT plugin!)
[ESP_Easy_mega_20210802_test_A_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6918935/ESP_Easy_mega_20210802_test_A_ESP8266_4M1M.zip)
[ESP_Easy_mega_20210802_test_A_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/6918912/ESP_Easy_mega_20210802_test_A_ESP32_4M316k.zip)

yohru, 2021-08-03:
Thank you for adding AHT10 sensor.
I was able to confirm that it works properly. 
It is working fine on the same I2C bus (Environment dependent)

[image]

TD-er, 2021-08-03:
Hmm looks like you're missing the CSS.
Please have a look at [Read The Docs - ExternalHostedStaticFiles](https://espeasy.readthedocs.io/en/latest/Reference/ExternalHostedStaticFiles.html?highlight=css#external-hosted-static-files)

yohru, 2021-08-04:
Thank you.
Is this the difference between graphical or not ... 
Well, so I understood. 

[image]

P.S.
I have total two AHT10 and ESP8266 combinations.
Both are working fine.

PR #3724 (2021-09-05): [P105] Move AHT10/20/21 plugin from Playground
Move plugin for AHT10/20/21 I2C environment sensor from the Playground.

- Some adjustments to the code, reformatted using Uncrustify
- Re-used Plugin ID P105
- Documentation

TODO:
- [x] Complete documentation
- [ ] Testing (to be reported)
  - [x] AHT10 (confirmed by ghtester, and dragonfly69)
  - [ ] AHT20
  - [x] AHT21

Resolves #3604 
Resolves #3466
Resolves #3436 
Resolves #3550
closed 2021-09-05 by TD-er