letscontrolit/ESPEasy #3469: Vagrant custom build - P016 error

ghtester, 2021-01-19:
After a longer period, I tried to make a custom build (through Vagrant + Custom.h) as none from the default binaries (except minimal) contains IR RX plugin (P016) anymore (which is much more important for me than IR TX).

Unfortunately the build ended up with this error and no binary was made:

  default: Compiling .pio/build/custom_ESP8266_4M1M/lib777/HT16K33/HT16K33.cpp                         .o
    default: Compiling .pio/build/custom_ESP8266_4M1M/lib9a1/LiquidCrystal_I2C/L                         iquidCrystal_I2C.cpp.o
    default: Compiling .pio/build/custom_ESP8266_4M1M/lib3c4/SparkFun_APDS-9960_                         Sensor_Arduino_Library/SparkFun_APDS9960.cpp.o
    default: Compiling .pio/build/custom_ESP8266_4M1M/lib94f/Adafruit_MPR121/Ada                         fruit_MPR121.cpp.o
    default: Compiling .pio/build/custom_ESP8266_4M1M/src/ESPEasy-Globals.cpp.o
    default: Compiling .pio/build/custom_ESP8266_4M1M/src/ESPEasy.ino.cpp.o
    default: /home/vagrant/GitHub/letscontrolit/ESPEasy/src/ESPEasy.ino:17:4: wa                         rning: #warning "**** Using Settings from Custom.h File ***" [-Wcpp]
    default:    #warning "**** Using Settings from Custom.h File ***"
    default:     ^
    default: /home/vagrant/GitHub/letscontrolit/ESPEasy/src/_P016_IR.ino:22:29:                          fatal error: IRremoteESP8266.h: No such file or directory
    default:
    default: *******************************************************************                         ******
    default: * Looking for IRremoteESP8266.h dependency? Check our library regis                         try!
    default: *
    default: * CLI  > platformio lib search "header:IRremoteESP8266.h"
    default: * Web  > https://platformio.org/lib/search?query=header:IRremoteESP                         8266.h
    default: *
    default: *******************************************************************                         ******
    default:  #include <IRremoteESP8266.h>
    default:                              ^
    default: compilation terminated.
    default: *** [.pio/build/custom_ESP8266_4M1M/src/ESPEasy.ino.cpp.o] Error 1
    default: ========================= [FAILED] Took 209.26 seconds ============                         =============
    default: Environment          Status    Duration
    default: -------------------  --------  ------------
    default: custom_ESP8266_4M1M  FAILED    00:03:29.262
    default: ==================== 1 failed, 0 succeeded in 00:03:29.262 ========                         ============

Any hint what can I do to fix the issue please?
Thanks for any kind of help on advance.

My plugin list from Custom.h looks like:

/*
 #######################################################################################################
 ###########     Plugins
 #######################################################################################################
 */

// #define USE_SERVO

 #define USES_P001   // Switch
 #define USES_P002   // ADC
 #define USES_P003   // Pulse
 #define USES_P004   // Dallas
 #define USES_P005   // DHT
// #define USES_P006   // BMP085
// #define USES_P007   // PCF8591
 #define USES_P008   // RFID
// #define USES_P009   // MCP

 #define USES_P010   // BH1750
// #define USES_P011   // PME
 #define USES_P012   // LCD
 #define USES_P013   // HCSR04
 #define USES_P014   // SI7021
// #define USES_P015   // TSL2561
 #define USES_P017   // PN532
// #define USES_P018   // Dust
 #define USES_P019   // PCF8574

// #define USES_P020   // Ser2Net
// #define USES_P021   // Level
// #define USES_P022   // PCA9685
 #define USES_P023   // OLED
// #define USES_P024   // MLX90614
// #define USES_P025   // ADS1115
 #define USES_P026   // SysInfo
 #define USES_P027   // INA219
 #define USES_P028   // BME280
 #define USES_P029   // Output

// #define USES_P031   // SHT1X
// #define USES_P032   // MS5611
 #define USES_P033   // Dummy
 #define USES_P034   // DHT12
// #define USES_P036   // FrameOLED
 #define USES_P037   // MQTTImport
// #define USES_P038   // NeoPixel
// #define USES_P039   // Environment - Thermocouple

// #define USES_P040   // RFID - ID12LA/RDM6300
// #define USES_P041   // NeoClock
// #define USES_P042   // Candle
 #define USES_P043   // ClkOutput
// #define USES_P044   // P1WifiGateway
// #define USES_P045   // MPU6050
// #define USES_P046   // VentusW266
// #define USES_P047   // I2C_soil_misture
// #define USES_P048   // Motoshield_v2
 #define USES_P049   // MHZ19

// #define USES_P050   // TCS34725 RGB Color Sensor with IR filter and White LED
 #define USES_P051   // AM2320
// #define USES_P052   // SenseAir
// #define USES_P053   // PMSx003
// #define USES_P054   // DMX512
// #define USES_P055   // Chiming
// #define USES_P056   // SDS011-Dust
// #define USES_P057   // HT16K33_LED
// #define USES_P058   // HT16K33_KeyPad
// #define USES_P059   // Encoder

// #define USES_P060   // MCP3221
 #define USES_P061   // Keypad
 #define USES_P062   // MPR121_KeyPad
 #define USES_P063   // TTP229_KeyPad
// #define USES_P064   // APDS9960 Gesture
// #define USES_P065   // DRF0299
// #define USES_P066   // VEML6040
// #define USES_P067   // HX711_Load_Cell
 #define USES_P068   // SHT3x
// #define USES_P069   // LM75A

// #define USES_P070   // NeoPixel_Clock
// #define USES_P071   // Kamstrup401
// #define USES_P072   // HDC1080
// #define USES_P073   // 7DG
// #define USES_P074   // TSL2561
// #define USES_P075   // Nextion
// #define USES_P076   // HWL8012   in POW r1
// #define USES_P077   // CSE7766   in POW R2
 #define USES_P078   // Eastron Modbus Energy meters
// #define USES_P079   // Wemos Motoshield

 #define USES_P080   // iButton Sensor  DS1990A
// #define USES_P081   // Cron
// #define USES_P082   // GPS
 #define USES_P083   // SGP30
// #define USES_P084   // VEML6070
// #define USES_P085   // AcuDC24x
 #define USES_P086   // Receiving values according Homie convention. Works together with C014 Homie controller
// #define USES_P087   // Serial Proxy
// #define USES_P088   // HeatpumpIR
// #define USES_P089   // Ping

// #define USES_P090   // CCS811
// #define USES_P091   // SerSwitch
// #define USES_P092   // DLbus
// #define USES_P093   // MitsubishiHP
// #define USES_P094   // CULReader
// #define USES_P095   // ILI9341
// #define USES_P096   // eInk
// #define USES_P097   // ESP32Touch
// #define USES_P098   //
// #define USES_P099   // XPT2046 touchscreen

// #define USES_P100   // DS2423 counter
// #define USES_P101   // WakeOnLan
// #define USES_P102   // PZEM004Tv3
// #define USES_P103   // Atlas_EZO_pH
// #define USES_P104   // Atlas_EZO_EC
// #define USES_P105   // Atlas_EZO_ORP
// #define USES_P106   // BME680
// #define USES_P107   // Si1145

// Special plugins needing IR library
 #define USES_P016   // IR
// #define P016_SEND_IR_TO_CONTROLLER false //IF true then the JSON replay solution is transmited back to the condroller.
// #define USES_P035   // IRTX
// #define P016_P035_Extended_AC // The following define is needed for extended decoding of A/C Messages and or using standardised
                                 //common arguments for controlling all deeply supported A/C units
// #define P016_P035_USE_RAW_RAW2 //Use the RAW and RAW2 encodings, disabling it saves 3.7Kb
// #define USES_P088   // Heatpump IR
// #define USES_P108   // DDS238-x ZN Modbus energy meters

/*
 #######################################################################################################
 ###########     Controllers
 #######################################################################################################
 */

// #define USES_C001   // Domoticz HTTP
 #define USES_C002   // Domoticz MQTT
// #define USES_C003   // Nodo telnet
 #define USES_C004   // ThingSpeak
 #define USES_C005   // Home Assistant (openHAB) MQTT
// #define USES_C006   // PiDome MQTT
// #define USES_C007   // Emoncms
 #define USES_C008   // Generic HTTP
// #define USES_C009   // FHEM HTTP
// #define USES_C010   // Generic UDP
 #define USES_C011   // Generic HTTP Advanced
// #define USES_C012   // Blynk HTTP
 #define USES_C013   // ESPEasy P2P network
 #define USES_C014   // homie 3 & 4dev MQTT
// #define USES_C015   // Blynk
// #define USES_C016   // Cache controller
// #define USES_C017   // Zabbix
// #define USES_C018   // TTN/RN2483

/*
 #######################################################################################################
 ###########     Notifiers
 #######################################################################################################
 */

 #define USES_N001   // Email
// #define USES_N002   // Buzzer

#endif // ESPEASY_CUSTOM_H

TD-er, 2021-01-19:
There is now a custom_IR PIO environment.
So you should select to build `custom_IR_ESP8266_4M1M` if you need IR plugins.

TD-er, 2021-01-19:
Hmm have to check that later this evening.
I split the IR stuff and also removed the libraries that were unlikely to be combined to get rid of build issues (mainly issues in Windows).

Is there a plugin you try to use which needs this Adafruit library?
Or else there may have been an update on another library now stating a dependency.

ghtester, 2021-01-19:
I am not sure which libraries are used in plugins selected... I am using plugin list above, with one exception - tried to enable also P035 - IRTX. 
I'll try to get more information and perhaps disable some plugins (suspected to contain Adafruit library).
Thanks for your help.

TD-er, 2021-01-19:
I can reproduce the build error.
Will fix it and then merge it to the mega branch so you can test it :)

TD-er, 2021-01-19:
Merged, so please let me know if we can keep this issue closed :)

closed 2021-01-19 by TD-er

ghtester, 2021-01-19:
An excellent, job, Gijs! Compilation finished. Thanks a lot! :-)

TD-er, 2021-01-19:
You're welcome.
Lots of new features for you to test now :)

PR #3470 (2021-01-19): [Build] Fix custom IR build #3469
Fixes: #3469
commit 30fc6d3d (2021-01-19): [Build] Fix custom IR build #3469

Fixes: #3469