letscontrolit/ESPEasy #3471: Please enable or fix mDNS or SSDP so i can reacht ESPEasy by hostname without a DNS-server

lansi002, 2021-01-21:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

TD-er, 2021-01-21:
Have you tried a build with mDNS enabled? 
`test_ESP8266_4M1M_VCC_MDNS_SD` is too big lately so that PIO env. is not included in the latest builds.

But in the "custom" builds, the `FEATURE_MDNS` flag is set so it should be included.

lansi002, 2021-01-22:
Ok, thank you. I will check on that.

In the case it is in the build, is mDNS enabled by default or do i need so set a webgui propertie like SSDP in the past (which is grey out now)?

Kind regards

TD-er, 2021-01-22:
mDNS is only included in a (very) limited number of builds, so if it isn't included, you cannot enable it.

The reason why it isn't included in most builds is 2-fold:
- Does increase build size and memory usage
- Was not working at all for quite a long time as it was broken in the esp8266/Arduino core libraries for quite some time. (not entirely sure if it is working now in the "beta" builds)

lansi002, 2021-01-26:
Thank you TD-er for you help. I will try your suggestion, but in the mean time i am trying to fix the same problem using MQTT all over the place on all my ESPEasy device so that ip-numbers are not needed anymore.

I use the default Domoticz idx integration wihtin the devices and also i use MQTT import to sent soms values to ESPEasy and process that in the rule engine.

For now i am stuck with a selector switch in Domoticz from which i am not able to receive MQTT values in ESPEasy. I tried several switch devices and the dummy device in ESPEasy, but i cannot manage to receive the selector switch value to be able to process it in the rules engine.

Here is my selector switch and the MQTT payload it sends to the broker.

Domoticz selector switch:
[cid:image001.png@01D6F41F.AAEDE810]

MQTT message from Domoticz to the broker:
I need the yellow value in the ESPEasy rules engine.

{ "Battery" : 255, "LevelActions" : "|http://192.168.0.22/control?cmd=VENTMODE,AUTO|http://192.168.0.22/control?cmd=VENTMODE,LOW|http://192.168.0.22/control?cmd=VENTMODE,MIDDLE|http://192.168.0.22/control?cmd=VENTMODE,HIGH|http://192.168.0.22/control?cmd=VENTMODE,PERMANENTLOW|http://192.168.0.22/control?cmd=VENTMODE,PERMANENTMIDDLE|http://192.168.0.22/control?cmd=VENTMODE,PERMANENTHIGH|http://192.168.0.22/control?cmd=VENTMODE,NOTHOME", "LevelNames" : "UIT (NA)|AUTO|LAAG|MIDDEN|HOOG|LAAG CONTINU|MIDDEN CONTINU|HOOG CONTINU|AFWEZIG", "LevelOffHidden" : "true", "RSSI" : 12, "SelectorStyle" : "1", "description" : "", "dtype" : "Light/Switch", "hwid" : "12", "id" : "000141D2", "idx" : 386, "name" : "DucoBox - Ventilatiestand", "nvalue" : 2, "stype" : "Selector Switch", "svalue1" : "10", "switchType" : "Selector", "unit" : 1 }

Do you know a way to get this value into the rules engine in ESPEasy ?

I really spent hours on ESPeasy and Google to fix this but no result so far. Hopefully you or somebody else can give me a help.

Thank you in advance.

André

tonhuisman, 2021-01-26:
I think your message is somewhat corrupted by the markup transformation. Try putting blocks of code between triple backticks, or shorter codes enclosed in single backticks: `single backtick enclosed so html special characters like <> are preserved`
```
Block of special character stuff 
or multi-
line
code using []{}/\&<> etc.
```

tonhuisman, 2021-01-26:
Now back on topic:
Parsing and filtering MQTT messages in json format is an open PR #3424 that is waiting to be merged. You can either build that yourself or wait for it to be merged.